libretiny-eu/libretiny #101: linking error

Elkropac, 2023-03-14:
Hi,
i'm on commit
```
commit 668bd82eadc78daa944b88fb2f1932876f11a7d7 (HEAD -> platform/libretuya, origin/platform/libretuya, origin/HEAD)
```

i tried to compile firmware for bulb from #100 , but it ends with
```
Linking .pioenvs/smart_bulb_wbl2_m1_001/firmware.elf
|-- Image 1: firmware.elf
/home/elkropac/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: .pioenvs/smart_bulb_wbl2_m1_001/src/esphome/core/application.cpp.o: in function `esphome::Scheduler::Scheduler()':
/home/elkropac/workspace/hx/libretuya-esphome/esphome/config/_build/smart_bulb_wbl2_m1_001/src/esphome/core/scheduler.h:13: undefined reference to `esphome::Mutex::Mutex()'
/home/elkropac/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: .pioenvs/smart_bulb_wbl2_m1_001/src/esphome/core/scheduler.cpp.o: in function `esphome::LockGuard::~LockGuard()':
/home/elkropac/workspace/hx/libretuya-esphome/esphome/config/_build/smart_bulb_wbl2_m1_001/src/esphome/core/helpers.h:554: undefined reference to `esphome::Mutex::unlock()'
/home/elkropac/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: .pioenvs/smart_bulb_wbl2_m1_001/src/esphome/core/scheduler.cpp.o: in function `esphome::LockGuard::LockGuard(esphome::Mutex&)':
/home/elkropac/workspace/hx/libretuya-esphome/esphome/config/_build/smart_bulb_wbl2_m1_001/src/esphome/core/helpers.h:553: undefined reference to `esphome::Mutex::lock()'
/home/elkropac/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: /home/elkropac/workspace/hx/libretuya-esphome/esphome/config/_build/smart_bulb_wbl2_m1_001/src/esphome/core/helpers.h:553: undefined reference to `esphome::Mutex::lock()'
/home/elkropac/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: /home/elkropac/workspace/hx/libretuya-esphome/esphome/config/_build/smart_bulb_wbl2_m1_001/src/esphome/core/helpers.h:553: undefined reference to `esphome::Mutex::lock()'
/home/elkropac/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: /home/elkropac/workspace/hx/libretuya-esphome/esphome/config/_build/smart_bulb_wbl2_m1_001/src/esphome/core/helpers.h:553: undefined reference to `esphome::Mutex::lock()'
/home/elkropac/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: /home/elkropac/workspace/hx/libretuya-esphome/esphome/config/_build/smart_bulb_wbl2_m1_001/src/esphome/core/helpers.h:553: undefined reference to `esphome::Mutex::lock()'
/home/elkropac/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: .pioenvs/smart_bulb_wbl2_m1_001/src/esphome/core/scheduler.cpp.o:/home/elkropac/workspace/hx/libretuya-esphome/esphome/config/_build/smart_bulb_wbl2_m1_001/src/esphome/core/helpers.h:553: more undefined references to `esphome::Mutex::lock()' follow
/home/elkropac/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: .pioenvs/smart_bulb_wbl2_m1_001/src/esphome/core/scheduler.cpp.o: in function `esphome::Scheduler::call()':
/home/elkropac/workspace/hx/libretuya-esphome/esphome/config/_build/smart_bulb_wbl2_m1_001/src/esphome/core/scheduler.cpp:247: undefined reference to `esphome::Mutex::unlock()'
collect2: error: ld returned 1 exit status
|-- Generated file not found: .pioenvs/smart_bulb_wbl2_m1_001/firmware.elf
*** [.pioenvs/smart_bulb_wbl2_m1_001/firmware.elf] Error 1
```
it has config
```
substitutions:
  <<: !include _common/subst.yaml
  dev_name: smart_bulb_wbl2_m1_001
  dev_name_network: smart-bulb-wbl2-m1-001.devices
  #dev_name_network: 192.168.23.194
  log_level: INFO
  #boot_brightness: "0.9"

packages:
  esphome: !include _packages/esphome_common.yaml
  text_sensor: !include _packages/text_sensor/common.yaml
  switch: !include _packages/switch/common.yaml
  wifi: !include _packages/wifi_common.yaml

libretuya:
  board: generic-bk7231n-qfn32-tuya
  framework:
    version: dev

```

I tried to compile for another device with config
```
substitutions:
  <<: !include _common/subst.yaml
  dev_name: lvl300_001
  dev_name_network: lvl300-001.devices
  #dev_name_network: 192.168.23.194
  log_level: INFO
  boot_brightness: "0.9"

packages:
  esphome: !include _packages/esphome_common.yaml
  light: !include _packages/light/lvl300.yaml
  output: !include _packages/output/lvl300.yaml
  sensor: !include _packages/sensor/common.yaml
  text_sensor: !include _packages/text_sensor/common.yaml
  switch: !include _packages/switch/common.yaml
  wifi: !include _packages/wifi_common.yaml

esphome:
  <<: !include .common/light_on_boot_01_w2.yaml

libretuya:
  board: cb3l
  framework:
    version: dev

wifi:
  # Optional manual IP
  manual_ip:
    static_ip: 192.168.23.59
    gateway: 192.168.23.1
    subnet: 255.255.255.0
```

and it ends with the same error
```
Linking .pioenvs/lvl300_001/firmware.elf
|-- Image 1: firmware.elf
/home/elkropac/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: .pioenvs/lvl300_001/src/esphome/core/application.cpp.o: in function `esphome::Scheduler::Scheduler()':
/home/elkropac/workspace/hx/libretuya-esphome/esphome/config/_build/lvl300_001/src/esphome/core/scheduler.h:13: undefined reference to `esphome::Mutex::Mutex()'
/home/elkropac/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: .pioenvs/lvl300_001/src/esphome/core/scheduler.cpp.o: in function `esphome::LockGuard::~LockGuard()':
/home/elkropac/workspace/hx/libretuya-esphome/esphome/config/_build/lvl300_001/src/esphome/core/helpers.h:554: undefined reference to `esphome::Mutex::unlock()'
/home/elkropac/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: .pioenvs/lvl300_001/src/esphome/core/scheduler.cpp.o: in function `esphome::LockGuard::LockGuard(esphome::Mutex&)':
/home/elkropac/workspace/hx/libretuya-esphome/esphome/config/_build/lvl300_001/src/esphome/core/helpers.h:553: undefined reference to `esphome::Mutex::lock()'
/home/elkropac/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: /home/elkropac/workspace/hx/libretuya-esphome/esphome/config/_build/lvl300_001/src/esphome/core/helpers.h:553: undefined reference to `esphome::Mutex::lock()'
/home/elkropac/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: /home/elkropac/workspace/hx/libretuya-esphome/esphome/config/_build/lvl300_001/src/esphome/core/helpers.h:553: undefined reference to `esphome::Mutex::lock()'
/home/elkropac/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: /home/elkropac/workspace/hx/libretuya-esphome/esphome/config/_build/lvl300_001/src/esphome/core/helpers.h:553: undefined reference to `esphome::Mutex::lock()'
/home/elkropac/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: /home/elkropac/workspace/hx/libretuya-esphome/esphome/config/_build/lvl300_001/src/esphome/core/helpers.h:553: undefined reference to `esphome::Mutex::lock()'
/home/elkropac/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: .pioenvs/lvl300_001/src/esphome/core/scheduler.cpp.o:/home/elkropac/workspace/hx/libretuya-esphome/esphome/config/_build/lvl300_001/src/esphome/core/helpers.h:553: more undefined references to `esphome::Mutex::lock()' follow
/home/elkropac/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: .pioenvs/lvl300_001/src/esphome/core/scheduler.cpp.o: in function `esphome::Scheduler::call()':
/home/elkropac/workspace/hx/libretuya-esphome/esphome/config/_build/lvl300_001/src/esphome/core/scheduler.cpp:247: undefined reference to `esphome::Mutex::unlock()'
collect2: error: ld returned 1 exit status
|-- Generated file not found: .pioenvs/lvl300_001/firmware.elf
*** [.pioenvs/lvl300_001/firmware.elf] Error 1
```

kuba2k2, 2023-03-14:
Thanks for the report, should be fixed in kuba2k2/libretuya-esphome@d076454f6f9db93169e3a5e090e8ab94830caf5e. Update and try again.

Elkropac, 2023-03-14:
Seems good, thanks
```
Linking .pioenvs/smart_bulb_wbl2_m1_001/firmware.elf
|-- Image 1: firmware.elf
|   |-- bk7231n_app_0x011000.rbl
|   |   |-- firmware.bin
|   |-- bk7231n_app_0x011000.crc
|   |-- bk7231n_app_0x129F0A.rblh
|   |-- bk7231n_app.ota.rbl
|   |-- bk7231n_app.ota.ug.bin
Building UF2 OTA image
|-- firmware.uf2
RAM:   [====      ]  35.8% (used 93900 bytes from 262144 bytes)
Flash: [=======   ]  65.1% (used 705041 bytes from 1083136 bytes)
============================================================================================================================================================================== [SUCCESS] Took 29.91 seconds ==============================================================================================================================================================================
```

Elkropac, 2023-03-14:
Btw, to do OTA from openbk to esphome, i should use this file  `bk7231n_app.ota.rbl`?

kuba2k2, 2023-03-14:
Yes:
https://docs.libretuya.ml/docs/flashing/esphome/#migrating-from-openbeken-ota

closed 2023-03-18 by kuba2k2