libretiny-eu/libretiny #102: Compile error LockGuard

klmi, 2023-03-15:
Hi,

i'm getting the folling error during compilation:

`c:\libretuya-esphome>python -m esphome compile test.yml
INFO Reading configuration test.yml...
INFO Generating C++ source...
INFO Compiling app...
Processing leckage (board: cb3s; framework: arduino; platform: libretuya)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Warning! Non-Git installations are NOT SUPPORTED.
CONFIGURATION: https://docs.platformio.org/page/boards/libretuya/cb3s.html
PLATFORM: LibreTuya (0.12.6) > CB3S Wi-Fi Module
HARDWARE: BK7231N 120MHz, 256KB RAM, 1.03MB Flash
DEBUG: Current (openocd) External (custom, openocd)
PACKAGES:
 - framework-arduino-api @ 3.0.0-a4cbfc+sha.3a4cbfc
 - framework-beken-bdk @ 0.0.0+v2021.06.07.sha.6491b8c
 - library-flashdb@03500fa @ 3500.0.0-fa+sha.03500fa
 - library-lwip@2.1.3-bdk @ 2.1.3-bdk+sha.57a7df1
 - library-printf@6.0.0 @ 6.0.0+sha.8b831c1
 - tool-ltchiptool @ 2.0.2+sha.7559033
 - toolchain-gccarmnoneeabi @ 1.40804.0 (4.8.4)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pioenvs\leckage\src\esphome\components\api\api_connection.cpp.o
Compiling .pioenvs\leckage\src\esphome\components\api\api_frame_helper.cpp.o
Compiling .pioenvs\leckage\src\esphome\components\api\api_pb2.cpp.o
Compiling .pioenvs\leckage\src\esphome\components\api\api_pb2_service.cpp.o
In file included from src/esphome/components/api/proto.h:5:0,
                 from src/esphome/components/api/api_pb2.h:5,
                 from src/esphome/components/api/api_pb2.cpp:3:
src/esphome/core/helpers.h: In constructor 'esphome::LockGuard::LockGuard(esphome::Mutex&)':
src/esphome/core/helpers.h:553:41: error: invalid initialization of non-const reference of type 'esphome::Mutex&' from an rvalue of type '<brace-enclosed initializer list>'
   LockGuard(Mutex &mutex) : mutex_{mutex} { mutex_.lock(); }
                                         ^
In file included from src/esphome/components/api/proto.h:5:0,
                 from src/esphome/components/api/api_pb2.h:5,
                 from src/esphome/components/api/api_pb2_service.h:5,
                 from src/esphome/components/api/api_pb2_service.cpp:3:
src/esphome/core/helpers.h: In constructor 'esphome::LockGuard::LockGuard(esphome::Mutex&)':
src/esphome/core/helpers.h:553:41: error: invalid initialization of non-const reference of type 'esphome::Mutex&' from an rvalue of type '<brace-enclosed initializer list>'
   LockGuard(Mutex &mutex) : mutex_{mutex} { mutex_.lock(); }
                                         ^
*** [.pioenvs\leckage\src\esphome\components\api\api_pb2_service.cpp.o] Error 1
In file included from src/esphome/components/api/proto.h:5:0,
                 from src/esphome/components/api/api_pb2.h:5,
                 from src/esphome/components/api/api_connection.h:4,
                 from src/esphome/components/api/api_connection.cpp:1:
src/esphome/core/helpers.h: In constructor 'esphome::LockGuard::LockGuard(esphome::Mutex&)':
src/esphome/core/helpers.h:553:41: error: invalid initialization of non-const reference of type 'esphome::Mutex&' from an rvalue of type '<brace-enclosed initializer list>'
   LockGuard(Mutex &mutex) : mutex_{mutex} { mutex_.lock(); }
                                         ^
In file included from src/esphome/components/api/api_frame_helper.cpp:5:0:
src/esphome/core/helpers.h: In constructor 'esphome::LockGuard::LockGuard(esphome::Mutex&)':
src/esphome/core/helpers.h:553:41: error: invalid initialization of non-const reference of type 'esphome::Mutex&' from an rvalue of type '<brace-enclosed initializer list>'
   LockGuard(Mutex &mutex) : mutex_{mutex} { mutex_.lock(); }
                                         ^
*** [.pioenvs\leckage\src\esphome\components\api\api_pb2.cpp.o] Error 1
*** [.pioenvs\leckage\src\esphome\components\api\api_frame_helper.cpp.o] Error 1
*** [.pioenvs\leckage\src\esphome\components\api\api_connection.cpp.o] Error 1
==================================================================================== [FAILED] Took 28.65 seconds ====================================================================================
`

`c:\libretuya-esphome>git log
commit d076454f6f9db93169e3a5e090e8ab94830caf5e (HEAD -> platform/libretuya, origin/platform/libretuya, origin/HEAD)
Author: Kuba Szczodrzyński <kuba@szczodrzynski.pl>
Date:   Tue Mar 14 11:42:15 2023 +0100

    libretuya: enable Mutex usage`

running windows 10 and trying to compile for cb3s ...

anyone any ideas on how to fix this?

kind regards

catalin2402, 2023-03-15:
please change libretuya: block to 
```
libretuya:
  board: wb3s
  framework:
    version: dev
```

kuba2k2, 2023-03-15:
From what I can see, you installed LibreTuya in an unsupported way:
```
 Warning! Non-Git installations are NOT SUPPORTED.
```

Try again after removing the version you have currently installed, adding the block mentioned above to ESPHome yaml config, and running compile again.

klmi, 2023-03-15:
@catalin2402  @kuba2k2  you guys are awesome, didn't took a minute to reply! THX!
changing from cb3s to wb3s and using dev version as mentioned above made it.

```
Archiving .pioenvs\test\libbdk_ble_4_2.a
Linking .pioenvs\test\firmware.elf
|-- Image 1: firmware.elf
|   |-- bk7231t_app_0x011000.rbl
|   |   |-- firmware.bin
|   |-- bk7231t_app_0x011000.crc
|   |-- bk7231t_app_0x129F0A.rblh
|   |-- bk7231t_app.ota.rbl
|   |-- bk7231t_app.ota.ug.bin
Building UF2 OTA image
Checking size .pioenvs\leckage\firmware.elf
|-- firmware.uf2
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [====      ]  43.9% (used 115148 bytes from 262144 bytes)
Flash: [=====     ]  50.1% (used 542445 bytes from 1083136 bytes)
=================================================================================== [SUCCESS] Took 881.98 seconds ===================================================================================
INFO Successfully compiled program.
```
Case closed

enjoy your evenings

catalin2402, 2023-03-15:
You don't have to change the board type, only the framework to be on dev ;)

kuba2k2, 2023-03-15:
Actually, you can't change the board, if cb2s was correct for your device. The WB/CB must match what's in your device.