libretiny-eu/libretiny #103: configUSE_TICK_HOOK

mczerski, 2023-03-16:
Could it be possible to enable configUSE_TICK_HOOK option in framework-beken-bdk ?
This would enable me do define vApplicationTickHook function which is called for very system tick interrupt.

kuba2k2, 2023-03-16:
Yes, you can create a file called `FreeRTOSConfig.h` in `libretuya/platform/beken-72xx/config/`. This file should be included before the SDK's config, so make sure to add `#include_next "FreeRTOSConfig.h"` in the first line.
Then, you can #define or #undef any options you want. I'm not 100% sure that this will work - in the structure-refactor branch, LT will not be using BDK's FreeRTOS, and instead a separate "clean" library will be used.

mczerski, 2023-03-16:
this is working if I do this in ~/.platformio/platforms/libretuya/platform/beken-72xx/config/FreeRTOSConfig.h. But is there a way to do that in project dir ?

mczerski, 2023-03-16:
ok, thanks

kuba2k2, 2023-03-23:
I've designed this syntax to make custom options possible:
```ini
custom_options.lwip =
	NETIF_NUM_PCB=1
custom_options.freertos =
	INCLUDE_vTaskDelay = 0
custom_options.sdk_config.h =
	SOC_TYPE = 123
```
The options are placed in `platformio.ini`. Note that you can either specify a complete header file name, or one of the defined shortcuts (currently only `lwip` and `freertos` - no idea if anything else is worth adding).

This will be available on the `structure-refactor` branch. If you have any suggestions, feel free to post them here!