libretiny-eu/libretiny #104: libtretuya-esphome: Mutex compile errors

Skince, 2023-03-19:
Hello,
I'm trying to compile libretuya-esphome and am getting compilation errors. 
Tried to update everything.
Similar to https://github.com/esphome/issues/issues/4324
```
esphome:
  name: lightswitch

libretuya:
  board: cb2s
  framework:
    version: dev

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: ""

ota:
  password: ""

wifi:
  ssid: ""
  password: ""

  # Enable fallback hotspot in case wifi connection fails
  ap:
    ssid: "Lightswitch Fallback Hotspot"
    password: ""

```
Output of `python -m esphome compile` 
```INFO Reading configuration config/lightswitches.yaml...
INFO Generating C++ source...
INFO Core config or version changed, cleaning build files...
INFO Compiling app...
Processing lightswitch (board: cb2s; framework: arduino; platform: https://github.com/kuba2k2/libretuya.git)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------
Library Manager: Installing esphome/noise-c @ 0.1.4
INFO Installing esphome/noise-c @ 0.1.4
Unpacking  [####################################]  100%
Library Manager: noise-c@0.1.4 has been installed!
INFO noise-c@0.1.4 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/libsodium @ 1.10018.1
INFO Installing esphome/libsodium @ 1.10018.1
Unpacking  [####################################]  100%
Library Manager: libsodium@1.10018.1 has been installed!
INFO libsodium@1.10018.1 has been installed!
HARDWARE: BK7231N 120MHz, 256KB RAM, 1.03MB Flash
 - framework-arduino-api @ 3.0.0-a4cbfc+sha.3a4cbfc
 - framework-beken-bdk @ 0.0.0+v2021.06.07.sha.6491b8c
 - library-flashdb@03500fa @ 3500.0.0-fa+sha.03500fa
 - library-lwip@2.1.3-bdk @ 2.1.3-bdk+sha.33191e0
 - library-printf@6.0.0 @ 6.0.0+sha.8b831c1
 - tool-ltchiptool @ 2.0.2+sha.7559033
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
Compiling .pioenvs/lightswitch/src/esphome/components/api/api_connection.cpp.o
Compiling .pioenvs/lightswitch/src/esphome/components/api/api_frame_helper.cpp.o
Compiling .pioenvs/lightswitch/src/esphome/components/api/api_pb2.cpp.o
Compiling .pioenvs/lightswitch/src/esphome/components/api/api_pb2_service.cpp.o
Compiling .pioenvs/lightswitch/src/esphome/components/api/api_server.cpp.o
Compiling .pioenvs/lightswitch/src/esphome/components/api/list_entities.cpp.o
Compiling .pioenvs/lightswitch/src/esphome/components/api/proto.cpp.o
Compiling .pioenvs/lightswitch/src/esphome/components/api/subscribe_state.cpp.o
In file included from src/esphome/components/api/proto.h:5:0,
                 from src/esphome/components/api/api_pb2.h:5,
                 from src/esphome/components/api/api_pb2_service.h:5,
                 from src/esphome/components/api/api_pb2_service.cpp:3:
src/esphome/core/helpers.h: In constructor 'esphome::LockGuard::LockGuard(esphome::Mutex&)':
src/esphome/core/helpers.h:553:41: error: invalid initialization of non-const reference of type 'esphome::Mutex&' from an rvalue of type '<brace-enclosed initializer list>'
   LockGuard(Mutex &mutex) : mutex_{mutex} { mutex_.lock(); }
                                         ^
In file included from src/esphome/components/api/api_frame_helper.cpp:5:0:
src/esphome/core/helpers.h: In constructor 'esphome::LockGuard::LockGuard(esphome::Mutex&)':
src/esphome/core/helpers.h:553:41: error: invalid initialization of non-const reference of type 'esphome::Mutex&' from an rvalue of type '<brace-enclosed initializer list>'
   LockGuard(Mutex &mutex) : mutex_{mutex} { mutex_.lock(); }
                                         ^
In file included from src/esphome/components/api/proto.h:5:0,
                 from src/esphome/components/api/api_pb2.h:5,
                 from src/esphome/components/api/api_pb2.cpp:3:
src/esphome/core/helpers.h: In constructor 'esphome::LockGuard::LockGuard(esphome::Mutex&)':
src/esphome/core/helpers.h:553:41: error: invalid initialization of non-const reference of type 'esphome::Mutex&' from an rvalue of type '<brace-enclosed initializer list>'
   LockGuard(Mutex &mutex) : mutex_{mutex} { mutex_.lock(); }
                                         ^
In file included from src/esphome/components/api/proto.h:5:0,
                 from src/esphome/components/api/api_pb2.h:5,
                 from src/esphome/components/api/api_connection.h:4,
                 from src/esphome/components/api/api_connection.cpp:1:
src/esphome/core/helpers.h: In constructor 'esphome::LockGuard::LockGuard(esphome::Mutex&)':
src/esphome/core/helpers.h:553:41: error: invalid initialization of non-const reference of type 'esphome::Mutex&' from an rvalue of type '<brace-enclosed initializer list>'
   LockGuard(Mutex &mutex) : mutex_{mutex} { mutex_.lock(); }
                                         ^
In file included from src/esphome/components/api/proto.h:5:0,
                 from src/esphome/components/api/proto.cpp:1:
src/esphome/core/helpers.h: In constructor 'esphome::LockGuard::LockGuard(esphome::Mutex&)':
src/esphome/core/helpers.h:553:41: error: invalid initialization of non-const reference of type 'esphome::Mutex&' from an rvalue of type '<brace-enclosed initializer list>'
   LockGuard(Mutex &mutex) : mutex_{mutex} { mutex_.lock(); }
                                         ^
In file included from src/esphome/core/component_iterator.h:5:0,
                 from src/esphome/components/api/list_entities.h:4,
                 from src/esphome/components/api/list_entities.cpp:1:
src/esphome/core/helpers.h: In constructor 'esphome::LockGuard::LockGuard(esphome::Mutex&)':
src/esphome/core/helpers.h:553:41: error: invalid initialization of non-const reference of type 'esphome::Mutex&' from an rvalue of type '<brace-enclosed initializer list>'
   LockGuard(Mutex &mutex) : mutex_{mutex} { mutex_.lock(); }
                                         ^
*** [.pioenvs/lightswitch/src/esphome/components/api/api_frame_helper.cpp.o] Error 1
In file included from src/esphome/components/api/proto.h:5:0,
                 from src/esphome/components/api/api_pb2.h:5,
                 from src/esphome/components/api/api_server.h:8,
                 from src/esphome/components/api/api_server.cpp:1:
src/esphome/core/helpers.h: In constructor 'esphome::LockGuard::LockGuard(esphome::Mutex&)':
src/esphome/core/helpers.h:553:41: error: invalid initialization of non-const reference of type 'esphome::Mutex&' from an rvalue of type '<brace-enclosed initializer list>'
   LockGuard(Mutex &mutex) : mutex_{mutex} { mutex_.lock(); }
                                         ^
*** [.pioenvs/lightswitch/src/esphome/components/api/proto.cpp.o] Error 1
*** [.pioenvs/lightswitch/src/esphome/components/api/api_pb2_service.cpp.o] Error 1
In file included from src/esphome/core/component_iterator.h:5:0,
                 from src/esphome/components/api/subscribe_state.h:4,
                 from src/esphome/components/api/subscribe_state.cpp:1:
src/esphome/core/helpers.h: In constructor 'esphome::LockGuard::LockGuard(esphome::Mutex&)':
src/esphome/core/helpers.h:553:41: error: invalid initialization of non-const reference of type 'esphome::Mutex&' from an rvalue of type '<brace-enclosed initializer list>'
   LockGuard(Mutex &mutex) : mutex_{mutex} { mutex_.lock(); }
                                         ^
*** [.pioenvs/lightswitch/src/esphome/components/api/api_connection.cpp.o] Error 1
*** [.pioenvs/lightswitch/src/esphome/components/api/list_entities.cpp.o] Error 1
*** [.pioenvs/lightswitch/src/esphome/components/api/api_server.cpp.o] Error 1
*** [.pioenvs/lightswitch/src/esphome/components/api/subscribe_state.cpp.o] Error 1
*** [.pioenvs/lightswitch/src/esphome/components/api/api_pb2.cpp.o] Error 1
==================================================================== [FAILED] Took 7.01 seconds ====================================================================

```

catalin2402, 2023-03-19:
You have to update libretuya to latest version using: `platformio platform install -f https://github.com/kuba2k2/libretuya` If this doesn't work, go to the folder where platformio installed libretuya, delete the libretuya folder from the platforms and try to compile again.

Skince, 2023-03-19:
Thanks! 
After `platformio platform install -f https://github.com/kuba2k2/libretuya` didn't work 
I had to delete `~/.platformio/platforms/libretuya` and `~/.platformio/platforms/libretuya@src-6a8f78c3f548d4e6960a5bf965ba3900`. That solved it.

closed 2023-03-19 by Skince