libretiny-eu/libretiny #105: Compilation issues with latest commits on libretuya-esphome

harsh2029, 2023-03-23:
I started to get below compilation errors after pulling the `platform/libretuya-esphome` to the latest version of March 14
```
Processing tuyaswitch-switch01 (board: cb2s; framework: arduino; platform: https://github.com/kuba2k2/libretuya.git)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
HARDWARE: BK7231N 120MHz, 256KB RAM, 1.03MB Flash
 - framework-arduino-api @ 3.0.0-a4cbfc+sha.3a4cbfc
 - framework-beken-bdk @ 0.0.0+v2021.06.07.sha.6491b8c
 - library-flashdb@03500fa @ 3500.0.0-fa+sha.03500fa
 - library-lwip@2.1.3-bdk @ 2.1.3-bdk+sha.33191e0
 - library-printf@6.0.0 @ 6.0.0+sha.8b831c1
 - tool-ltchiptool @ 2.0.2+sha.7559033
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- ESPAsyncWebServer-esphome @ 3.0.0
|   |-- AsyncTCP-esphome @ 2.0.0
|-- ArduinoJson @ 6.18.5
Compiling .pioenvs/tuyaswitch-switch01/src/esphome/components/api/api_connection.cpp.o
Compiling .pioenvs/tuyaswitch-switch01/src/esphome/components/api/api_frame_helper.cpp.o
Compiling .pioenvs/tuyaswitch-switch01/src/esphome/components/api/api_pb2.cpp.o
Compiling .pioenvs/tuyaswitch-switch01/src/esphome/components/api/api_pb2_service.cpp.o
In file included from src/esphome/components/api/proto.h:5:0,
                 from src/esphome/components/api/api_pb2.h:5,
                 from src/esphome/components/api/api_pb2.cpp:3:
src/esphome/core/helpers.h: In constructor 'esphome::LockGuard::LockGuard(esphome::Mutex&)':
src/esphome/core/helpers.h:553:41: error: invalid initialization of non-const reference of type 'esphome::Mutex&' from an rvalue of type '<brace-enclosed initializer list>'
   LockGuard(Mutex &mutex) : mutex_{mutex} { mutex_.lock(); }
                                         ^
In file included from src/esphome/components/api/proto.h:5:0,
                 from src/esphome/components/api/api_pb2.h:5,
                 from src/esphome/components/api/api_connection.h:4,
                 from src/esphome/components/api/api_connection.cpp:1:
src/esphome/core/helpers.h: In constructor 'esphome::LockGuard::LockGuard(esphome::Mutex&)':
src/esphome/core/helpers.h:553:41: error: invalid initialization of non-const reference of type 'esphome::Mutex&' from an rvalue of type '<brace-enclosed initializer list>'
   LockGuard(Mutex &mutex) : mutex_{mutex} { mutex_.lock(); }
                                         ^
In file included from src/esphome/components/api/proto.h:5:0,
                 from src/esphome/components/api/api_pb2.h:5,
                 from src/esphome/components/api/api_pb2_service.h:5,
                 from src/esphome/components/api/api_pb2_service.cpp:3:
src/esphome/core/helpers.h: In constructor 'esphome::LockGuard::LockGuard(esphome::Mutex&)':
src/esphome/core/helpers.h:553:41: error: invalid initialization of non-const reference of type 'esphome::Mutex&' from an rvalue of type '<brace-enclosed initializer list>'
   LockGuard(Mutex &mutex) : mutex_{mutex} { mutex_.lock(); }
                                         ^
In file included from src/esphome/components/api/api_frame_helper.cpp:5:0:
src/esphome/core/helpers.h: In constructor 'esphome::LockGuard::LockGuard(esphome::Mutex&)':
src/esphome/core/helpers.h:553:41: error: invalid initialization of non-const reference of type 'esphome::Mutex&' from an rvalue of type '<brace-enclosed initializer list>'
   LockGuard(Mutex &mutex) : mutex_{mutex} { mutex_.lock(); }
                                         ^
*** [.pioenvs/tuyaswitch-switch01/src/esphome/components/api/api_pb2_service.cpp.o] Error 1
*** [.pioenvs/tuyaswitch-switch01/src/esphome/components/api/api_frame_helper.cpp.o] Error 1
*** [.pioenvs/tuyaswitch-switch01/src/esphome/components/api/api_connection.cpp.o] Error 1
*** [.pioenvs/tuyaswitch-switch01/src/esphome/components/api/api_pb2.cpp.o] Error 1
================================================================================= [FAILED] Took 5.93 seconds =================================================================================
```

I had to go back to the Feb 14 commit for successful compilation

[image]

Did I miss something in the configuration yaml to work with this new mutex feature?

kuba2k2, 2023-03-23:
You need to update LibreTuya, not ESPHome-LibreTuya. Use framework: version: dev in YAML.

harsh2029, 2023-03-23:
Thanks!

I was already using dev framework
```
libretuya:
  board: cb2s
  framework:
    version: dev
```
However somehow the libretuya platformio installation was not updating.
I reinstalled everything and it worked after that.

closed 2023-03-24 by kuba2k2