libretiny-eu/libretiny #107: error installing platform on Windows

Masterz69, 2023-04-11:
```
C:\_libretuya_>platformio platform install -f https://github.com/kuba2k2/libretuya

WARNING: This command is deprecated and will be removed in the next releases.                                            
Please use `pio pkg install` instead.                                                                                                                                                                                                             

Installing/updating ltchiptool                                                                                           
'C:\Program' is not recognized as an internal or external command,                                                      
operable program or batch file.                                                                                          
!!! Installing ltchiptool failed, or version outdated. Please install ltchiptool manually using pip. Cannot continue. <member 'name' of 'ImportError' objects>: No module named 'ltchiptool'                                                      
Error: Unknown development platform 'libretuya'
```

Looks some software fails to work with Windows pathes with spaces inside.
Any idea how to fix - welcome.

Masterz69, 2023-04-11:
After manually installing package by ``` pip install ltchiptool``` - platform installation continues.

Masterz69, 2023-04-11:
> Your Python is probably installed in `C:\Program Files`

Yes. Thank You.

kuba2k2, 2023-08-18:
This should be fixed for some time now.

closed 2023-08-18 by kuba2k2