libretiny-eu/libretiny #108: Cannot update the device post initial install

fokcuk, 2023-04-13:
Hello
For some reason the update has stopped working. I changed the code, but even though it says it compiled and OTA pushed through ESPHome, the device is still not updated - old sensor names.

I tried deleting the device and recreating it in ESPHome, but its still the same thing - no update

kuba2k2, 2023-04-13:
What version of libretuya and what chip are you using? What board name did you choose?

fokcuk, 2023-04-14:
turned out to be a device specific issue - if you hold the button after powering it off and then plugging back in with button pressed, it will start to use firmware. So its loading it, but not using is for some reason.
Closing the issue

closed 2023-04-14 by fokcuk

kuba2k2, 2023-04-14:
This is very weird, can you tell me more about this device? Is it Realtek or Beken?

fokcuk, 2023-04-14:
Beken

kuba2k2, 2023-04-14:
Then it's not really possible to update depending on the button presses. The bootloader should update the firmware as soon as it receives it.