libretiny-eu/libretiny #109: openBK ota not working

klmi, 2023-04-16:
Hi,

i successfully flashed an bk7231n with openbk.
now i'm trying to include it in esp home with ota update but it doesn't accept the file as suggested [here ](https://community.home-assistant.io/t/wifi-night-light-with-motion-globe-50239-wb3s-localtuya-openbeken-libretuya-esphome/459505/11)

(renamed bk7231n_app.ota.rbl to bk7231n_esphome.rbl, and all other files, non worked):
[image]

Also flashing it directly to the chip via ltchiptool oder bk7231flasher doesnt work...the chip just starts, prints a few lines about itself and gets stuck there

any idea why ota doesn't work? is there a way to some logging about it?

kuba2k2, 2023-04-16:
As said in the migration guide, the file has to be named OpenBK7231N_esphome.rbl, not just bk7231n.

klmi, 2023-04-16:
*facepalm* thx @kuba2k2

closed 2023-04-16 by klmi

kuba2k2, 2023-05-19:
The folder is created as soon as you run the `esphome compile` command. The program needs to be compiled first, only then the RBL will be available.
If you'll having issues with getting the folder to show up, you can try [ESPHome-Kickstart](https://github.com/libretiny-eu/esphome-kickstart), which has `.ota.rbl` binaries that you can use. After that, watch [this video](https://youtu.be/sSj8f-HCHQ0?t=497) (start at 8:17) to see how to proceed from then.

ElVit, 2023-05-19:
Thanks for your help.
I'm using the Home Assistant Addon `LibreTiny ESPHome` and unfortunately there is no `Compile` button anymore.
So I used Portainer to open a terminal in the container of the addon.
After execting the command `esphome compile {devicename}` I could find the RBL files in the `.pioenvs` folder.

I could also flash the RBL file and now its working 😃 
Thanks again.

kuba2k2, 2023-05-19:
Well, that's not exactly what I meant, but I'm glad it worked.

The files generate after compiling the program, which equals to pressing "Install" in home assistant. I previously assumed that you were using the command line. After compiling in HA, you would find the required files using Portainer.