libretiny-eu/libretiny #110: NVS length does not match (0!=98)

klmi, 2023-04-16:
Hallo again,

this time I'm facing a loop during wifi connection start...
I setup a testnetwork with only wpa2 but connection can't be asstablished as shown below.
whenever it's suppose to connect 'NVS length does not match (0!=98)' does appear and it's scanning again for networks, finds the configured one and fail.

'''
[I][wifi:258]: WiFi Connecting to 'IoTNetwork'...
[V][wifi:260]: Connection Params:
[V][wifi:261]:   SSID: 'IoTNetwork'
[V][wifi:264]:   BSSID: redacted
[V][wifi:284]:   Password: 'IoTNetwork2023'
[V][wifi:289]:   Channel: 1
[V][wifi:298]:   Using DHCP IP
[V][wifi:300]:   Hidden: NO
I [      0.000] LibreTuya v0.12.6 on cb3s, compiled at Apr 16 2023 15:59:00
[I][logger:299]: Log initialized
[VV][lt.preferences:064]: fdb_kv_get_blob: key: 233825507, len: 4
[C][ota:477]: There have been 6 suspected unsuccessful boot attempts.
[VV][lt.preferences:041]: s_pending_save: key: 233825507, len: 4
[D][lt.preferences:104]: Saving 1 preferences to flash...
[VV][lt.preferences:113]: Checking if FDB data 233825507 has changed
[V][lt.preferences:115]: sync: key: 233825507, len: 4
[D][lt.preferences:133]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[I][app:029]: Running through setup()...
[V][app:030]: Sorting components by setup priority...
[VV][scheduler:063]: set_interval(name='update', interval=30000, offset=2966)
[D][binary_sensor:034]: 'Leckage 1 Status': Sending initial state OFF
[V][text_sensor:013]: 'Leckage 1 ESPHome Version': Received new state 2023.4.0-dev Apr 16 2023, 15:53:24
[D][text_sensor:064]: 'Leckage 1 ESPHome Version': Sending state '2023.4.0-dev Apr 16 2023, 15:53:24'
[V][text_sensor:013]: 'Leckage 1 Mac Wifi Address': Received new state redacted
[D][text_sensor:064]: 'Leckage 1 Mac Wifi Address': Sending state 'redacted'
[C][wifi:038]: Setting up WiFi...
[C][wifi:039]:   Local MAC: redacted
[VV][lt.preferences:061]: NVS length does not match (0!=98)
[V][wifi_lt:034]: Enabling STA.
[V][wifi_lt:262]: Event: WiFi STA start
[V][wifi:064]: Setting Power Save Option failed!
[D][wifi:387]: Starting scan...
[VV][scheduler:226]: Running interval 'update' with interval=30000 last_execution=4294934390 (now=476)
[V][sensor:059]: 'Leckage 1 Uptime Sensor': Received new state 0.482000
[D][sensor:110]: 'Leckage 1 Uptime Sensor': Sending state 0.48200 s with 0 decimals of accuracy
[V][wifi_lt:256]: Event: WiFi Scan Done status=0 number=32 scan_id=0
[D][wifi:402]: Found networks:
[I][wifi:446]: - 'IoTNetwork' (redacted) ▂▄▆█
[D][wifi:447]:     Channel: 1
[D][wifi:448]:     RSSI: -44 dB
[...]
[I][wifi:258]: WiFi Connecting to 'IoTNetwork'...
[V][wifi:260]: Connection Params:
[V][wifi:261]:   SSID: 'IoTNetwork'
[V][wifi:264]:   BSSID: redacted
[V][wifi:284]:   Password: 'IoTNetwork2023'
[V][wifi:289]:   Channel: 1
[V][wifi:298]:   Using DHCP IP
[V][wifi:300]:   Hidden: NO
I [      0.000] LibreTuya v0.12.6 on cb3s, compiled at Apr 16 2023 15:59:00
[I][logger:299]: Log initialized
[...}
'''
AP Log;
[image]

my current config: (same result if used dev instead of latest)
'''
substitutions:
  name: leckage
  friendly_name: "Leckage 1"
  ota_password: !secret ota_password
  api_key:  !secret api_key
  wifi_ssid: !secret wifi_ssid_wpa2
  wifi_password: !secret wifi_password_wpa2
  domain: !secret wifi_domain
  ap_point: "Leckage-Fallback-AP"
  ap_point_password: "Test#123"
  log_level: VERY_VERBOSE
  
esphome:
  name: leckage

libretuya:
  board: cb3s 
  framework:
    version: latest
#  lt_config:
#    LT_LOG_HEAP: 1
#    LT_UART_DEFAULT_PORT: 2
#    LT_UART_SILENT_ALL: 0

logger:
  level: $log_level
api:
  encryption:
    key: $api_key

ota:
  password: $ota_password

wifi:
  ssid: $wifi_ssid
  password: $wifi_password
  domain: $domain

  ap:
    ssid: $ap_point
    password: $ap_point_password

captive_portal:

web_server:
    port: 80

switch:
  - platform: restart
    name: '${friendly_name} restart'

binary_sensor:
  - platform: status
    name: '${friendly_name} Status'

sensor:
  - platform: wifi_signal
    name: '${friendly_name} WiFi Signal'
    update_interval: 60s
  - platform: uptime
    name: '${friendly_name} Uptime Sensor'
    update_interval: 30s

text_sensor:
  - platform: version
    name: '${friendly_name} ESPHome Version'
  - platform: wifi_info
    ip_address:
      name: '${friendly_name} IP Address'
    ssid:
      name: '${friendly_name} Connected SSID'
    bssid:
      name: '${friendly_name} Connected BSSID'
    mac_address:
      name: '${friendly_name} Mac Wifi Address'
    scan_results:
      name: '${friendly_name} Latest Scan Results'
'''

Any idea?
in the meantime i'll try with lower loglevel

kuba2k2, 2023-04-16:
This is a verbose message, not necessarily an error. You probably don't need VV loglevel, so the message will go away anyways.

As for why it's rebooting, I assume you're powering it from an UART adapter or something other on your bench. The power supply is most likely not enough for wifi connection, thus causing reboots.

klmi, 2023-04-16:
Hi,

was also my first guess but it's powered by an laboratory power supply (3.3V) and power ist directly attached to the chip.

Sure, NVS Messages is a VV message, but it's the most i can get from the logs.

The device itself is sth like [this](https://www.elektroda.com/rtvforum/topic3954826.html)
But i guess it's a hardware failure, I've a couple of other sensors all wb3s based and will go on with these

closed 2023-07-13 by kuba2k2