libretiny-eu/libretiny #113: EMW3080/RTL8710BN: WiFi connection failure after upgrade to LT1.0.1

hn, 2023-04-29:
I upgraded from LibreTuya to LibreTiny via OTA and the EMW3080 stick did not reconnect to network. Analysis via serial port showed:

```
[wifi:258]: WiFi Connecting to 'MY-WIFI'...
[   1697.075] WIFI: Connection failed; ret=-1
[W][wifi_lt:119]: esp_wifi_connect failed! 4
[E][wifi:304]: wifi_sta_connect_ failed!
[W][wifi_lt:288]: Event: Disconnected ssid='' bssid=00:00:00:00:00:00 reason='Unspecified'
```
I took the stick out of the room where the solar inverter is located and placed it near the router antenna. With this, the stick gets an IP address but restarts shortly after during the boot process.

With LibreTuya the WiFi connection never was a problem (even in the room where the solar inverter is stored).

For now, I re-flashed an old LibreTuya firmware bin and everything is working fine again.

```
Processing solis-inv (board: generic-rtl8710bx-4mb-980k; framework: arduino; platform: libretiny)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------
HARDWARE: RTL8710BN 125MHz, 256KB RAM, 980KB Flash
 - framework-arduino-api @ 2022.8.24+sha.237b10a 
 - framework-realtek-amb1 @ 0.0.0+v2022.06.21.sha.c4e44ef 
 - library-flashdb @ 1.2.0+sha.d5c892f 
 - library-freertos @ 8.1.2+sha.776ae6c 
 - library-freertos-port @ 2023.3.13+sha.bd96e82 
 - library-lwip @ 2.1.3-amb1+sha.6297b80 
 - library-printf @ 6.1.0+sha.28a79bd 
 - tool-ltchiptool @ 4.0.0+sha.788ba4e 
PLATFORM VERSIONS:
 - libretiny @ 1.0.1+sha.d332315
 - ltchiptool @ 4.0.1
CUSTOM OPTIONS:
 - fw_name = esphome
 - fw_version = 2023.5.0-dev
Dependency Graph
|-- ESPAsyncWebServer-esphome @ 3.0.0+sha.9f822c0
|   |-- AsyncTCP-esphome @ 2.0.0+sha.aab1fe4
|-- DNSServer @ 1.1.0
Compiling .pioenvs/solis-inv/src/main.cpp.o
```

kuba2k2, 2023-04-29:
There were some wifi related changes in Realtek that were meant to make reconnection better (and it did in my case). Apparently not for everyone.

Is the old bin compiled with latest 0.13.0/0.12.6 LibreTuya? Or something older than this?

hn, 2023-04-29:
This is the bin I re-flashed (and which is working ok):

```
$ strings LibreTuya-image_0x00B000.ota1.bin | grep -i compi
ESPHome version 2023.4.0-dev compiled on %s
VDIWEFLibreTuya v0.12.6+sha.24832d3.dirty on generic-rtl8710bx-4mb-980k, compiled at Mar 29 2023 21:40:21
```

If you give me a rough hint where to look I can do some trial-and-error changing/testing.

kuba2k2, 2023-04-29:
Well, I guess you could compare the WiFi impl between 0.13.0 and 1.0.0 on Realtek... I sadly don't have time to do this right now, but that's what I would begin with.

hn, 2023-05-07:
WiFi seems to be OK for normal operation, at least for the last days.

However, this morning (the inverter switches off during the night), the stick was initially unable to connect to WiFi (`wifi_sta_connect_ failed! ... bssid=00:00:00:00:00:00 reason='Unspecified'`) and also failed to join the WiFi for an hour, although the network was shown in the network scan. I had to power cycle the stick. But this effect only happened once, I can't tell yet if it's a real problem or just a glitch in the matrix. I will report again if there are any new findings.

closed 2023-05-07 by hn

kuba2k2, 2023-05-07:
This is a problem I couldn't fix on Realtek. The SDK doesn't like disconnecting from the AP, it usually can't reconnect back to it. I have tried many options, and none of them worked.

hn, 2023-05-25:
> This is a problem I couldn't fix on Realtek. The SDK doesn't like disconnecting from the AP, it usually can't reconnect back to it. I have tried many options, and none of them worked.

Within [ESPhome WiFi component](https://esphome.io/components/wifi.html) there is a 'reboot_timeout=15min' setting ("The amount of time to wait before rebooting when no WiFi connection exists"). Unfortunately this problem seems not to be catched by this counter, the device is caught in an endless loop. Is there a way to reboot the device if WiFi STA fails (re-)connecting?

ShonP40, 2023-05-29:
> This is a problem I couldn't fix on Realtek. The SDK doesn't like disconnecting from the AP, it usually can't reconnect back to it. I have tried many options, and none of them worked.

Is there a way to use `v0.13.0` in ESPHome then?

kuba2k2, 2023-05-29:
No, there's no easy way to do this.
As per https://github.com/hn/ginlong-solis/issues/2#issuecomment-1531514620, the issue causing WiFi problems is already fixed.

ShonP40, 2023-05-29:
I'm still getting it with a RTL8710BN on the latest v1.0.2 though

Like, sometimes it'll connect, but sometimes I'll need to disconnect&reconnect it a few times for it to connect to my WiFi

Log shows the same `Event: Disconnected ssid='' bssid=00:00:00:00:00:00 reason='Unspecified'` error @hn got

kuba2k2, 2023-05-29:
And that's how it's been for the whole time, I think. I was getting these errors on 0.13.0 as well.

ShonP40, 2023-05-29:
ah, well, I can't install the Midea UART integration anyways (requires the `ESP8266WiFi` lib) so kinda no point for me to debug it at the moment

LibreTiny works great on Tuya ESP clones tho