libretiny-eu/libretiny #115: Unable to build docker image with libretiny

ryuzaki-ryuga, 2023-04-30:
Hi,

I'm unable to build the docker image of libretiny, with this git error code. I am running Fedora 37, is anyone able to help? Thanks in advance.

`sudo python docker/build.py --tag libretiny --arch amd64 --build-type docker build
$ docker buildx build --build-arg BASEIMGTYPE=docker --build-arg BUILD_VERSION=libretiny --cache-from type=registry,ref=ghcr.io/esphome/esphome-amd64:cache-dev --file docker/Dockerfile --platform linux/amd64 --target docker --tag esphome/esphome-amd64:libretiny --tag esphome/esphome-amd64:dev --tag ghcr.io/esphome/esphome-amd64:libretiny --tag ghcr.io/esphome/esphome-amd64:dev .
[+] Building 9.8s (7/15)                                                                                               
 => [internal] load .dockerignore                                                                                 0.7s
 => => transferring context: 1.38kB                                                                               0.0s
 => [internal] load build definition from Dockerfile                                                              0.6s
 => => transferring dockerfile: 4.69kB                                                                            0.0s
 => [internal] load metadata for docker.io/library/debian:bullseye-20230208-slim                                  1.8s
 => importing cache manifest from ghcr.io/esphome/esphome-amd64:cache-dev                                         1.6s
 => CACHED [base-docker 1/1] FROM docker.io/library/debian:bullseye-20230208-slim@sha256:8eaee63a5ea83744e62d5bf  0.0s
 => [internal] load build context                                                                                 1.4s
 => => transferring context: 8.95MB                                                                               0.5s
 => ERROR [base 1/5] RUN     apt-get update     && apt-get install -y --no-install-recommends         python3=3.  5.6s
------                                                                                                                 
 > [base 1/5] RUN     apt-get update     && apt-get install -y --no-install-recommends         python3=3.9.2-3         python3-pip=20.3.4-4+deb11u1         python3-setuptools=52.0.0-4         python3-pil=8.1.2+dfsg-0.3+deb11u1         python3-cryptography=3.3.2-1         python3-venv=3.9.2-3         iputils-ping=3:20210202-1         git=1:2.30.2-1         curl=7.74.0-1.3+deb11u7         openssh-client=1:8.4p1-5+deb11u1     && rm -rf         /tmp/*         /var/{cache,log}/*         /var/lib/apt/lists/*:                                                                                       
#0 1.379 Get:1 http://deb.debian.org/debian bullseye InRelease [116 kB]
#0 1.397 Get:2 http://deb.debian.org/debian-security bullseye-security InRelease [48.4 kB]
#0 1.398 Get:3 http://deb.debian.org/debian bullseye-updates InRelease [44.1 kB]
#0 1.559 Get:4 http://deb.debian.org/debian bullseye/main amd64 Packages [8183 kB]
#0 2.124 Get:5 http://deb.debian.org/debian-security bullseye-security/main amd64 Packages [237 kB]
#0 2.130 Get:6 http://deb.debian.org/debian bullseye-updates/main amd64 Packages [14.6 kB]
#0 3.340 Fetched 8643 kB in 2s (4380 kB/s)
#0 3.340 Reading package lists...
#0 3.937 Reading package lists...
#0 4.577 Building dependency tree...
#0 4.716 Reading state information...
#0 4.836 E: Version '1:2.30.2-1' for 'git' was not found
------
Dockerfile:18
--------------------
  17 |     
  18 | >>> RUN \
  19 | >>>     apt-get update \
  20 | >>>     # Use pinned versions so that we get updates with build caching
  21 | >>>     && apt-get install -y --no-install-recommends \
  22 | >>>         python3=3.9.2-3 \
  23 | >>>         python3-pip=20.3.4-4+deb11u1 \
  24 | >>>         python3-setuptools=52.0.0-4 \
  25 | >>>         python3-pil=8.1.2+dfsg-0.3+deb11u1 \
  26 | >>>         python3-cryptography=3.3.2-1 \
  27 | >>>         python3-venv=3.9.2-3 \
  28 | >>>         iputils-ping=3:20210202-1 \
  29 | >>>         git=1:2.30.2-1 \
  30 | >>>         curl=7.74.0-1.3+deb11u7 \
  31 | >>>         openssh-client=1:8.4p1-5+deb11u1 \
  32 | >>>     && rm -rf \
  33 | >>>         /tmp/* \
  34 | >>>         /var/{cache,log}/* \
  35 | >>>         /var/lib/apt/lists/*
  36 |     
--------------------
ERROR: failed to solve: process "/bin/sh -c apt-get update     && apt-get install -y --no-install-recommends         python3=3.9.2-3         python3-pip=20.3.4-4+deb11u1         python3-setuptools=52.0.0-4         python3-pil=8.1.2+dfsg-0.3+deb11u1         python3-cryptography=3.3.2-1         python3-venv=3.9.2-3         iputils-ping=3:20210202-1         git=1:2.30.2-1         curl=7.74.0-1.3+deb11u7         openssh-client=1:8.4p1-5+deb11u1     && rm -rf         /tmp/*         /var/{cache,log}/*         /var/lib/apt/lists/*" did not complete successfully: exit code: 100
Command failed`

blfriedman, 2023-05-01:
I have the same issue.

ryuzaki-ryuga, 2023-05-03:
Fix was to go into docker/Dockerfile and replace **git=1:2.30.2-1** to **git**

kuba2k2, 2023-05-03:
The esphome fork repository has been updated already, please pull the latest changes and it will work.

closed 2023-05-03 by kuba2k2