libretiny-eu/libretiny #118: Compile error after installing the latest libretiny version

rmandjes, 2023-05-02:
I've updated to the latest tinylibre version. If I want to compile my esphome config, I get the following error message:
INFO Reading configuration /config/libretuya-esphome/smart-socket.yaml...
INFO Detected timezone 'Europe/Amsterdam'
INFO Generating C++ source...
INFO Compiling app...
Processing smart-socket (board: cb2s; framework: arduino; platform: https://github.com/kuba2k2/libretuya.git)
--------------------------------------------------------------------------------
HARDWARE: BK7231N 120MHz, 256KB RAM, 1.03MB Flash
 - framework-arduino-api @ 2022.8.24+sha.237b10a 
 - framework-beken-bdk @ 0.0.0+v2021.06.07.sha.6491b8c 
 - library-flashdb @ 1.2.0+sha.d5c892f 
 - library-freertos @ 9.0.0+sha.95cc959 
 - library-freertos-port @ 2023.3.13+sha.bd96e82 
 - library-lwip @ 2.1.3-bdk+sha.4ee4d34 
 - library-printf @ 6.1.0+sha.28a79bd 
 - tool-ltchiptool @ 4.0.0+sha.788ba4e 
PLATFORM VERSIONS:
 - libretiny @ 1.0.1+sha.86924d8
 - ltchiptool @ 4.0.1
CUSTOM OPTIONS:
 - fw_name = esphome
 - fw_output = firmware.uf2
 - fw_version = 2023.4.0-dev
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- ESPAsyncWebServer-esphome @ 3.0.0
|-- DNSServer @ 1.1.0
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
|-- ArduinoJson @ 6.18.5
Compiling /data/smart-socket/.pioenvs/smart-socket/src/esphome/components/api/api_connection.cpp.o
Compiling /data/smart-socket/.pioenvs/smart-socket/src/esphome/components/api/api_frame_helper.cpp.o
Compiling /data/smart-socket/.pioenvs/smart-socket/src/esphome/components/api/api_pb2.cpp.o
Compiling /data/smart-socket/.pioenvs/smart-socket/src/esphome/components/api/api_pb2_service.cpp.o
In file included from src/esphome/components/api/proto.h:5,
                 from src/esphome/components/api/api_pb2.h:5,
                 from src/esphome/components/api/api_pb2_service.h:5,
                 from src/esphome/components/api/api_pb2_service.cpp:3:
src/esphome/core/helpers.h:543:3: error: 'SemaphoreHandle_t' does not name a type; did you mean 'xSemaphoreHandle'?
  543 |   SemaphoreHandle_t handle_;
      |   ^~~~~~~~~~~~~~~~~
      |   xSemaphoreHandle
In file included from src/esphome/components/api/proto.h:5,
                 from src/esphome/components/api/api_pb2.h:5,
                 from src/esphome/components/api/api_pb2.cpp:3:
src/esphome/core/helpers.h:543:3: error: 'SemaphoreHandle_t' does not name a type; did you mean 'xSemaphoreHandle'?
  543 |   SemaphoreHandle_t handle_;
      |   ^~~~~~~~~~~~~~~~~
      |   xSemaphoreHandle
In file included from src/esphome/components/api/proto.h:5,
                 from src/esphome/components/api/api_pb2.h:5,
                 from src/esphome/components/api/api_connection.h:4,
                 from src/esphome/components/api/api_connection.cpp:1:
src/esphome/core/helpers.h:543:3: error: 'SemaphoreHandle_t' does not name a type; did you mean 'xSemaphoreHandle'?
  543 |   SemaphoreHandle_t handle_;
      |   ^~~~~~~~~~~~~~~~~
      |   xSemaphoreHandle
In file included from src/esphome/components/api/api_frame_helper.cpp:5:
src/esphome/core/helpers.h:543:3: error: 'SemaphoreHandle_t' does not name a type; did you mean 'xSemaphoreHandle'?
  543 |   SemaphoreHandle_t handle_;
      |   ^~~~~~~~~~~~~~~~~
      |   xSemaphoreHandle
*** [/data/smart-socket/.pioenvs/smart-socket/src/esphome/components/api/api_pb2_service.cpp.o] Error 1
*** [/data/smart-socket/.pioenvs/smart-socket/src/esphome/components/api/api_frame_helper.cpp.o] Error 1
*** [/data/smart-socket/.pioenvs/smart-socket/src/esphome/components/api/api_pb2.cpp.o] Error 1
*** [/data/smart-socket/.pioenvs/smart-socket/src/esphome/components/api/api_connection.cpp.o] Error 1
========================= [FAILED] Took 12.74 seconds =========================

kuba2k2, 2023-05-02:
You need to update LibreTiny ESPHome as well.

rmandjes, 2023-05-02:
I've installed this add-on repository in Home Assistant:
https://github.com/libretiny-eu/esphome-hass-addon

I was expecting that the libretiny esphome version also was updated. Maybe that is not the case. How can I update ed LibreTiny ESPHome version in Home Assistant?

catalin2402, 2023-05-02:
The addon is not update yet, i have a PR opened for it, you can install it from my fork

rmandjes, 2023-05-02:
How must I do that from my homeassitant installation?

kuba2k2, 2023-05-02:
I believe the guide has been updated now:
https://github.com/libretiny-eu/esphome-hass-addon

closed 2023-05-03 by kuba2k2

rmandjes, 2023-05-03:
Thx. That was indeed solving the issue.