libretiny-eu/libretiny #119: Update Failed: ard=8,uf2=2, what does it mean?

GrumpyMeow, 2023-05-03:
Hi,

I've managed to flash a custom ESPHome firmware to one of my LSC smart ceiling lights. The initial flashing went as intended. The device is also connected with my WiFi. But i need to reflash the device with other pinouts as i'm unable to turn the lights on/off. But now when i try to flash my device via the webserver, it gives the error: `Update Failed: ard=8,uf2=2`. Unfortunatly the 6 times power-cycling also doesn´t provide the WiFi access-point which initially is used to flash the firmware. 

What does the error actually indicate?
```
Selected Device Slug: lsc-3007213-970787-ceiling-light
Selected Profile: oem-bk7231n-light-ty-1.2.16-sdk-2.3.1-40.00
Selected Firmware: /work/custom-firmware/overlooplamp_bk7231n_app.ota.ug.bin
```
Output of initial flash:
```
Using interface wlxd037451af94b with hwaddr d0:37:45:1a:f9:4b and ssid "cloudcutterflash"
wlxd037451af94b: interface state UNINITIALIZED->ENABLED
wlxd037451af94b: AP-ENABLED 
Using PSK v1 - Received PSK ID version 01
Processing endpoint /v2/url_config
Processing endpoint tuya.device.active
Processing endpoint tuya.device.dynamic.config.get
Processing endpoint tuya.device.uuid.pskkey.get
[MQTT Sending] Triggering firmware update message.
Processing endpoint tuya.device.upgrade.get
Processing endpoint tuya.device.upgrade.status.update
Processing endpoint /files/overlooplamp_bk7231n_app.ota.ug.bin
Firmware update progress: 28%
Firmware update progress: 30%
Processing endpoint atop.online.debug.log
Firmware update progress: 49%
Processing endpoint tuya.device.dynamic.config.get
Processing endpoint tuya.device.dynamic.config.ack
Firmware update progress: 60%
Firmware update progress: 87%
Firmware update progress: 90%
[Firmware Upload] /files/overlooplamp_bk7231n_app.ota.ug.bin send complete, request range: bytes=0-544639/544640
Processing endpoint tuya.device.upgrade.silent.get
Processing endpoint tuya.device.timer.count
Firmware update progress: 98%
Firmware file has been sent and MQTT reported a progress of nearly complete.  Waiting 15 seconds to ensure flashing completes.
Flashing should be complete.  It takes about 15 seconds for the device to reboot and verify the flash was valid.
Please wait about 30 seconds then look for signs of activity from the firmware you supplied (either watch for AP mode or check if it joined your network).
Device MAC address: 10:5a:17:7a:73:1a
```
ESPHome config:
```
libretuya:
  board: generic-bk7231n-qfn32-tuya
  framework:
    version: dev
```

kuba2k2, 2023-05-03:
The uf2 error 2 means that you're selecting the wrong firmware file to upload. You need to use the firmware.uf2 file, instead of the UG.bin file. Also, it seems that you're using an old version (LibreTuya) - please update to LibreTiny v1.0.0.

It's recommended to use the `esphome upload` command, which will automatically choose the right file to flash.

rakalex, 2023-05-03:
I have the same issue, I tried all firmwares, but no luck, same error. It happened when I uploaded the kickstart firmware It has 
kickstart-bk7231n AP, so I can navigate to 192.168.4.1, then select the WIFI, then it will open the Pinscan UI. But when I tries to upload the new firmware it shows that error.

kuba2k2, 2023-05-03:
@rakalex this is not the same issue, because the error code is different. Your error code 5 means that you're trying to flash old firmware. Please update LibreTiny to v1.0.0 and try again, it will work then.
LibreTuya v0.13.0 has been renamed to LibreTiny v1.0.0, and migrating backwards is not directly possible.

GrumpyMeow, 2023-05-03:
I updated the repository:
```
git fetch 
git pull
git status
On branch platform/libretuya
Your branch is up to date with 'origin/platform/libretuya'.
```
Still the error: 
```
root@LibreTuya-ESPHome:~/esphome/libretuya-esphome# esphome upload overlooplamp.yaml
INFO Reading configuration overlooplamp.yaml...
INFO Resolving IP address of overlooplamp.local
INFO  -> 192.168.178.106
INFO Uploading .esphome/build/overlooplamp/.pioenvs/overlooplamp/firmware.uf2 (2651648 bytes)
Uploading: [                                                            ] 0% 
ERROR Error sending data: [Errno 104] Connection reset by peer
ERROR Backend error code: 0x830b
```