libretiny-eu/libretiny #120: Pulse counter not working in ESPHome which block HLW8012

mobrembski, 2023-05-04:
Hi,

I found out that HLW8012 module is not working in libretiny. After correct configuration, all electricity related values remains 0.
I think i found what the issue is. Pulse counter (on which HLW8012 module is based on ) is by default attaching a interrupt at ANY EDGE.
Since Beken platform didnt support any edges, in libretiny we're forcing it to be falling edge:
`wiring_irq.c:

		case CHANGE:
			event	= GPIO_INT_LEVEL_FALLING;
			modeNew = INPUT_PULLUP;
			change	= true;
` 
And by default pulse counter in HLW module is using following configuration:
`pulse_counter_sensor.h:
  PulseCounterCountMode rising_edge_mode{PULSE_COUNTER_INCREMENT};
  PulseCounterCountMode falling_edge_mode{PULSE_COUNTER_DISABLE};
`
And in it's interrupt handler it is checking the state of pin which generates interrupt:
`pulse_counter_sensor.cpp:
  PulseCounterCountMode mode = arg->isr_pin.digital_read() ? arg->rising_edge_mode : arg->falling_edge_mode;
  switch (mode) {
    case PULSE_COUNTER_DISABLE:
      break;
    case PULSE_COUNTER_INCREMENT:
      arg->counter++;
      break;
    case PULSE_COUNTER_DECREMENT:
      arg->counter--;
      break;
`

Now if we force that CHANGE is falling, and by default pulse sensor reacts on rising edge only, this boolean operator returns falling_edge_mode due to the fact that during interrupt handler exectuion state of pin is low.
And for falling_edge_mode as we can see pulse counter is by default set up to PULSE_COUNTER_DISABLE.

If we change in libretiny CHANGE is rising, then state of pin during interrupt should be high, and boolean operator will return rising_edge_mode for which default state is PULSE_COUNTER_INCREMENT.

catalin2402, 2023-05-04:
You can try to use : `inverted: true` on the pins of HLW8012 configuration. I have a plug with HLW8012 that is using libretiny & libretiny-esphome (latest versions) and it is working as expected. We found that on some devices, the pins need to be inverted and the pulse counter started to work on these devices. On devices with BL0937 (that uses the same HLW8012 component) when inverting the pins, it is working as expected. On mine for example, with HLW8012 i didn't had to invert the pins

mobrembski, 2023-05-04:
Yes, i know it, but unfortunately it doesnt work for me.
Ive debugged it for a while and it seems that i'm not getting interrupts, this helps.

kuba2k2, 2023-07-13:
Hi, is this still an issue? I haven't had any problems with HLW8012/BL0937 on the current versions, and neither did I see anyone reporting issues.

muchtall, 2023-07-31:
I can confirm this is still an issue on 2023.7.0

I ran into this issue on a SMARTGREENER TGWF115PQM (https://www.elektroda.com/rtvforum/topic3951685.html) which uses a BK7231N and the BL0937. I banged my head on it way too long before realizing that the underlying pulse counter didn't work either. If I just use PIN_P6 (connected to CF1) as a binary sensor, I can see it toggling like crazy. But if I use it under pulse_counter, usually nothing. Occasionally I'll see an extremely low value like ~1 pulse/min but that's rare.

muchtall, 2023-07-31:
> You can try to use : `inverted: true` on the pins of HLW8012 configuration.

Ok. Now this makes more sense as to where this confusion arises. The ESPHome documentation for HLW8012 makes it clear that the SEL pin on the BL0937 can be inverted, however there's no mention of possible inversion of the CF and CF1 pins as well (https://esphome.io/components/sensor/hlw8012.html#sel-pin-inversion). Turns out those are inverted as well, at least on this product. When I invert the CF and CF1 connected pins, I start to see valid data come through.

Example YAML:
`sensor:
  - platform: hlw8012
    model: BL0937
    sel_pin:
      number: PIN_P24
      inverted: true
    cf_pin:
      number: PIN_P7
      inverted: true
    cf1_pin:
      number: PIN_P6
      inverted: true
    current_resistor: ${current_res}
    voltage_divider: ${voltage_div}
    energy:
      name: "${plug_name} Energy"
      unit_of_measurement: Wh
    current:
      name: "${plug_name} Amperage"
      unit_of_measurement: A
    voltage:
      name: "${plug_name} Voltage"
      unit_of_measurement: V
    power:
      name: "${plug_name} Wattage"
      unit_of_measurement: W
    change_mode_every: 8
    update_interval: 3s`
    
 With the above YAML, I am now seeing valid data come through. Now I just need to tweak the resistor values to match reality.
 
 @mobrembski , can you confirm this addresses the issue for you as well?

kuba2k2, 2023-08-02:
@mobrembski do I understand correctly that it works now without adding your changes?

mobrembski, 2023-08-02:
Yes, can be closed now

closed 2023-08-03 by kuba2k2