libretiny-eu/libretiny #121: BK7231N - UART Compile Error with libretiny

kaizday, 2023-05-04:
Hello,

First, thanks so much for great work.

I have a device that is currently running LibreTuya 2023.04, moving to LibreTiny, I am getting this error something related to UART. Could you please take a look?

Thanks

INFO Reading configuration /config/smart-air-sensors.yaml... INFO Generating C++ source... INFO Compiling app... Processing smart-air-sensors (board: cb3s; framework: arduino; platform: https://github.com/kuba2k2/libretiny.git) -------------------------------------------------------------------------------- HARDWARE: BK7231N 120MHz, 256KB RAM, 1.03MB Flash - framework-arduino-api @ 2022.8.24+sha.237b10a - framework-beken-bdk @ 0.0.0+v2021.06.07.sha.6491b8c - library-flashdb @ 1.2.0+sha.d5c892f - library-freertos @ 9.0.0+sha.95cc959 - library-freertos-port @ 2023.3.13+sha.bd96e82 - library-lwip @ 2.1.3-bdk+sha.4ee4d34 - library-printf @ 6.1.0+sha.28a79bd - tool-ltchiptool @ 4.0.0+sha.788ba4e PLATFORM VERSIONS: - libretiny @ 1.0.1+sha.86924d8 - ltchiptool @ 4.0.1 CUSTOM OPTIONS: - fw_name = esphome - fw_version = 2023.5.0-dev No dependencies Compiling .pioenvs/smart-air-sensors/src/esphome/components/uart/uart_component_libretiny.cpp.o Compiling .pioenvs/smart-air-sensors/src/esphome/core/scheduler.cpp.o Compiling .pioenvs/smart-air-sensors/src/esphome/core/string_ref.cpp.o Compiling .pioenvs/smart-air-sensors/src/esphome/core/util.cpp.o src/esphome/components/uart/uart_component_libretiny.cpp: In member function 'virtual void esphome::uart::LibreTinyUARTComponent::setup()': src/esphome/components/uart/uart_component_libretiny.cpp:72:85: error: 'PIN_SERIAL2_RX' was not declared in this scope; did you mean 'PIN_SERIAL2_TX'? 72 | else if ((tx_pin == -1 || tx_pin == PIN_SERIAL2_TX) && (rx_pin == -1 || rx_pin == PIN_SERIAL2_RX)) { | ^~~~~~~~~~~~~~ | PIN_SERIAL2_TX *** [.pioenvs/smart-air-sensors/src/esphome/components/uart/uart_component_libretiny.cpp.o] Error 1 Compiling .pioenvs/smart-air-sensors/src/main.cpp.o ========================= [FAILED] Took 21.97 seconds =========================

kuba2k2, 2023-05-04:
Can you post your config?

kaizday, 2023-05-04:
For the uart component, i tried both TX1/RX1 and the actual pins P11/P10. Really appreciate your work.

libretiny:
  board: cb3s
  framework:
    version: dev

# Enable logging
#logger:
#  baud_rate: 0

uart:
  rx_pin: RX1
  tx_pin: TX1
  baud_rate: 9600

# Register the Tuya MCU connection
tuya:

kuba2k2, 2023-05-04:
Try using `generic-bk7231n-qfn32-tuya` board instead; apparently CB3S doesn't have UART2 RX pin and it's messing with the pin macros...

kaizday, 2023-05-04:
Yessss! That works. Thanks so much!

closed 2023-05-04 by kaizday