libretiny-eu/libretiny #122: DHCP client sends wrong ARP packets

sciurius, 2023-05-05:
I have a number of bk7231t devices successfully flashed with libretiny.
Occasionally, when they try to set up the DHCP lease, wrong packets are sent. Instead of asking for an IP address, or reclaiming it's current address, the device effectively claims the IP address of the DHCP server.

From the system log:

    May  4 05:39:59 srv1 dhcpcd[449]: eth0: hardware address xx:xx:xx:xx:xx::ce claims 192.168.1.10

where 192.168.1.10 is the address of the DHCP server.

While this seems harmless, if a second package arrives at the DHCP server within 10 seconds it shuts down the network connection and, without DHCP server, the LAN slowly collapses.

The devices are GU10 lamps and TH06 sensors. All devices run the latest libretuya-esphome 

    v1.0.1+sha.8999cb9.dirty on generic-bk7231t-qfn32-tuya, compiled at May 4 2023 21:42:27, GCC 10.3.1 (-O1)

Nine out of ten devices have shown this behaviour in the last couple of weeks. Additional information (configs, ...) available on demand.

blfriedman, 2023-05-11:
I see the same issue with Gosund SW5 switches flashed with the latest libretiny-esphome, also bk7231t devices.

blfriedman, 2023-05-16:
Some more information which might help in identifying the location of the fault - I have noticed that for about five seconds after a restart/reboot, there is a sucessful DHCP reservation - as I watch it live on my DHCP server, I see the reservation for the active host, but after about five seconds, the reservation disappears.  My DHCP server is on my Mikrotik RB750Gr3 router, Rev r4, with 6.48.6 firmware.

voed, 2023-05-23:
Same issue on bk7231n, 1.0.2+sha.5c4da6e . The device connects once, but after reboot it just falling to the fallback hotspot. Added static IP to the config and it works fine.