libretiny-eu/libretiny #123: Where to share success stories?

GrumpyMeow, 2023-05-06:
I've managed to flash multiple devices which took quite some time to figure the correct configuration out. 
What would be the best place to share our success stories for each other?
I´d uploaded a post to elektroda, but the provided libretiny configuration was not appreciated..
Might the Wiki of this GitHub be a good place?

kuba2k2, 2023-05-06:
Interesting, I tried to find that post but I couldn't, did they remove it? 🤣

We have a #esphome-configs forum channel on Discord, you can share your stories there. There isn't any Wiki yet (nor am I planning to create one) but there are documentation pages. If you feel that something important is missing from the docs, or that you would describe some part of the process better, feel free to submit a PR to the docs.

kaizday, 2023-05-12:
I also have this same thought, I would wish someone shared the configs so I didnt have to lol.

Perhaps to have a page similar to devices.esphome.io for us to share device yaml configs?

kuba2k2, 2023-05-12:
As I mentioned, there's a channel on Discord made for this. You can post device info, photos, and configs there.

The Discord server is linked on the main page of the docs.

closed 2023-05-12 by GrumpyMeow