libretiny-eu/libretiny #127: Error: Unknown development platform 'libretiny'

rajil, 2023-05-14:
Hello,

I am trying to build using docker but it is failing.

docker was started using,
```
$ docker run  --rm -p 6052:6052 -v "${PWD}":/config -it ghcr.io/libretiny-eu/libretiny-esphome-docker:latest
```

Here is the log,
```
INFO Reading configuration /config/bedroom-fan.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing bedroom-fan (board: generic-bk7231n-qfn32-tuya; framework: arduino; platform: https://github.com/kuba2k2/libretiny.git)
--------------------------------------------------------------------------------
Installing/updating ltchiptool
Collecting ltchiptool<5.0,>=4.0.1
  Using cached ltchiptool-4.1.0-py3-none-any.whl (134 kB)
Collecting py-datastruct<0.4.0,>=0.3.0
  Using cached py_datastruct-0.3.0-py3-none-any.whl (24 kB)
Collecting semantic-version<3.0.0,>=2.10.0
  Using cached semantic_version-2.10.0-py2.py3-none-any.whl (15 kB)
Collecting click<9.0.0,>=8.1.3
  Using cached click-8.1.3-py3-none-any.whl (96 kB)
Collecting bk7231tools<2.0.0,>=1.3.5
  Using cached bk7231tools-1.3.5-py3-none-any.whl (28 kB)
Collecting importlib-metadata<5.0.0,>=4.12.0
  Using cached importlib_metadata-4.13.0-py3-none-any.whl (23 kB)
Collecting pycryptodomex<4.0.0,>=3.15.0
  Using cached pycryptodomex-3.17-cp35-abi3-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (2.1 MB)
Collecting colorama<0.5.0,>=0.4.5
  Using cached colorama-0.4.6-py2.py3-none-any.whl (25 kB)
Collecting xmodem<0.5.0,>=0.4.6
  Using cached xmodem-0.4.6-py3-none-any.whl
Collecting prettytable<4.0.0,>=3.3.0
  Using cached prettytable-3.7.0-py3-none-any.whl (27 kB)
Collecting pyserial<4.0,>=3.5
  Using cached pyserial-3.5-py2.py3-none-any.whl (90 kB)
Collecting zipp>=0.5
  Using cached zipp-3.15.0-py3-none-any.whl (6.8 kB)
Collecting wcwidth
  Using cached wcwidth-0.2.6-py2.py3-none-any.whl (29 kB)
Installing collected packages: zipp, wcwidth, pyserial, xmodem, semantic-version, pycryptodomex, py-datastruct, prettytable, importlib-metadata, colorama, click, bk7231tools, ltchiptool
  Attempting uninstall: zipp
    Found existing installation: zipp 3.15.0
    Uninstalling zipp-3.15.0:
      Successfully uninstalled zipp-3.15.0
  Attempting uninstall: wcwidth
    Found existing installation: wcwidth 0.2.6
    Uninstalling wcwidth-0.2.6:
      Successfully uninstalled wcwidth-0.2.6
  Attempting uninstall: semantic-version
    Found existing installation: semantic-version 2.10.0
    Uninstalling semantic-version-2.10.0:
ERROR: Could not install packages due to an EnvironmentError: [Errno 22] Invalid argument: '/usr/local/lib/python3.9/dist-packages/semantic_version-2.10.0.dist-info/'

!!! Installing ltchiptool failed, or version outdated. Please install ltchiptool manually using pip. Cannot continue. <member 'name' of 'ImportError' objects>: No module named 'ltchiptool'
Error: Unknown development platform 'libretiny'
```

kuba2k2, 2023-05-14:
This is odd. Try cleaning up the image and container:
```
docker stop <container name>
docker container prune -a
docker image prune -a
```
and then pulling the image again.

rajil, 2023-05-14:
Didnt help. Same error as before.

```
$ docker run  --rm -p 6052:6052 -v "${PWD}":/config -it ghcr.io/libretiny-eu/libretiny-esphome-docker:latest
Unable to find image 'ghcr.io/libretiny-eu/libretiny-esphome-docker:latest' locally
latest: Pulling from libretiny-eu/libretiny-esphome-docker
bb263680fed1: Pull complete 
55a99726d0af: Pull complete 
4f4fb700ef54: Pull complete 
176574bf4911: Pull complete 
4a4afc723d71: Pull complete 
bae5e13f5019: Pull complete 
ec7b96b091f3: Pull complete 
f3d78d88eece: Pull complete 
679b2adec694: Pull complete 
544e7b8ebe86: Pull complete 
Digest: sha256:a6f4172053fb8816b7eabbd67061b4994175e41d3dba9457caaa9ffaec6f9889
Status: Downloaded newer image for ghcr.io/libretiny-eu/libretiny-esphome-docker:latest
2023-05-14 15:16:14,857 INFO Starting dashboard web server on http://0.0.0.0:6052 and configuration dir /config...
2023-05-14 15:16:14,965 INFO 304 GET /devices (192.168.1.5) 2.12ms
2023-05-14 15:16:19,968 INFO 304 GET /devices (192.168.1.5) 1.23ms
2023-05-14 15:16:20,039 INFO 304 GET / (192.168.1.5) 7.30ms
2023-05-14 15:16:20,094 INFO 200 GET /static/fonts/material-icons/material-icons.css?hash=ead4c276 (192.168.1.5) 7.28ms
2023-05-14 15:16:20,096 INFO 200 GET /static/fonts/material-icons/MaterialIcons-Regular.woff2?hash=570eb838 (192.168.1.5) 1.71ms
2023-05-14 15:16:20,097 INFO 200 GET /static/css/esphome-2.css?hash=e81a30ae (192.168.1.5) 1.90ms
2023-05-14 15:16:20,100 INFO 200 GET /static/js/esphome/index-f2af38a5.js (192.168.1.5) 4.21ms
2023-05-14 15:16:20,136 INFO 200 GET /static/images/logo-text.svg (192.168.1.5) 1.25ms
2023-05-14 15:16:20,139 INFO 304 GET /devices (192.168.1.5) 1.68ms
2023-05-14 15:16:20,154 INFO 200 GET /static/js/esphome/c.ff44c2ea.js (192.168.1.5) 16.92ms
2023-05-14 15:16:20,343 INFO 200 GET /static/js/esphome/c.5cd5d307.js (192.168.1.5) 3.19ms
2023-05-14 15:16:20,344 INFO 200 GET /static/js/esphome/c.419250e6.js (192.168.1.5) 1.32ms
2023-05-14 15:16:23,981 INFO 200 GET /static/js/esphome/c.30a0677d.js (192.168.1.5) 1.11ms
2023-05-14 15:16:24,015 INFO 200 GET /static/js/esphome/c.63fcbfd6.js (192.168.1.5) 0.98ms
2023-05-14 15:16:24,017 INFO 200 GET /static/js/esphome/c.33591792.js (192.168.1.5) 1.41ms
2023-05-14 15:16:24,018 INFO 200 GET /static/js/esphome/c.32c39114.js (192.168.1.5) 1.11ms
2023-05-14 15:16:24,019 INFO 200 GET /static/js/esphome/c.f7445c67.js (192.168.1.5) 1.50ms
2023-05-14 15:16:24,048 INFO 304 GET /info?configuration=bedroom-fan.yaml (192.168.1.5) 1.05ms
2023-05-14 15:16:25,141 INFO 304 GET /devices (192.168.1.5) 1.48ms
2023-05-14 15:16:26,343 INFO 200 GET /static/js/esphome/c.fc1c6728.js (192.168.1.5) 1.47ms
2023-05-14 15:16:26,353 INFO 200 GET /static/js/esphome/c.13f7e761.js (192.168.1.5) 1.35ms
2023-05-14 15:16:26,366 INFO 101 GET /compile (192.168.1.5) 1.12ms
2023-05-14 15:16:26,369 INFO Running command 'esphome --dashboard compile /config/bedroom-fan.yaml'
2023-05-14 15:16:30,146 INFO 304 GET /devices (192.168.1.5) 1.73ms
2023-05-14 15:16:31,338 INFO Process exited with return code 1
2023-05-14 15:16:35,180 INFO 304 GET /devices (192.168.1.5) 1.29ms
2023-05-14 15:16:40,147 INFO 304 GET /devices (192.168.1.5) 1.67ms
2023-05-14 15:16:45,145 INFO 304 GET /devices (192.168.1.5) 1.29ms
2023-05-14 15:16:50,142 INFO 304 GET /devices (192.168.1.5) 1.51ms

```

rajil, 2023-05-14:
Logged in the container and install the packages manually. After that the compilation worked.

```
$ docker exec -it 01a0fd348624 sh
# pip install zipp wcwidth pyserial xmodem semantic-version pycryptodomex py-datastruct prettytable importlib-metadata colorama click bk7231tools ltchiptool
Requirement already satisfied: zipp in /usr/local/lib/python3.9/dist-packages (3.15.0)
Requirement already satisfied: wcwidth in /usr/local/lib/python3.9/dist-packages (0.2.6)
Collecting pyserial
  Using cached pyserial-3.5-py2.py3-none-any.whl (90 kB)
Collecting xmodem
  Using cached xmodem-0.4.6-py3-none-any.whl
Requirement already satisfied: semantic-version in /usr/local/lib/python3.9/dist-packages (2.10.0)
Collecting pycryptodomex
  Using cached pycryptodomex-3.17-cp35-abi3-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (2.1 MB)
Collecting py-datastruct
  Using cached py_datastruct-0.3.0-py3-none-any.whl (24 kB)
Collecting prettytable
  Using cached prettytable-3.7.0-py3-none-any.whl (27 kB)
Collecting importlib-metadata
  Downloading importlib_metadata-6.6.0-py3-none-any.whl (22 kB)
Requirement already satisfied: colorama in /usr/local/lib/python3.9/dist-packages (0.4.6)
Requirement already satisfied: click in /usr/local/lib/python3.9/dist-packages (8.1.3)
Collecting bk7231tools
  Using cached bk7231tools-1.3.5-py3-none-any.whl (28 kB)
Collecting ltchiptool
  Using cached ltchiptool-4.1.0-py3-none-any.whl (134 kB)
Collecting importlib-metadata
  Using cached importlib_metadata-4.13.0-py3-none-any.whl (23 kB)
Installing collected packages: pyserial, xmodem, pycryptodomex, py-datastruct, prettytable, importlib-metadata, bk7231tools, ltchiptool
Successfully installed bk7231tools-1.3.5 importlib-metadata-4.13.0 ltchiptool-4.1.0 prettytable-3.7.0 py-datastruct-0.3.0 pycryptodomex-3.17 pyserial-3.5 xmodem-0.4.6
```

closed 2023-07-13 by kuba2k2