libretiny-eu/libretiny #128: Libretiny-Esphome, Unexpected error (Windows +Python)

Brokly, 2023-05-15:
Hi !
Thanks for the cool addition to ESPHOME !
I am trying to compile a project under python in windows, but I get an unexpected error.
At the same time, everything works successfully in the libretuya-esphome  fork.
I have already fulfilled all the recommendations from the last lines of the log.
Maybe I 'm doing something wrong ? Thank you in advance for your help.

C:\Users\Brokly\esphome\libretiny-esphome>python -m esphome compile d:/config/testtiny.yaml
INFO Reading configuration d:/config/testtiny.yaml...
INFO Generating C++ source...
INFO Core config or version changed, cleaning build files...
INFO Compiling app...
Processing testtiny (board: cb2s; framework: arduino; platform: https://github.com/kuba2k2/libretiny.git)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Installing/updating ltchiptool
Error: Traceback (most recent call last):
  File "D:\Python\lib\site-packages\platformio\__main__.py", line 102, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "D:\Python\lib\site-packages\click\core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "D:\Python\lib\site-packages\click\core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "D:\Python\lib\site-packages\platformio\cli.py", line 70, in invoke
    return super().invoke(ctx)
  File "D:\Python\lib\site-packages\click\core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "D:\Python\lib\site-packages\click\core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "D:\Python\lib\site-packages\click\core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "D:\Python\lib\site-packages\click\decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "D:\Python\lib\site-packages\platformio\run\cli.py", line 145, in cli
    process_env(
  File "D:\Python\lib\site-packages\platformio\run\cli.py", line 208, in process_env
    ).process()
  File "D:\Python\lib\site-packages\platformio\run\processor.py", line 81, in process
    install_project_env_dependencies(
  File "D:\Python\lib\site-packages\platformio\package\commands\install.py", line 131, in install_project_env_dependencies
    _install_project_env_platform(project_env, options),
  File "D:\Python\lib\site-packages\platformio\package\commands\install.py", line 148, in _install_project_env_platform
    PlatformPackageManager().install(
  File "D:\Python\lib\site-packages\platformio\package\manager\platform.py", line 60, in install
    p.configure_project_packages(project_env, project_targets)
  File "D:\Python\lib\site-packages\platformio\platform\base.py", line 183, in configure_project_packages
    self.configure_default_packages(options, targets or [])
  File "C:\Users\Brokly\.platformio\platforms\libretiny@src-3806c7394b736d894a622c447e992805\platform.py", line 178, in configure_default_packages
    board = self.get_boards(options.get("board"))
  File "C:\Users\Brokly\.platformio\platforms\libretiny@src-3806c7394b736d894a622c447e992805\platform.py", line 286, in get_boards
    return self.update_board(result)
  File "C:\Users\Brokly\.platformio\platforms\libretiny@src-3806c7394b736d894a622c447e992805\platform.py", line 305, in update_board
    family = ltchiptool.Family.get(short_name=family)
  File "D:\Python\lib\site-packages\ltchiptool\models\family.py", line 61, in get
    for family in cls.get_all():
  File "D:\Python\lib\site-packages\ltchiptool\models\family.py", line 39, in get_all
    LT_FAMILIES = [cls(f) for f in lt_read_json("families.json")]
  File "D:\Python\lib\site-packages\ltchiptool\models\family.py", line 39, in <listcomp>
    LT_FAMILIES = [cls(f) for f in lt_read_json("families.json")]
  File "D:\Python\lib\site-packages\ltchiptool\models\family.py", line 28, in __init__
    for key, value in data.items():
AttributeError: 'str' object has no attribute 'items'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq/index.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

kuba2k2, 2023-05-15:
It seems that you have multiple versions of LibreTiny/LibreTuya installed. Uninstall all versions, by deleting directories in `C:\Users\Brokly\.platformio\platforms\` (all dirs related to `libretuya` or `libretiny`). Then try compiling again.

Brokly, 2023-05-15:
You're right, there really were several copies. Unfortunately, deleting them didn't help :(

C:\Users\Brokly\esphome\libretiny-esphome>python -m esphome compile d:/config/testtiny.yaml
INFO Reading configuration d:/config/testtiny.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing testtiny (board: cb2s; framework: arduino; platform: https://github.com/kuba2k2/libretiny.git)**

Platform Manager: Installing git+https://github.com/kuba2k2/libretiny.git
INFO Installing git+https://github.com/kuba2k2/libretiny.git
git version 2.40.1.windows.1
Cloning into 'C:\Users\Brokly\.platformio\.cache\tmp\pkg-installing-lmfjqiyu'...
remote: Enumerating objects: 600, done.
remote: Counting objects: 100% (600/600), done.
remote: Compressing objects: 100% (441/441), done.
remote: Total 600 (delta 137), reused 448 (delta 127), pack-reused 0Receiving objects:  99% (594/600)
Receiving objects: 100% (600/600), 367.18 KiB | 1.56 MiB/s, done.
Resolving deltas: 100% (137/137), done.
Platform Manager: libretiny@1.0.2+sha.f7c28ee has been installed!
[32mINFO [32mlibretiny@1.0.2+sha.f7c28ee has been installed![0m[0m
Installing/updating ltchiptool
Error: Traceback (most recent call last):
  File "D:\Python\lib\site-packages\platformio\__main__.py", line 102, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "D:\Python\lib\site-packages\click\core.py", line 1130, in __call__
..........

kuba2k2, 2023-05-15:
Well, I'll need to see the entire error message again. The most important part is actually at the end, not at the start.

Brokly, 2023-05-16:
Of course....

C:\Users\Brokly\esphome\libretiny-esphome>python -m esphome compile d:/config/testtiny.yaml
INFO Reading configuration d:/config/testtiny.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing testtiny (board: cb2s; framework: arduino; platform: https://github.com/kuba2k2/libretiny.git)

Platform Manager: Installing git+https://github.com/kuba2k2/libretiny.git
INFO Installing git+https://github.com/kuba2k2/libretiny.git
git version 2.40.1.windows.1
Cloning into 'C:\Users\Brokly\.platformio\.cache\tmp\pkg-installing-lmfjqiyu'...
remote: Enumerating objects: 600, done.
remote: Counting objects: 100% (600/600), done.
remote: Compressing objects: 100% (441/441), done.
remote: Total 600 (delta 137), reused 448 (delta 127), pack-reused 0Receiving objects:  99% (594/600)
Receiving objects: 100% (600/600), 367.18 KiB | 1.56 MiB/s, done.
Resolving deltas: 100% (137/137), done.
Platform Manager: libretiny@1.0.2+sha.f7c28ee has been installed!
[32mINFO [32mlibretiny@1.0.2+sha.f7c28ee has been installed![0m[0m
Installing/updating ltchiptool
Error: Traceback (most recent call last):
  File "D:\Python\lib\site-packages\platformio\__main__.py", line 102, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "D:\Python\lib\site-packages\click\core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "D:\Python\lib\site-packages\click\core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "D:\Python\lib\site-packages\platformio\cli.py", line 70, in invoke
    return super().invoke(ctx)
  File "D:\Python\lib\site-packages\click\core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "D:\Python\lib\site-packages\click\core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "D:\Python\lib\site-packages\click\core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "D:\Python\lib\site-packages\click\decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "D:\Python\lib\site-packages\platformio\run\cli.py", line 145, in cli
    process_env(
  File "D:\Python\lib\site-packages\platformio\run\cli.py", line 208, in process_env
    ).process()
  File "D:\Python\lib\site-packages\platformio\run\processor.py", line 81, in process
    install_project_env_dependencies(
  File "D:\Python\lib\site-packages\platformio\package\commands\install.py", line 131, in install_project_env_dependencies
    _install_project_env_platform(project_env, options),
  File "D:\Python\lib\site-packages\platformio\package\commands\install.py", line 148, in _install_project_env_platform
    PlatformPackageManager().install(
  File "D:\Python\lib\site-packages\platformio\package\manager\platform.py", line 60, in install
    p.configure_project_packages(project_env, project_targets)
  File "D:\Python\lib\site-packages\platformio\platform\base.py", line 183, in configure_project_packages
    self.configure_default_packages(options, targets or [])
  File "C:\Users\Brokly\.platformio\platforms\libretiny\platform.py", line 178, in configure_default_packages
    board = self.get_boards(options.get("board"))
  File "C:\Users\Brokly\.platformio\platforms\libretiny\platform.py", line 286, in get_boards
    return self.update_board(result)
  File "C:\Users\Brokly\.platformio\platforms\libretiny\platform.py", line 305, in update_board
    family = ltchiptool.Family.get(short_name=family)
  File "D:\Python\lib\site-packages\ltchiptool\models\family.py", line 61, in get
    for family in cls.get_all():
  File "D:\Python\lib\site-packages\ltchiptool\models\family.py", line 39, in get_all
    LT_FAMILIES = [cls(f) for f in lt_read_json("families.json")]
  File "D:\Python\lib\site-packages\ltchiptool\models\family.py", line 39, in <listcomp>
    LT_FAMILIES = [cls(f) for f in lt_read_json("families.json")]
  File "D:\Python\lib\site-packages\ltchiptool\models\family.py", line 28, in __init__
    for key, value in data.items():
AttributeError: 'str' object has no attribute 'items'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq/index.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

C:\Users\Brokly\esphome\libretiny-esphome>

kuba2k2, 2023-05-16:
This is weird. Can you open a CMD.exe in the D:\Python directory and run:
python -m pip install --upgrade ltchiptool
and post the output of that?

Brokly, 2023-05-16:
Yes.

D:\Python>python -m pip install --upgrade ltchiptool
WARNING: Ignoring invalid distribution -latformio (d:\python\lib\site-packages)
Requirement already satisfied: ltchiptool in d:\python\lib\site-packages (3.0.2)
Collecting ltchiptool
  Downloading ltchiptool-4.1.1-py3-none-any.whl (134 kB)
     ---------------------------------------- 134.6/134.6 kB 1.1 MB/s eta 0:00:00
Collecting bk7231tools<2.0.0,>=1.3.6 (from ltchiptool)
  Downloading bk7231tools-1.3.6-py3-none-any.whl (28 kB)
Requirement already satisfied: click<9.0.0,>=8.1.3 in d:\python\lib\site-packages (from ltchiptool) (8.1.3)
Requirement already satisfied: colorama<0.5.0,>=0.4.5 in d:\python\lib\site-packages (from ltchiptool) (0.4.6)
Requirement already satisfied: importlib-metadata<5.0.0,>=4.12.0 in d:\python\lib\site-packages (from ltchiptool) (4.13.0)
Requirement already satisfied: prettytable<4.0.0,>=3.3.0 in d:\python\lib\site-packages (from ltchiptool) (3.6.0)
Collecting py-datastruct<0.4.0,>=0.3.0 (from ltchiptool)
  Downloading py_datastruct-0.3.0-py3-none-any.whl (24 kB)
Requirement already satisfied: pycryptodomex<4.0.0,>=3.15.0 in d:\python\lib\site-packages (from ltchiptool) (3.17)
Requirement already satisfied: semantic-version<3.0.0,>=2.10.0 in d:\python\lib\site-packages (from ltchiptool) (2.10.0)
Requirement already satisfied: xmodem<0.5.0,>=0.4.6 in d:\python\lib\site-packages (from ltchiptool) (0.4.6)
Requirement already satisfied: pyserial<4.0,>=3.5 in d:\python\lib\site-packages (from bk7231tools<2.0.0,>=1.3.6->ltchiptool) (3.5)
Requirement already satisfied: zipp>=0.5 in d:\python\lib\site-packages (from importlib-metadata<5.0.0,>=4.12.0->ltchiptool) (3.15.0)
Requirement already satisfied: wcwidth in d:\python\lib\site-packages (from prettytable<4.0.0,>=3.3.0->ltchiptool) (0.2.6)
WARNING: Ignoring invalid distribution -latformio (d:\python\lib\site-packages)
Installing collected packages: py-datastruct, bk7231tools, ltchiptool
  Attempting uninstall: bk7231tools
    Found existing installation: bk7231tools 1.3.1
    Uninstalling bk7231tools-1.3.1:
      Successfully uninstalled bk7231tools-1.3.1
  Attempting uninstall: ltchiptool
    Found existing installation: ltchiptool 3.0.2
    Uninstalling ltchiptool-3.0.2:
      Successfully uninstalled ltchiptool-3.0.2
Successfully installed bk7231tools-1.3.6 ltchiptool-4.1.1 py-datastruct-0.3.0

Brokly, 2023-05-16:
It helped ! Thank you so much for your help and for a great project !

closed 2023-06-10 by Brokly