libretiny-eu/libretiny #131: mDNS does not work

rajil, 2023-05-18:
I have regular esphome and libretiny in the same container. The regular esphome is able to see all the devices via mDNS but libretiny does not. 

`tcpdump -i eth0 -n ip multicast` shows all the regular esphome devices but no libretiny devices ( i have several flashed).

Is there something missing in libretiny for mDNS?

Thanks

closed 2023-05-18 by rajil

kuba2k2, 2023-05-18:
I see the issue was closed, did it start working? I personally didn't have any issues with mDNS on the current version.

rajil, 2023-05-19:
I have both esphome and libretiny on the same machine. That means both want to use port 6052 at the same for the dashboard. I tried changing libretiny port by using the following,
`--net=host -p 6053:6052`
but docker did not accept it. Eventually, i had to forego mdns and used `-p 6053:6052 -e ESPHOME_DASHBOARD_USE_PING=true` with libretiny.