libretiny-eu/libretiny #132: Is there a problem with efuse ?

Brokly, 2023-05-27:
Hi ! I stumbled upon a problem and hope to find help here. I have a CB2L device. After the firmware in libretiny-esphome, it does not start. Restoring from backup brings the device back to its previous working state. Unfortunately, this device is not Tuya. Could the wrong data in efuse be the cause?

kuba2k2, 2023-05-27:
CB2L is a Tuya-specific module, so what cloud is this device connected to?

What firmware are you trying to flash? Can you post the esphome config?

What flashing tools are you using and how?

Brokly, 2023-05-27:
Yes, this manufacturer used to produce devices compatible with Tuya, but now it has switched to its own cloud, and its devices have stopped connecting to the Tuya cloud. The log shows that he is using a completely different software. I made a full backup using BK7231Flasher.exe . I restore the firmware using BK7231Flasher.exe or bk_writer.exe . I tried to flash the device with the firmware received from a working device with the same chip, but the result is the same: the device does not start. For compilation and firmware, I use the libretiny-esphome version for python under Windows. This is not the first device that I have redesigned, more than a dozen are working successfully. To give more information, I will show a fragment of the log with the factory firmware. This is clearly not Tuya :(

BK7231n_1.0.8
CPSR:0x000000D3
R0:0x00800000
R1:0x00000000
R2:0x005AA000
R3:0x00000006
R4:0x00400001
R13:0x00401C1C
R14(LR):0x000033AC
ST:0x00000000
[32;22m[I/FAL] Fal(V0.4.0)success[0m
[36;22m[I/OTA] RT-Thread OTA package(V0.2.4) initialize success.[0m

go os_addr(0x10000)..........
[0][ERR][DEVINFO]app partition signature not found
[480][ERR][STORAGE_PW]failed to read key:WIFI_SSID
[480][ERR][STORAGE_PW]failed to read key:WIFI_KEY
[10480][ERR][TELEMETRY]bk_wlan_get_link_status faied, err: -1
[10482][ERR][TELEMETRY]bk_wlan_get_link_status faied, err: -1
[10484][ERR][MQTT_SERVICE_PAHO]Cannot publish message to the topic:  (err -1)
[10486][ERR][TELEMETRY]Cannot send DeviceTelemetry

kuba2k2, 2023-05-27:
[ltchiptool](https://docs.libretiny.eu/docs/flashing/tools/ltchiptool/) is the recommended software for flashing LibreTiny firmware. It can also dump flash and restore stock firmware from dump file.

BK7231Flasher should work too, but using bk_writer is strongly discouraged, because of its instability and CRC checking problems. Also, these tools can't check for file type, so it can also be an issue that you're flashing the wrong file. For example, if you flash LibreTiny's `firmware.uf2` or `firmware.bin` with BKFlasher, it just won't boot up.

It seems like your device has an unusual bootloader version (1.0.8, while all other devices have 1.0.1). This can cause booting problems, but I can't tell for sure since you haven't posted logs of failed bootup.

Here's a copy of the 1.0.1 bootloader:
[bk7231n-1.0.1.zip](https://github.com/kuba2k2/libretiny/files/11582176/bk7231n-1.0.1.zip)

Download it, unzip the file, and put it into ltchiptool. Set the options like this:
[image]

Then, connect the device as usual, and press Start to begin the flashing process.

After doing that, flash `firmware.uf2` with the same program.

Brokly, 2023-05-27:
Unfortunately, it is not possible to write the loader.
[image]
But if I flash the dump read from the working module (Tuya), then the bootloader is flashed and the firmware is flashed, but the device remains dead.

kuba2k2, 2023-05-27:
Hm, maybe it can't flash because it's not aligned to block boundary... Try this one instead:
[boot_1.0.1.zip](https://github.com/kuba2k2/libretiny/files/11582288/boot_1.0.1.zip)

Even if the error shows up, the flashing should be successful. Check if the chip boots then. Post UART logs after flashing the bootloader (even with an error) with LibreTiny firmware.

Brokly, 2023-05-27:
Unfortunately, there is nothing in the UART2 logs. The checksum mismatch occurs not only when trying to flash the bootloader, but also in the next firmware firmware.uf2 . The device remains dead. Here are the ltchiptool logs :+1: 

D: Detection: Detection(name='D:\\download\\boot_1.0.1.bin', size=69632, type=<Type.UNRECOGNIZED: 'Unrecognized'>, file_type=None, offset=0, skip=0, length=69632, family=None, soc=None)
D: Started FlashThread
D: Starting WRITE operation; file = D:\download\boot_1.0.1.bin, port = COM4 @ Auto
D: Connecting
D: Starting file position: 0 / 0x0 / 0 B
D: Trying to unprotect flash memory...
D: Erasing and writing at 0x200000 (0.00%)
...
D: Erasing and writing at 0x210000 (94.12%)
D: Verifying CRC
E: ValueError: Chip CRC value 516A29EE does not match calculated CRC value B399F8A6
E: |-- File "ltchiptool\gui\work\flash.py", line 63, in run_impl
E: |-- File "ltchiptool\gui\work\flash.py", line 151, in _do_write
E: |-- File "ltchiptool\soc\bk72xx\flash.py", line 175, in flash_write_raw
E: |-- File "ltchiptool\util\flash.py", line 44, in update_from
E: |-- File "bk7231tools\serial\__init__.py", line 136, in program_flash
D: Stopped FlashThread
D: Detection: Detection(name='D:\\config\\.esphome\\build\\lamp-rgbww-sbe27-7\\.pioenvs\\lamp-rgbww-sbe27-7\\firmware.uf2', size=3075584, type=<Type.VALID_UF2: 'UF2'>, file_type='UF2 - esphome 2023.5.0-dev', offset=0, skip=0, length=3075584, family=<Family: BK7231N(0x7B3EF230), name=beken-7231n, parent=beken-72xx-gen1>, soc=None)
D: Started FlashThread
D: Starting WRITE operation; file = D:\config\.esphome\build\lamp-rgbww-sbe27-7\.pioenvs\lamp-rgbww-sbe27-7\firmware.uf2, port = COM4 @ Auto
D: Connecting
D: Trying to unprotect flash memory...
D: Erasing and writing at 0x211000 (0.00%)
...
D: Erasing and writing at 0x2F8000 (99.47%)
D: Erasing and writing at 0x2F9000 (99.90%)
D: Verifying CRC
E: ValueError: Chip CRC value 30633265 does not match calculated CRC value 6112AD94
E: |-- File "ltchiptool\gui\work\flash.py", line 63, in run_impl
E: |-- File "ltchiptool\gui\work\flash.py", line 119, in _do_write
E: |-- File "ltchiptool\soc\bk72xx\flash.py", line 203, in flash_write_uf2
E: |-- File "ltchiptool\util\flash.py", line 44, in update_from
E: |-- File "bk7231tools\serial\__init__.py", line 136, in program_flash
D: Stopped FlashThread

kuba2k2, 2023-05-27:
Is this the only device having such strange issues?
My guess is either;
- the program fails to unprotect the flash, and can't write any data
- we're dealing with a different version of BK7231N, and the CRC algorithm is different.

If you've used BKFlasher successfully, can you try writing the bootloader using that program? Offset 0x0, the entire bootloader file after unzipping. Then, check UART2 output for any logs, and post them here.

Brokly, 2023-05-27:
BKFlasher controls the size of the firmware and does not allow you to fill only the bootloader, the size is too small. My only hope is for efuse, but I don't understand how to read or write them. It looks like bk_writer only claims to be able to write them, in fact it doesn't.

kuba2k2, 2023-05-27:
I don't even know if efuses are on BK7231, but I'm 99% sure that's not the problem here.

If BKWriter doesn't allow smaller files, upload a full dump. For example, from here:
https://github.com/tuya-cloudcutter/tuya-cloudcutter/issues/400

Brokly, 2023-05-27:
I saw that there is a file 15wRGBCWWifiDownlight-rom.bin in the archive there. Apparently this is a ROM firmware file. But how to use it ?
[image]

And yes, there is a mention of efuse in the technical documentation. And the data contained in it is of interest.
[image]

kuba2k2, 2023-05-27:
You don't need the ROM file, just flash the firmware dump file `15wRGBCWWifiDownlight.bin`.
From what I understand, reading EFuse needs an SPI programmer, and most likely some proprietary software and commands. 
If the EFuse would be set to disable flash writing, I think the device would still have its original firmware running. The fact that it either works or not, depending on what file you flash, indicates that it's not write-protected.

Brokly, 2023-05-27:
Unfortunately, chips with N do not have the ability to SPI firmware. Firmware file 15wRGBCWWifiDownlight.bin did not bring any result. :(Apparently I ran into some kind of protection, which I don't have enough knowledge to break :(

kuba2k2, 2023-05-27:
Flash the dump file, using BKFlasher (as that seems to be the only tool that works for you). Then read it back (using any tool that works). Then zip the file, and post it here.

Who said that N chips don't have SPI flashing possibility? The datasheet states: `The 32-byte eFUSE could be written through download port GPIO20 ~ GPIO23 at download mode`. These GPIOs are actually SPI flashing GPIOs (shared with JTAG afaik).

Brokly, 2023-05-27:
[BK7231n_1_0_8_trouble_reflash.zip](https://github.com/kuba2k2/libretiny/files/11582828/BK7231n_1_0_8_trouble_reflash.zip)

When I wrote about SPI, I meant the CB2L module. It is very difficult to solder to the qfn contacts directly :)

kuba2k2, 2023-05-27:
> This is the source dump of the device that cannot be flashed. Or did I misunderstand ?

Yes, I think you did - I meant for you to flash the firmware from issue #400 (linked above), and then dump it back, right after flashing. That way we can check if it was written successfully.

Brokly, 2023-05-27:
Yes, I did it.

[readResult_BK7231N_QIO_back_read_2023-27-5--22-20-34.zip](https://github.com/kuba2k2/libretiny/files/11582891/readResult_BK7231N_QIO_back_read_2023-27-5--22-20-34.zip)

kuba2k2, 2023-05-27:
Writing works. I've found the cause.

After running your original dump through bk7231tools, there's no usable data in the output - everything looks pretty much random. It seems like the manufacturer **changed the encryption keys** of the firmware. That's why the it doesn't boot up with any other firmware.

It seems like the only option here would be to actually use the EFuse, and overwrite the manufacturer's encryption keys. But I'm not sure how to do that yet.

Another option would be to use JTAG, to read out the keys from bootloader (unless they've disabled JTAG too).

kuba2k2, 2023-05-27:
Actually... I was able to extract the keys, I think. I'll keep you posted about the progress here.

Brokly, 2023-05-28:
Super, your help is always very effective! I will be looking forward to the news! I could try reading or writing keys. I have a JTAG programmer. But I couldn't find any methods. It would probably be correct to write down the old Tuya key, but I don't know it. This is very similar to the fact that bk_writer can write keys via UART.

bk_writer's help is very crooked. Apparently Chinese is used there.

D:\download>bk_writer --help
bkwriter 1.2.6.5: build @ Dec  8 2021 16:59:59
# bk_writer command line

```
bk_writer -BBK7231S_1.0.3 -tBK7231 -pCOM7@1500000 -fbk7231s_common_1.0.1_uart_1.00.bin@00011000 -e2 -R --getbus-timeout=15 --retry-cnt=20 -efuse-key=0x510fb093,0xa3cbeadc,0x5993a17e,0xc7adeb03 --secureboot=1 --publicKey=0xbce4f9bc,0x76979838,0xedc80be0,0xea050f59,0x7e382b92,0x290c2c2b,0xc02250b6,0x249e2865,0x2543b188,0xf12a2d6c,0x9f40377e,0x24a9d5d3,0xf8e3fc88,0x17311e85,0x4c146c24,0x995af150 --enable-Sec-boot=1 --deviceID=0x21650820 --publicKey-file=efuse_7271.txt

-B ?????,???boot?????????,???, BK7231N???
-t BK7231/BK7231N ???????BK7231??BK7231N
-p ??COM?????
-f ??????????
-R ??Reset
-e2 ???
--getbus-timeout Host?Target????
--retry-cnt ??????
--efuse-key ??efuse key, 4?32bit16???, ??BK7231N??
--efuse-file ??efuse file
--secureboot SecureBoot????
--enable-Sec-boot ????secureboot ???,??????
--publicKey secureboot ?????
--publicKey-file ??????
--deviceID ????DeviceID (???BK7271)
```

kuba2k2, 2023-05-28:
I don't think that the UART protocol allows to write keys. Beken has their own dedicated SPI programmer, so maybe bk_writer has to be used with that to write the keys.

Also, you can't read keys from efuse. What I did was to decrypt the bootloader image, using known-plaintext parts. The decrypted image contains the keys, which gave me the possibility to decrypt everything else.

LibreTiny allows to change the keys that are compiled with the firmware. I'll let you know how to do that, ans you'll be able to run ESPHome firmware without having to change keys.

kuba2k2, 2023-05-28:
Hi

1. Flash the stock firmware (to bring back the original bootloader)
2. Verify that it boots
3. Add this to your ESPHome YAML:
```yaml
esphome:
  name: my-device
  platformio_options:
    board_build.bkcrypt_coeffs: 79bffed7a3fbeafd5dd3abffdfbfff5b
    board_build.bkota.key: BDB4CE110F4787C2F539BF50E30A14E0
    board_build.bkota.iv: 46DE464946314329
```
4. Compile ESPHome with this config, as usual.
5. Flash `firmware.uf2` (if ltchiptool works for you). Otherwise flash `image_bk7231n_app.0x011000.rbl` (with BKFlasher) to offset 0x11000.
6. Optionally, dump the flash again and post it here (so that we can verify if the firmware is flashed correctly).
7. After doing that, ESPHome should work. Remember not to remove this `platformio_options:` block!

Brokly, 2023-05-28:
Unfortunately, ltchiptools terminates the firmware with a checksum error. bk_flasher does not allow you to set the program address and does not support *.rbl files and does not allow to set the offset. For this reason, I cannot act on this plan:(

kuba2k2, 2023-05-28:
Right. I haven't used BKFlasher, so I thought it allows for more than just flashing OpenBeken, apparently it doesn't.

In this case, we have to find out why ltchiptool is failing to calculate the checksum:
1. Flash the [1.0.1 bootloader](https://github.com/kuba2k2/libretiny/files/11582288/boot_1.0.1.zip) again - using ltchiptool. It will show a CRC error - copy & paste the error log here, so that I can see the exact CRC values it calculated.
2. During flashing, there's "Chip info" written on the "Start" button. That information can be helpful too.
3. After flashing bootloader, dump the flash again, using BKFlasher.
4. Read the flash with ltchiptool as well.

Post here: 
- both flash dumps after writing bootloader (one from BKFlasher, second from ltchiptool)
- chip info from ltchiptool
- CRC errors and logs from ltchiptool

Hopefully that will help me solve the problem.

Brokly, 2023-05-28:
Logs ltchiptool:
D: Started FlashThread
D: Starting WRITE operation; file = D:\download\boot_1.0.1.bin, port = COM4 @ Auto
D: Connecting
D: Connecting
D: Connecting
D: Connecting
D: Connecting
D: Connecting
D: Connecting
D: Connecting
D: Connecting
D: Connecting
I: Connect UART1 of the BK7231 to the USB-TTL adapter:
I: 
I:     --------+        +------------------------
I:          PC |        | BK7231                 
I:     --------+        +------------------------
I:          RX | ------ | TX1 (GPIO11 / P11)     
I:          TX | ------ | RX1 (GPIO10 / P10)     
I:         RTS | ------ | CEN (or RST, optional) 
I:             |        |                        
I:         GND | ------ | GND                    
I:     --------+        +------------------------
I:  
I: Make sure to use a good 3.3V power supply, otherwise the adapter might
I: lose power during chip reset. Usually, the adapter's power regulator
I: is not enough and an external power supply is needed (like AMS1117).
I:  
I: If you didn't connect RTS to CEN, after running the command you have
I: around 20 seconds to reset the chip manually. In order to do that,
I: you need to bridge CEN to GND with a wire.
D: Connecting
D: Connecting
D: Connecting
D: Connecting
D: Connecting
D: Connecting
D: Connecting
D: Starting file position: 0 / 0x0 / 0 B
D: Trying to unprotect flash memory...
D: Erasing and writing at 0x200000 (0.00%)
D: Erasing and writing at 0x201000 (5.88%)
D: Erasing and writing at 0x202000 (11.76%)
D: Erasing and writing at 0x203000 (17.65%)
D: Erasing and writing at 0x204000 (23.53%)
D: Erasing and writing at 0x205000 (29.41%)
D: Erasing and writing at 0x206000 (35.29%)
D: Erasing and writing at 0x207000 (41.18%)
D: Erasing and writing at 0x208000 (47.06%)
D: Erasing and writing at 0x209000 (52.94%)
D: Erasing and writing at 0x20A000 (58.82%)
D: Erasing and writing at 0x20B000 (64.71%)
D: Erasing and writing at 0x20C000 (70.59%)
D: Erasing and writing at 0x20D000 (76.47%)
D: Erasing and writing at 0x20E000 (82.35%)
D: Erasing and writing at 0x20F000 (88.24%)
D: Erasing and writing at 0x210000 (94.12%)
D: Verifying CRC
E: ValueError: Chip CRC value 516A29EE does not match calculated CRC value B399F8A6
E: |-- File "ltchiptool\gui\work\flash.py", line 63, in run_impl
E: |-- File "ltchiptool\gui\work\flash.py", line 151, in _do_write
E: |-- File "ltchiptool\soc\bk72xx\flash.py", line 175, in flash_write_raw
E: |-- File "ltchiptool\util\flash.py", line 44, in update_from
E: |-- File "bk7231tools\serial\__init__.py", line 136, in program_flash
D: Stopped FlashThread

[image]

read bkflasher:
[bkflasher_after_flash_loader.zip](https://github.com/kuba2k2/libretiny/files/11584514/bkflasher_after_flash_loader.zip)

rom: 
[ltchiptool_rom.zip](https://github.com/kuba2k2/libretiny/files/11584485/ltchiptool_rom.zip)

flash:
[ltchiptool_flash.zip](https://github.com/kuba2k2/libretiny/files/11584531/ltchiptool_flash.zip)

kuba2k2, 2023-05-28:
The ROM is standard. Probably the flash model is not supported, these kind of issues are very hard to debug.

You can try with BkWriter, if it works for you:
https://images.tuyacn.com/smart/bk_writer1.60/bk_writer1.60.exe
You can put the RBL file in it, and set the offset to 0x11000. Just make sure to update your esphome config and get the correct file, as per this comment:
https://github.com/kuba2k2/libretiny/issues/132#issuecomment-1566066632

Brokly, 2023-05-28:
Unfortunately, BKWriter also ends up with a checksum error. Apparently the problem is not simple and it is impossible to solve it yet. In any case , thank you very much for your help !

kuba2k2, 2023-05-28:
Of course it's possible to solve, we're very close in fact :)

Yes, BkWriter can't calculate CRC properly, but the firmware should be flashed nevertheless. If you flash your stock firmware bootloader, along with LibreTiny ESPHome firmware compiled with the special keys, it should work. If not, there should at least be something on the UART logs.

Brokly, 2023-05-29:
I tried to flash in a full dump from another device from Tuya, with a similar chip. This can be done using BK7231Flasher.exe . After that, the rest of the programs are flashed without complaints about the checksum. But the device does not wake up :(

kuba2k2, 2023-05-29:
Again, your chip has different encryption keys - even if flashing dump from another device will work, it will **not boot up**. That's perfectly normal.

Try to do what I suggested in the previous message. Make sure that the correct 1.0.8 bootloader is on the chip, and flash the compiled esphome code then.

Brokly, 2023-05-29:
In this case, there is a problem with the checksum. And as a result, the firmware remains the same. Because the recording is not completed :(I do not know a tool that would burn from 0x11000 address and successfully complete the operation.

kuba2k2, 2023-05-29:
Did you check that the firmware remains the same? BkWriter can write, but can't calculate CRC. That is a known issue, even with "normal" chips. Even if it says that CRC is wrong, it should have written the data successfully.

Brokly, 2023-05-29:
No, I didn't compare dumps. But there was no result either. The device was still silent.

I acted like this: 

1. Restored the factory firmware. 
2. Checked the output of debugging data. Everything worked. The loader was 1.0.8 
3. Burned the loader 1.0.1. There was a checksum error 
4. Burned image_bk7231n_app.0x011000.rbl. I am sure that new keys have been installed in the configuration. There was also a checksum error. 

As a result, the log is quiet, the device does not communicate.

kuba2k2, 2023-05-29:
That's because you burned 1.0.1 loader. Try the same steps, but without step 3. Use BkWriter, not ltchiptool. It should show CRC error, that's expected.
The 1.0.1 loader won't work, as it's encrypted using different keys.

Brokly, 2023-05-29:
Oops... It seems to me that I did. But I've done so many manipulations that I'm not sure anymore. Unfortunately, I won't be able to check it until the day after tomorrow:(

Brokly, 2023-06-01:
Yes! It worked! Compiled the firmware, using keys and just flashed it without using any separate programmer. Librettina did everything herself. Thank you very much! However, now I have two chips with the same MAC - the result of my experiments. Thank you very much for your help !

kuba2k2, 2023-06-01:
You should be able to program it using OTA now (as long as you keep the keys in there). 
The MAC is stored somewhere in the flash, and if you don't have the stock firmware backup, you can't restore it (you could at least change it manually, to some made-up value).

deepcoder, 2023-06-01:
Can you share the product vendor and model info? Thx!

Brokly, 2023-06-01:
Vendor: https://sberdevices.ru .
 Model: SBDV-00115. 
The manufacturer is unknown, possibly Tuya OEM.
Made in China

Brokly, 2023-06-01:
It turned out that the problem was not completely solved. I can't download the firmware via OTA. The boot process is going fine, but after a reboot, the firmware remains the same :(

kuba2k2, 2023-06-01:
Watch the bootloader logs, it should be printing some messages about OTA. It's likely that the OTA address is different, I didn't check that.

Brokly, 2023-06-02:
Loader log without errors.

[0;36m[D][ota:147]: Starting OTA Update from 192.168.193.169...[0m
[0;37m[V][ota:178]: OTA features is 0x01[0m
[0;37m[V][ota:198]: Auth: Nonce is 6760b96d7ed5b08412319f3eb2df9265[0m
[0;37m[V][ota:218]: Auth: CNonce is 724240ca14aa28485843c51f62c82f2b[0m
[0;37m[V][ota:225]: Auth: Result is ca5bd709eca93072e52665d18508da89[0m
[0;37m[V][ota:233]: Auth: Response is ca5bd709eca93072e52665d18508da89[0m
[0;37m[V][ota:261]: OTA size is 3118592 bytes[0m
[0;37m[V][ota:278]: Update: Binary MD5 is 33318de6af39b64ce5f5dbcae9738892[0m
[0;36m[D][ota:316]: OTA in progress: 0.0%[0m
[0;36m[D][ota:316]: OTA in progress: 5.7%[0m
[0;36m[D][ota:316]: OTA in progress: 13.5%[0m
[0;36m[D][ota:316]: OTA in progress: 20.7%[0m
[0;36m[D][ota:316]: OTA in progress: 27.7%[0m
[0;36m[D][ota:316]: OTA in progress: 32.4%[0m
[0;36m[D][ota:316]: OTA in progress: 37.3%[0m
[0;36m[D][ota:316]: OTA in progress: 42.3%[0m
[0;36m[D][ota:316]: OTA in progress: 45.5%[0m
[0;36m[D][ota:316]: OTA in progress: 50.1%[0m
[0;36m[D][ota:316]: OTA in progress: 57.1%[0m
[0;36m[D][ota:316]: OTA in progress: 62.8%[0m
[0;36m[D][ota:316]: OTA in progress: 66.7%[0m
[0;36m[D][ota:316]: OTA in progress: 70.8%[0m
[0;36m[D][ota:316]: OTA in progress: 75.0%[0m
[0;36m[D][ota:316]: OTA in progress: 79.2%[0m
[0;36m[D][ota:316]: OTA in progress: 83.0%[0m
[0;36m[D][ota:316]: OTA in progress: 87.0%[0m
[0;36m[D][ota:316]: OTA in progress: 90.9%[0m
[0;36m[D][ota:316]: OTA in progress: 92.4%[0m
[0;36m[D][ota:316]: OTA in progress: 95.3%[0m
[0;36m[D][ota:316]: OTA in progress: 99.0%[0m
[0;32m[I][ota:349]: OTA update finished![0m
[0;32m[I][app:134]: Rebooting safely...[0m
[0;36m[D][lt.preferences:104]: Saving 1 preferences to flash...[0m
[0;37m[V][lt.preferences:115]: sync: key: 233825507, len: 4[0m
[0;36m[D][lt.preferences:132]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed[0m
BK7231n_1.0.8
CPSR:0x000000D3
R0:0x00800000
R1:0x00000000
R2:0x005AA000
R3:0x00000006
R4:0x00400001
R13:0x00401C1C
R14(LR):0x000033AC
ST:0x00010001
[32;22m[I/FAL] Fal(V0.4.0)success[0m
[36;22m[I/OTA] RT-Thread OTA package(V0.2.4) initialize success.[0m

go os_addr(0x10000)..........
[1;32mI [      0.000] [0mLibreTiny v1.0.2+sha.f7c28ee on cb2l, compiled at Jun  2 2023 15:26:21, GCC 10.3.1 (-O1)

It's strange that if I compile the firmware without the board_build keys.bkota.key and board_build.bkota.iv , then the log still looks normal. But in any case, the firmware remains old.

[0;36m[D][ota:147]: Starting OTA Update from 192.168.193.169...[0m
[0;37m[V][ota:178]: OTA features is 0x01[0m
[0;37m[V][ota:198]: Auth: Nonce is 43442ab1f8e346803b3d95cb91f6a5fe[0m
[0;37m[V][ota:218]: Auth: CNonce is 4f494a6513adccf44e5d45bac2565f92[0m
[0;37m[V][ota:225]: Auth: Result is 09bb2957b6a113a2caf9c289f698f3f0[0m
[0;37m[V][ota:233]: Auth: Response is 09bb2957b6a113a2caf9c289f698f3f0[0m
[0;37m[V][ota:261]: OTA size is 3118592 bytes[0m
[0;37m[V][ota:278]: Update: Binary MD5 is 2397b9dd6b4a64e588a9d22345a80982[0m
[0;36m[D][ota:316]: OTA in progress: 0.0%[0m
[0;36m[D][ota:316]: OTA in progress: 7.7%[0m
[0;36m[D][ota:316]: OTA in progress: 12.2%[0m
[0;36m[D][ota:316]: OTA in progress: 18.6%[0m
[0;36m[D][ota:316]: OTA in progress: 24.1%[0m
[0;36m[D][ota:316]: OTA in progress: 29.2%[0m
[0;36m[D][ota:316]: OTA in progress: 33.9%[0m
[0;36m[D][ota:316]: OTA in progress: 38.1%[0m
[0;36m[D][ota:316]: OTA in progress: 43.0%[0m
[0;36m[D][ota:316]: OTA in progress: 49.0%[0m
[0;36m[D][ota:316]: OTA in progress: 52.1%[0m
[0;36m[D][ota:316]: OTA in progress: 55.1%[0m
[0;36m[D][ota:316]: OTA in progress: 57.9%[0m
[0;36m[D][ota:316]: OTA in progress: 60.5%[0m
[0;36m[D][ota:316]: OTA in progress: 62.9%[0m
[0;36m[D][ota:316]: OTA in progress: 65.6%[0m
[0;36m[D][ota:316]: OTA in progress: 69.7%[0m
[0;36m[D][ota:316]: OTA in progress: 74.1%[0m
[0;36m[D][ota:316]: OTA in progress: 78.5%[0m
[0;36m[D][ota:316]: OTA in progress: 82.9%[0m
[0;36m[D][ota:316]: OTA in progress: 87.1%[0m
[0;36m[D][ota:316]: OTA in progress: 89.9%[0m
[0;36m[D][ota:316]: OTA in progress: 94.0%[0m
[0;36m[D][ota:316]: OTA in progress: 96.8%[0m
[0;36m[D][ota:316]: OTA in progress: 98.7%[0m
[0;32m[I][ota:349]: OTA update finished![0m
[0;32m[I][app:134]: Rebooting safely...[0m
[0;36m[D][lt.preferences:104]: Saving 1 preferences to flash...[0m
[0;37m[V][lt.preferences:115]: sync: key: 233825507, len: 4[0m
[0;36m[D][lt.preferences:132]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed[0m
BK7231n_1.0.8
CPSR:0x000000D3
R0:0x00800000
R1:0x00000000
R2:0x005AA000
R3:0x00000006
R4:0x00400001
R13:0x00401C1C
R14(LR):0x000033AC
ST:0x00010001
[32;22m[I/FAL] Fal(V0.4.0)success[0m
[36;22m[I/OTA] RT-Thread OTA package(V0.2.4) initialize success.[0m

go os_addr(0x10000)..........
[1;32mI [      0.000] [0mLibreTiny v1.0.2+sha.f7c28ee on cb2l, compiled at Jun  2 2023 15:26:21, GCC 10.3.1 (-O1)

Could the OTA keys be with an error ?

Brokly, 2023-06-02:
I accidentally closed the problem :(

kuba2k2, 2023-06-02:
The OTA partition is on a different address. Add the following to platformio_options block:
`board_flash.download: "0x132000+0x9E000"`

Keep the keys as well.

closed 2023-06-02 by Brokly

cheburah, 2023-11-28:
Hi. I flashed the chip with esphome firmware via ltchiptool, but did not first backup the factory firmware. Is it possible to find out the keys for the platformio_options block? Or can the device be thrown away?

kuba2k2, 2023-11-28:
Does your device come with non-Tuya firmware that uses different encryption keys? How do you know that?

ltchiptool v4.8.0 has "Read chip info" option, which will show you the encryption keys. Note that it's only possible on BK7231N, not T or any other version.

cheburah, 2023-11-28:
Thanks, I'll try to download version 4.8.0 and see the information

kuba2k2, 2023-11-28:
CRC error is unrelated to encryption keys. Tuya firmware almost always uses default keys - OP's device wasn't manufactured with Tuya Cloud. Try the other fixes mentioned in this thread, such as using BkWriter or other tools. Please also post info from the Chip Info window, so that we can check the flash ID.

cheburah, 2023-11-28:
I'll try it tonight after work. But BkWriter doesn't understand .rbl files, only .bin
[image]

cheburah, 2023-11-28:
Home assustant downloads only .rbl or uf2 firmware for this chip.

kuba2k2, 2023-11-28:
RBL is the format flashable to 0x11000. Rename it to bin and then flash it. Don't do this with UF2, it will not work.

cheburah, 2023-11-28:
0x11000 is Start Address? I don't understand well, sorry.

kuba2k2, 2023-11-28:
Yes, it's the start address. Please read the LibreTiny documentation on Beken 72xx.

cheburah, 2023-11-28:
BkWriter says
Flash ID:0x00000000
Time:2.803s
Speed of Programm:115200bps
deprotecting flashFAILED

cheburah, 2023-11-28:
[image]

cheburah, 2023-11-28:
[image]

cheburah, 2023-11-28:
But BkWriter read the dump

cheburah, 2023-11-28:
Understood thanks

Cossid, 2023-11-28:
`chip_type` is 3 which means you have a BL0942 which should be on `RX1` and `TX`

cheburah, 2023-11-28:
26 | P10/UART1_RXD
27 | P11/UART1_TXD
Right?
There should also be sel_pin

kuba2k2, 2023-11-28:
Since you already have ltchiptool, you can just put the dump file in UPK2ESPHome tab, and it will generate ESPHome YAML config for you.

[30 more replies]