libretiny-eu/libretiny #133: LibreTiny-ESPHome: arm-none-eabi-g++: not found

nahchetay69, 2023-05-27:
Hi,

   I previously had LibreTuya-ESPhome working in a docker container and have several blk72* devices setup off a raspberry pi.   I noticed that it has noe become LibreTiny-ESPhome and I cloned LibreTiny-ESPhome and installed the LibreTiny module as directed.   Then build the docker image, when I tried to create the esphome binary I get the error arm-none-eabi-g++: not found.   So I then tried via the cli method in my local account and I get the same message here is the output from a sample yaml using the wizard for a wb2s:

          rpaul@libretuya:~/libretiny-esphome $ python -m esphome compile example.yml
          INFO Reading configuration example.yml...
          INFO Generating C++ source...
          INFO Core config or version changed, cleaning build files...
          INFO Compiling app...
          Processing test-device (board: cb2s; framework: arduino; platform: https://github.com/kuba2k2/libretiny.git)
          ------------------------------------------------------------------------------------------------------------------------------------
          Platform Manager: Installing git+https://github.com/kuba2k2/libretiny.git
          INFO Installing git+https://github.com/kuba2k2/libretiny.git
          git version 2.30.2
          Cloning into '/home/rpaul/.platformio/.cache/tmp/pkg-installing-k0tcxmgu'...
          remote: Enumerating objects: 597, done.
          remote: Counting objects: 100% (597/597), done.
          remote: Compressing objects: 100% (443/443), done.
          remote: Total 597 (delta 135), reused 448 (delta 123), pack-reused 0
          Receiving objects: 100% (597/597), 367.38 KiB | 3.09 MiB/s, done.
          Resolving deltas: 100% (135/135), done.
          Platform Manager: libretiny@1.0.2+sha.5c4da6e has been installed!
          INFO libretiny@1.0.2+sha.5c4da6e has been installed!
          Tool Manager: Installing git+https://github.com/libretiny-eu/framework-beken-bdk#v2021.06.07
          INFO Installing git+https://github.com/libretiny-eu/framework-beken-bdk#v2021.06.07
          git version 2.30.2
          Cloning into '/home/rpaul/.platformio/.cache/tmp/pkg-installing-wiuoxsr3'...
          remote: Enumerating objects: 3486, done.
          remote: Counting objects: 100% (3486/3486), done.
          remote: Compressing objects: 100% (2785/2785), done.
          remote: Total 3486 (delta 586), reused 3430 (delta 586), pack-reused 0
          Receiving objects: 100% (3486/3486), 24.50 MiB | 9.47 MiB/s, done.
          Resolving deltas: 100% (586/586), done.
          Note: switching to '6491b8c001b737c790cba0bee16e548859d5e6fe'.
          
          You are in 'detached HEAD' state. You can look around, make experimental
          changes and commit them, and you can discard any commits you make in this
          state without impacting any branches by switching back to a branch.
          
          If you want to create a new branch to retain commits you create, you may
          do so (now or later) by using -c with the switch command. Example:
          
            git switch -c <new-branch-name>
          
          Or undo this operation with:
          
            git switch -
          
          Turn off this advice by setting config variable advice.detachedHead to false
          
          Updating files: 100% (3077/3077), done.
          Tool Manager: framework-beken-bdk@0.0.0+v2021.06.07.sha.6491b8c has been installed!
          INFO framework-beken-bdk@0.0.0+v2021.06.07.sha.6491b8c has been installed!
          Tool Manager: Installing git+https://github.com/libretiny-eu/ArduinoCore-API#2022.08.24
          INFO Installing git+https://github.com/libretiny-eu/ArduinoCore-API#2022.08.24
          git version 2.30.2
          Cloning into '/home/rpaul/.platformio/.cache/tmp/pkg-installing-_h2j7cfx'...
          remote: Enumerating objects: 145, done.
          remote: Counting objects: 100% (145/145), done.
          remote: Compressing objects: 100% (106/106), done.
          remote: Total 145 (delta 59), reused 84 (delta 32), pack-reused 0
          Receiving objects: 100% (145/145), 197.01 KiB | 2.29 MiB/s, done.
          Resolving deltas: 100% (59/59), done.
          Note: switching to '237b10a4f6da14bf32e5e0963b7f28865216a60c'.
          
          You are in 'detached HEAD' state. You can look around, make experimental
          changes and commit them, and you can discard any commits you make in this
          state without impacting any branches by switching back to a branch.
          
          If you want to create a new branch to retain commits you create, you may
[... 66 lines cut ...]
          *** [.pioenvs/test-device/src/esphome/components/api/api_frame_helper.cpp.o] Error 127
          *** [.pioenvs/test-device/src/esphome/components/api/api_pb2.cpp.o] Error 127
          sh: 1: arm-none-eabi-g++: not found
          *** [.pioenvs/test-device/src/esphome/components/api/api_pb2_service.cpp.o] Error 127
          =================================================== [FAILED] Took 22.50 seconds ===================================================
          
during the install of LibreTiny module via platoformio I do see the toolchain was installed:

      Tool Manager: library-printf@6.1.0+sha.28a79bd has been installed!
      INFO:Tool Manager:library-printf@6.1.0+sha.28a79bd has been installed!
      Tool Manager: Installing toolchain-gccarmnoneeabi
      INFO:Tool Manager:Installing toolchain-gccarmnoneeabi
      Downloading  [####################################]  100%
      Unpacking  [####################################]  100%
      Tool Manager: toolchain-gccarmnoneeabi@1.100301.220327 has been installed!
      INFO:Tool Manager:toolchain-gccarmnoneeabi@1.100301.220327 has been installed!
      Tool Manager: Installing git+https://github.com/libretiny-eu/ltchiptool#v4.0.0
      INFO:Tool Manager:Installing git+https://github.com/libretiny-eu/ltchiptool#v4.0.0

Any insight on what I might be missing or if you need any other output.

Thanks for creating this port so we don't have to solder replace with an ESP chip.

Thanks
Rick

catalin2402, 2023-05-27:
You can use prebuilt docker images for libretiny now: https://www.youtube.com/watch?v=sSj8f-HCHQ0&t=100s

kuba2k2, 2023-05-27:
From the posted logs I'm not sure if you're running this inside docker or not, but I suspect that you aren't. The paths printed are referring to your /home directory, which Docker wouldn't have access to.

What Raspberry Pi are you running this on, and is this a 32 or 64 bit OS?

nahchetay69, 2023-05-27:
the logs I posted above was not from inside docker. it was in my local account on the pi.  When the docker was not working I figured would try the cli path outside of docker to confirm all was installed correctly and noticed it was getting the same failure.   running 64 bit
Linux libretuya 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

Awesome video and huge kudos for the HASS add on, which is a much better place where I would like it.   

back to the rpi, so I did the docker pull of the ghcr.io image and started it.  the dashboard came up and when I tried to compile one of the devices the same error occurred again.  Have attached the log output from the dashboard running in the rpi docker image.
[logs_kitchen-main-2-switch_compile.txt](https://github.com/kuba2k2/libretiny/files/11582952/logs_kitchen-main-2-switch_compile.txt)

Interestingly the image running via my HA installing compiles just fine so I will move my yaml's over to there and use it from there.    My HA runs on a intel nuc so is a different image.

not sure why it is not working for me on the pi.   I guess next step is to basically wipe the pi out and start fresh again.  I use it mainly for the tuya-cloudcutter path.

I will wipe and retry later today, if that resolves it, i will close this issue out.

Thanks

nahchetay69, 2023-05-28:
Well a fresh install did not solve it.   This time I followed your steps in the video on installing the 32 bit lite version, install python and docker.  Then installed the docker image and I still get the compile error:

Compiling .pioenvs/garage_west_switch/src/esphome/components/api/api_connection.cpp.o
sh: 1: arm-none-eabi-g++: not found
Compiling .pioenvs/garage_west_switch/src/esphome/components/api/api_frame_helper.cpp.o
sh: 1: arm-none-eabi-g++: not found
Compiling .pioenvs/garage_west_switch/src/esphome/components/api/api_pb2.cpp.o
sh: 1: arm-none-eabi-g++: not found
*** [.pioenvs/garage_west_switch/src/esphome/components/api/api_connection.cpp.o] Error 127
*** [.pioenvs/garage_west_switch/src/esphome/components/api/api_frame_helper.cpp.o] Error 127
*** [.pioenvs/garage_west_switch/src/esphome/components/api/api_pb2.cpp.o] Error 127
Compiling .pioenvs/garage_west_switch/src/esphome/components/api/api_pb2_service.cpp.o
sh: 1: arm-none-eabi-g++: not found
*** [.pioenvs/garage_west_switch/src/esphome/components/api/api_pb2_service.cpp.o] Error 127

Attached is the screen shot of the fresh install, only slight mod was I changed the hostname.  Also is the logs of the docker compile along with the yml file.   This was compiling under LibreTuya and not sure why it is not under LibreTiny.
Here are the logs:
[logs_garage-west-switch_compile.txt](https://github.com/kuba2k2/libretiny/files/11583689/logs_garage-west-switch_compile.txt)
[libretiny docker install.txt](https://github.com/kuba2k2/libretiny/files/11583691/libretiny.docker.install.txt)
[garage_west_switch.txt](https://github.com/kuba2k2/libretiny/files/11583700/garage_west_switch.txt) had to change ext from yml to txt to upload.

As I mentioned previously it is working for me under the HASS add on and will use it from there.  I wanted to confirm if it was something on the previous image that was causing it.  Now with a fresh install and still seeing it, will leave it up to you if you want more info.

catalin2402, 2023-05-28:
Try with framework version dev instead of latest

nahchetay69, 2023-05-28:
Still same compile error.   I even tried using just the generic config when you create a new device and still see the compile error.   using this yaml file:

esphome:
  name: test-device
  friendly_name: test-device

libretiny:
  board: generic-bk7231t-qfn32-tuya
  framework:
    version: dev

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "tC8kAdNSGgeHIlyungY+56AbCGWrKAIUFTxiNKrCt6g="

ota:
  password: "eefa6e79797b95506b9d43e6358c1ffc"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot in case wifi connection fails
  ap:
    ssid: "Test-Device Fallback Hotspot"
    password: "p3LlgjjyaSK4"

closed 2023-07-13 by kuba2k2