libretiny-eu/libretiny #134: [BUG/Missing] [BK7231N / CB3S] deep sleep support

dskw, 2023-06-01:
Hello, 

I'm experimenting with libretiny and a battery powered door/window sensor.
My initial tries were done using OpenBeken, documented on this page: 
https://www.elektroda.com/rtvforum/viewtopic.php?p=20586001#20586001
It uses the CB3S board with a BK7231N chip and no additional Tuya MCU, running on a battery.

After finding out about libretiny I wanted to give it a try as well.
So far the sensors work fine and connections can be established fast.

Deep Sleep support seems to be missing though.
The chip supports wake up on pin interrupt like ESP32, but it is not supported on libretiny yet.
And even when using the regular time based deep sleep it instantly wakes up again:

`[08:02:52][C][lt.component:013]: LibreTiny:
[08:02:52][C][lt.component:014]:   Version: 1.0.2+sha.5c4da6e
[08:02:52][C][lt.component:015]:   Loglevel: 3
[08:02:52][C][deep_sleep:049]: Setting up Deep Sleep...
[08:02:52][C][deep_sleep:052]:   Sleep Duration: 300000 ms
[08:02:52][D][api:102]: Accepted 192.168.23.5
[08:02:52][D][api.connection:958]: Home Assistant 2023.5.4 (192.168.23.5): Connected successfully
[08:02:53][D][sensor:102]: 'battery_adc': Sending state 329.00000 V with 0 decimals of accuracy
[08:02:53][D][sensor:102]: 'Battery Voltage': Sending state 2431.39038 mV with 0 decimals of accuracy
[08:02:53][D][sensor:102]: 'Battery Level': Sending state 28.92380 % with 0 decimals of accuracy
[08:02:53][D][main:098]: Updated battery
[08:02:53][I][deep_sleep:116]: Beginning Deep Sleep
[08:02:53][I][deep_sleep:118]: Sleeping for 300000000us
[08:02:53][D][lt.preferences:104]: Saving 1 preferences to flash...
[08:02:53][D][lt.preferences:132]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
INFO Processing expected disconnect from ESPHome API for 192.168.23.164
WARNING Disconnected from API
INFO Successfully connected to 192.168.23.164
[08:02:57][D][sensor:102]: 'battery_adc': Sending state 328.00000 V with 0 decimals of accuracy
[08:02:57][D][sensor:102]: 'Battery Voltage': Sending state 2424.00000 mV with 0 decimals of accuracy
[08:02:57][D][sensor:102]: 'Battery Level': Sending state 28.00000 % with 0 decimals of accuracy
[08:02:57][D][main:098]: Updated battery
[08:02:57][I][deep_sleep:116]: Beginning Deep Sleep
[08:02:57][I][deep_sleep:118]: Sleeping for 300000000us
[08:02:57][D][lt.preferences:104]: Saving 1 preferences to flash...
[08:02:57][D][lt.preferences:132]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
INFO Processing expected disconnect from ESPHome API for 192.168.23.164
WARNING Disconnected from API
INFO Successfully connected to 192.168.23.164
[08:03:01][D][sensor:102]: 'battery_adc': Sending state 328.00000 V with 0 decimals of accuracy
[08:03:01][D][sensor:102]: 'Battery Voltage': Sending state 2424.00000 mV with 0 decimals of accuracy
[08:03:01][D][sensor:102]: 'Battery Level': Sending state 28.00000 % with 0 decimals of accuracy
[08:03:01][D][main:098]: Updated battery
[08:03:01][I][deep_sleep:116]: Beginning Deep Sleep
[08:03:01][I][deep_sleep:118]: Sleeping for 300000000us
[08:03:01][D][lt.preferences:104]: Saving 1 preferences to flash...
[08:03:01][D][lt.preferences:132]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
INFO Processing expected disconnect from ESPHome API for 192.168.23.164
WARNING Disconnected from API`

kuba2k2, 2023-06-01:
Correct - deep sleep is not supported yet. It will be, in the future. Let's keep this issue open.

dskw, 2023-06-01:
Thanks for the reply kuba!
I could try using my device to replicate the OpenBeken implementation and make it available for the BK7231N platform similar to ESP32.
Does it make sense to do this in this repo, or do you plan to shut it down once it has been merged into esphome? (I'm following the integration ticket there)

kuba2k2, 2023-06-01:
This repo, the one where you're posting this issue (kuba2k2/libretiny) will stay - it's the base platform of the port.
The kuba2k2/libretiny-esphome repo will be outdated as soon as it gets merged into ESPHome upstream.

However, adding new features like this needs to be done in **both of these repos**. The base platform needs to expose a common API (C/C++ headers and classes) for a particular feature. Then, it can be implemented within the ESPHome HAL.

dskw, 2023-06-01:
Oh dang.. should have noticed this.. Sorry, I got confused with the repo links. In that case I'll have a look if I can get something in. Worst case will have to wait for another contributor with more knowledge of the code base.
Thanks for the clarification!

closed 2023-07-13 by kuba2k2