libretiny-eu/libretiny #135: Pullup does not work in ESPHome on WB3S

dungeon77, 2023-06-01:
Hello, Kuba. Thank you for the great work!
I can't get one moment.
I have a lamp on a jb3 chip, it has an IR receiver on pin P1. In ESPHome, the receiver does not want to work, no matter what settings I make. But it works fine in Openbecken. And also it starts working in ESPHome if I solder a 10k resistor from the data leg to 3.3v
Tell me, what am I doing wrong? Maybe you can advise something

kuba2k2, 2023-06-01:
Can you post your config?

dungeon77, 2023-06-01:
It's useless, I've tried everything. For example,

remote_receiver:
   - id: receiver_ir
     pin:
       number: P1
       inverted: false
       mode:
         input: true
         pullup: true
     dump: all

kuba2k2, 2023-06-01:
It's possible that pull-up doesn't work with interrupts. It's being fixed on the AmebaZ2 branch, but it's not ready yet. You could test it if you wanted, but that requires a fair bit of setup.

dungeon77, 2023-06-01:
just tell me what I should do :)

dungeon77, 2023-06-01:
OK, I'll try))

dungeon77, 2023-06-01:
Kuba, I'm sorry, but what if I don't use PlatformIO? I have libretiny (although still libretuya)) installed on python in windows. As well as the libretiny addon in homeassistant. 
I have platformio installed, but my knowledge of it is extremely small

kuba2k2, 2023-06-01:
Yeah - it requires some complicated setup. The best solution for you would be to wait until we merge the changes :smiley:

kuba2k2, 2023-08-18:
@dungeon77 can you check if it works on latest LibreTiny version? (1.2.1 right now)

dungeon77, 2023-08-18:
I still have the old version. How can it be updated? I have the Libertiny ESPHome addon in Home Assistant. Or just wait?

PLATFORM VERSIONS:
 - libretiny @ 1.0.1+sha.113b2fc
 - ltchiptool @ 4.2.3
CUSTOM OPTIONS:
 - fw_name = esphome
 - fw_version = 2023.7.0-dev

kuba2k2, 2023-08-18:
Update the addon then. You might need to clear its data too.

catalin2402, 2023-08-18:
Or simply uninstall and reinstall it. This way all the cached data is wiped and only the configs will be retained

dungeon77, 2023-08-18:
the addon has been updated. And libretiny is still 1.0.1
How to update it? I thought the addon would download, but it's not

catalin2402, 2023-08-18:
Uninstall and reinstall it. It will retain your configs and it will delete all the cached build files (cached platofrms, libraries, etc.)

dungeon77, 2023-08-21:
Sorry, a lot of work and little free time.
In general, it turned out to update all the components, I flashed one of the lamps with a new firmware, but... The first attempts failed. While the remote does not react the same way as before. When there is more time, I will make more attempts.

HARDWARE: BK7231T 120MHz, 256KB RAM, 1.03MB Flash
 - framework-arduino-api @ 2022.8.24+sha.237b10a 
 - framework-beken-bdk @ 0.0.0+v2021.06.07.sha.6491b8c 
 - library-flashdb @ 1.2.0+sha.d5c892f 
 - library-freertos @ 9.0.0+sha.95cc959 
 - library-freertos-port @ 2023.5.23+sha.a917d93 
 - library-lwip @ 2.1.3-bdk+sha.4ee4d34 
 - library-printf @ 6.1.0+sha.28a79bd 
 - library-uf2ota @ 5.0.0+sha.f955412 
PLATFORM VERSIONS:
 - libretiny @ 1.2.1
 - ltchiptool @ 4.2.3
CUSTOM OPTIONS:
 - fw_name = esphome
 - fw_version = 2023.9.0-dev

Config:

remote_receiver:
  id: ir_rec
  pin:
    number: P1
    mode:
      pullup: true
      input: true
    inverted: true
  dump: 
    - lg

dungeon77, 2023-08-21:
with this configuration, but with a soldered resistor, the receiver works fine