libretiny-eu/libretiny #136: Error: Could not find one of 'package.json' manifest files in the package

Brokly, 2023-06-10:
Hi ! I am running esphome-libretiny under Windows + Python. I tried to update libretiny and apparently made some mistake. Now when I try to compile any firmware, I get an error: 

D:\esphome\libretiny-esphome>python -m esphome compile d:/config/lamp-rgbww-sbgu10-7.yaml
INFO ESPHome 2023.6.0-dev
INFO Reading configuration d:/config/lamp-rgbww-sbgu10-7.yaml...
INFO Generating C++ source...
INFO Compiling app...
**************************************************************************************************
Obsolete PIO Core v6.1.6 is used (previous was 6.1.7)
Please remove multiple PIO Cores from a system:
https://docs.platformio.org/en/latest/core/installation/troubleshooting.html
***************************************************************************************************
Processing lamp-rgbww-sbgu10-7 (board: wb2l; framework: arduino; platform: https://github.com/kuba2k2/libretiny.git)
---------------------------------------------------------------------------------------------------------------------------------------------------Error: Could not find one of 'package.json' manifest files in the package

Maybe you can help me fix this problem ?

kuba2k2, 2023-06-10:
Try removing everything related to LibreTiny (or even entire .platformio/platforms and .platformio/packages directories).

Brokly, 2023-06-10:
Yes, deleting the .platformio/platforms and .platformio/packages directories helped perfectly! Thanks again for your help !

closed 2023-06-10 by Brokly