libretiny-eu/libretiny #137: Aubess 20A - plug SW reboot each night at 3 am

ladykira2022, 2023-06-13:
Hi.
I have several Aubess 20A plugs and all of them perform a SW reboot at 3 am.

They are using : v1.0.2+sha.5c4da6e on generic-bk7231n-qfn32-tuya, compiled at Jun 8 2023 09:18:46, GCC 10.3.1 (-O1)

[image]
[logs_3-aubess_logs.txt](https://github.com/kuba2k2/libretiny/files/11731180/logs_3-aubess_logs.txt)

Yaml Below:

`esphome:
  name: "3-aubess"

libretiny:
  board: generic-bk7231n-qfn32-tuya
  framework:
    version: dev

logger:

web_server:

captive_portal:

mdns:

api:
  password: !secret api_password

ota:
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ap:

button:
- platform: restart
  name: Restart

debug:
  update_interval: 30s

text_sensor:
- platform: debug
  reset_reason:
    name: Reset Reason
- platform: libretiny
  version:
    name: LibreTiny Version

sensor:
- platform: uptime
  name: Uptime
- platform: hlw8012
  model: BL0937
  cf_pin:
    number: P7
    inverted: true
  cf1_pin:
    number: P6
    inverted: true
  sel_pin:
    number: P24
    inverted: true
  current:
    name: BL0937 Current
  voltage:
    name: BL0937 Voltage
  power:
    name: BL0937 Power
  energy:
    name: BL0937 Energy
  voltage_divider: 800
  current_resistor: 0.001 ohm

binary_sensor:
- platform: gpio
  id: binary_switch_1
  pin:
    number: P23
    inverted: true
    mode: INPUT_PULLUP
  on_press:
    then:
    - switch.toggle: switch_1

switch:
- platform: gpio
  id: switch_1
  name: Relay 1
  pin: P26

status_led:
  pin: P8
`

dr3amr, 2023-06-13:
Same thing happens to me. Just recently converted two of Aubess 20A to libretiny based firmware and I get SW Reboot each night. The libretiny version seems current at 1.0.2. To note that the firmware is compiled using the LibreTiny ESPHome Home Assistant Addon.

kuba2k2, 2023-06-13:
ESPHome will auto-reboot if it fails to connect with Home Assistant for more than 15 minutes. Are you sure your HA instance (or network routers/access points) are available at that time? I can't think of any other reason why it would reboot every 24 hours.

dr3amr, 2023-06-13:
@kuba2k2 indeed, I have a router cronjob set at 4:30 each night to reboot the router. Seems that all esp devices connected to HA have same behavior (which I did not noticed before). Is there any way to prevent this auto-reboot on connectivity loss with HA API ?

dr3amr, 2023-06-13:
seems there is reboot_timeout parameter under api: which has default value of 15min, 0s disables it. will try that for the moment. Thanks for clarifications.

kuba2k2, 2023-06-13:
Yes, that's exactly this parameter. Note that there's a separate parameter like this for WiFi I think.

dr3amr, 2023-06-14:
@kuba2k2 seems even after adding the reboot_timeout with value 0s to api: and wifi:, and compiling new firmware, still I get same device auto-reboot, when my router is restarted. The restart should not take anyway 15 min but still device auto-reboots. Will do some more research on how to prevent this auto-reboot.

closed 2023-07-13 by kuba2k2