libretiny-eu/libretiny #138: Flashing esphome on EKAZA Smart Plug 16A - TYWE2S - MCU V1.1.7 - The profile you selected did not result in a successful exploit

camnop, 2023-06-15:
Hello everyone,

I'm seeking help regarding flashing esphome in wireless mode on the [EKAZA Smart Plug 16A](
https://www.ekaza.com.br/automacao/tomada-inteligente-16a/) with energy meter. Here are some details about the device and its firmware:

Device details:

Model: TYWE2S
TUYA MCU V1.1.7

I have attempted to flash the device using the following options:

- 1.1.7 - BK7231N / CL_DREAM_RGB_STRIP_19KEY_1KEY_5V_BK7231N
- 1.1.7 - BK7231N / oem_bk7231n_ceiling_light_ty
- 1.1.7 - BK7231N / oem_bk7231n_plug
- 1.1.7 - BK7231T / oem_bk7231s_light_pir_ty
- 1.1.71 - BK7231T / bk7231t_common_user_config_ty

Unfortunately, none of these attempts have been successful. I consistently receive the following error message:

> "Found access point name: 'SmartLife-5229', trying to connect...
> Device 'wlp1s0' successfully activated with '0eec4e65-xxxxxx-xxxx-xxe16a0ebf88'.
> Connected to access point.
> [!] The profile you selected did not result in a successful exploit."

I would greatly appreciate any assistance or guidance you can provide to help me successfully flash esphome on this device in wireless mode to instlal esphome

Thank you in advance for your support!

Best regards,

[image]
[image]
[image]

[image]

Cossid, 2023-06-15:
TYWE2S is an ESP chip, which is not compatible with LibreTiny or Tuya-CloudCutter.

You can try Tuya-Convert for wireless flashing, but that exploit has been patched for quite some time.  If that doesn't work, your only option will be serial flashing with regular ESPHome (non-LibreTiny).

camnop, 2023-06-15:
Thanks a bunch for getting back to me with your suggestions. So, it seems like the Tuya-Convert wireless flashing option might be a no-go since the exploit has been patched. You mentioned that serial flashing with regular ESPHome is the only remaining option. Fair enough.

Here's the thing that's been tickling my curiosity: Is there any way we could spice things up and make LibreTiny compatible with ESP chips like TYWE2S?  Any insights or info about the possibility of expanding LibreTiny's horizons to embrace ESP chips? 

Thanks a million for your help, you rock!

Cossid, 2023-06-15:
> Here's the thing that's been tickling my curiosity: Is there any way we could spice things up and make LibreTiny compatible with ESP chips like TYWE2S? Any insights or info about the possibility of expanding LibreTiny's horizons to embrace ESP chips?

ESPHome itself is based around ESP chips already, there is no need to use LibreTiny at all (LibreTiny is just a framework to make other chips compatible with ESPHome).  TYWE2S is ESP8266 and is already fully supported by stock ESPHome.

camnop, 2023-06-15:
Thanks for clarifying that ESPHome is already designed for ESP chips, including the TYWE2S. Good to know it's fully supported out of the box!

I have this wild idea floating in my mind. You know that cool trick from LibreTiny where you can flash chips wirelessly using Wi-Fi? I was wondering if we could borrow that nifty exploit and apply it to ESP chips as well. Imagine being able to flash them without any wires, just pure wireless magic!

Do you reckon it's technically possible to adapt that trick for ESP chips too? I'd love to hear your thoughts on this adventurous idea. haha

Cossid, 2023-06-15:
There is technically a chance it's possible, but nobody with the understanding to do it has come along yet.  ESP is vulnerable to the same exploit, but is a totally different assembly language and may or may not be able to manipulate the exploit in a usable way against that assembly code.  Tuya-CloudCutter is a completely separate entity from LibreTiny.  Tuya-CloudCutter is more akin to Tuya-Convert in that it's a tool that uses exploits to force firmware changes.  There is no expectation Tuya-CloudCutter will be expanding to ESP chips anytime soon (and Tuya has pretty much discontinued use of ESP chips, so it's only very old stock that still has them anyway).

closed 2023-06-16 by kuba2k2