libretiny-eu/libretiny #139: LibreTiny container image

vxav, 2023-06-15:
Just a quick question, are there any plans to set up actions to build images for https://github.com/kuba2k2/libretiny-esphome and push them to dockerhub/gchr?

It would be great to have upstream images 😄

catalin2402, 2023-06-15:
They are published already:
ghcr.io/libretiny-eu/libretiny-esphome-docker:latest - for the docker image
And https://github.com/libretiny-eu/esphome-hass-addon if you're using Home Assistant

vxav, 2023-06-15:
That's great thanks a lot 👍 
Maybe a quick one-liner in the [doc](https://docs.libretiny.eu/docs/projects/esphome/#find-your-devices-board) would be good as it only mentions building the image.

kuba2k2, 2023-06-16:
Since we're working on preparing ESPHome for merging with upstream, I don't think we should introduce such features right now. The fork is unversioned, tagging is theoretically possible but it would mean that there are two different v2023.5 tags of ESPHome and that's messy.

About the docs, they haven't been updated for quite some time, but you're welcome to fix it if you'd like.

vxav, 2023-06-16:
I see, pretty big deal to add support for Beken chips upstream, nice one ! 👍

closed 2023-06-16 by vxav