libretiny-eu/libretiny #145: WR02 doesn't boot

Sonic-Amiga, 2023-07-02:
After finally learning to flash these things (https://github.com/libretiny-eu/ltchiptool/issues/5) i proceeded to building esphome for my device. And it doesn't boot up. I made this minimalist .yml for testing:
```
esphome:
  name: yndx-0007

libretiny:
  board: wr2
  framework:
    version: latest
  # verbose/trace/debug/info/warn/error/fatal
  loglevel: debug
  # suppress chip's SDK log messages
  # (same as LT_UART_SILENT_ALL above)
  sdk_silent: false
  # disable SWD/JTAG so that all GPIOs can be used
  # set to false if you want to attach a debugger
  gpio_recover: false

logger:
api:
  password: ""
ota:
  password: ""

wifi:
  ssid: "MyWifi"
  password: "TopSecret"
  ap:
    ssid: "yndx-0007"
    password: ""

captive_portal:
web_server:
```
And... Here goes nothing! I get no new IP on wifi; no AP, nothing. Silence. The boot log says only this:
```
ROM:[V0.1]
FLASHRATE:4
BOOT TYPE:0 XTAL:40000000
IMG1 DATA[1168:10002000]
IMG1 ENTRY[8000541:100021ef]
IMG1 ENTER
CHIPID[000000ff]
read_mode idx:0, flash_speed idx:0
calibration_result:[1:9:9][5:d]
calibration_result:[2:13:7][1:d]
calibration_result:[3:3:3][1:3]
calibration_ok:[2:13:7]
FLASH CALIB[NEW OK]
OTA2 ADDR[80d0000]
OTAx SELE[fffffffc]
OTA1 USE
IMG2 DATA[0x8073b90:3720:0x10005000]
IMG2 SIGN[RTKWin(10005008)]
IMG2 ENTRY[0x10005000:0x8033c55]
===== Enter Image 2 ====
System_Init1
OSC8M: 7fea9f
boot reason: 0
System_Init2
```
 I guess this means the application isn't alive. Shouldn't it say at least "Hi" on the debug log ? Note i've cranked up all the debug options. Or am i just doing something stupid ?

I am flashing the device using "python -m esphome upload wr02-basic.yml", so it shouldn't be wrong file chosen.

Sonic-Amiga, 2023-07-02:
And here comes the first problem. I've found a generated C++ code, and right in the beginning it says:
```
  logger_logger = new logger::Logger(115200, 512);
  logger_logger->set_uart_selection(logger::UART_SELECTION_UART0);
  logger_logger->pre_setup();
  logger_logger->set_component_source("logger");
  App.register_component(logger_logger);
```
 which is obviously wrong, since i want UART2. Anyways UART0 pins are occupied by other functions (LED and button). But the framework doesn't let me select it! According to original ESPHome docs, i write in my UML:
```
logger:
  hardware_uart: "UART2"
```
 And the python script swears at me:
```
Failed config

logger: [source wr02-basic.yml:18]

  Unknown value 'UART2', did you mean 'UART0'?.
  hardware_uart: UART2
```
 Something unsupported/unimplemented ?

kuba2k2, 2023-07-02:
The current version considers "UART0" as the default UART port. To choose other ports you have to use SERIAL# instead of UART#.  This will be fixed in the future (feature/libretiny-prerelease branch, you can test it already if you want).

The "default" port, however, is chosen based on the available pins. Since your board has the TX2/RX2 on the back, and it's not programmed into the pinout diagrams, LT considers it as unavailable, and chooses UART1 as the default output port. This should be fixed as soon as I add the correct pinout.

TL;DR use generic-rtl8710bn-2mb-788k instead. It has all the pins available.

Sonic-Amiga, 2023-07-02:
Just found out SERIAL2, tried this setting, the same result. Well, going generic...

Sonic-Amiga, 2023-07-02:
board: generic-rtl8710bn-2mb-788k
hardware_uart: "SERIAL2"

Worked, i am getting signs of life. But it crashes at "Starting wifi scan", i guess that's another story

closed 2023-07-02 by Sonic-Amiga