libretiny-eu/libretiny #148: BK7231 SPI support

Maxx-W, 2023-07-13:
Hello, I need the Adafruit SPIFlash Library for my project, but SPI currently isn't supported by libretiny with the BK7231N. 
Can you please tell me whether SPI support is planned for the near future? Alternatively I will have to replace the BK7231N, so it would be of great help.

I found two projects with drivers, but the migration to libretiny is way above my level.
https://github.com/openshwprojects/OpenBK7231T_App/blob/main/src/driver/drv_spi.c
https://github.com/bekencorp/armino/blob/main/middleware/driver/spi/spi_driver.c

kuba2k2, 2023-07-13:
BK7231 has built-in flash, and functions for reading/writing the flash are available in LibreTiny. The [flash memory map](https://docs.libretiny.eu/boards/generic-bk7231n-qfn32-tuya/#flash-memory-map) has a "user data" partition, which you can freely use to store whatever you'd like. It's not super big, but should be way more than enough if done properly.

kuba2k2, 2023-07-13:
Yes, you can use the OTA partition, but if you ever decide to flash an OTA update, it will get erased.

Also, data stored directly in the very first block of the OTA partition may confuse the bootloader (and it may wipe that then), so better store the data at a bit bigger offset (like, move 0x1000 from the beginning of OTA partition).

Just make sure to keep away from Network, calibration, etc. partitions.

Maxx-W, 2023-07-13:
Thank you so much for the help and libretiny!

fonix232, 2023-12-12:
@kuba2k2 I have a Tuya LED controller using the CBU module (BK7231T) that utilises SPI for communicating with the LEDs. Unfortunately it's not clear which driver they use, however I'd be more than willing to help testing out an SPI component!

yesitsme007, 2024-01-18:
There are appearing a lot of devices now with addressable LEDs. Christmas lights strings, etc. like [this](https://www.elektroda.com/rtvforum/topic3990174.html#gallery-1). They often seem to be based on a module labelled "WB8" that is not documented on the Tuya developer page. Apparently they use the SPI interface to interface with the LED chain (which seems not be "real" SPI but more a "misuse" to get a high baud rate). I see that there currently is a branch "i2c-spi" 👍 . @kuba2k2: Is this already in a state to give it a try for this purpose?

kuba2k2, 2024-01-18:
SPI is not implemented for Beken chips, not even on that branch.

How do you know it isn't a real SPI interface? Is the LED driver connected to SPI pins of the Beken chip or normal GPIOs? Do you have a pinout of this module?

yesitsme007, 2024-01-18:
Thanks for quick response. According to information from this [thread](https://www.elektroda.com/rtvforum/topic4011975.html#20796676) these gadgets use a chain of WS2812 or compatible LEDs. Those kind of LEDs do not have an SPI interface but you can use SPI to address them. The mechanism how this works is explained [here](https://controllerstech.com/ws2812-leds-using-spi/). The reason for using SPI is _"they need a very precisely clocked data, which on Beken is only possible with P16, through the SPI"_ (from this thread). I fear no one has figured out a pinout yet. The mentioned forum post says it might be compatible with WB3L. I was not able so far to address UART for reading or flashing firmware. I can only confirm that Vcc and GND matches the pinout of WB3L. Sorry for not having more solid information.