libretiny-eu/libretiny #149: Arlec Grid Connect pc44ha Outdoor Plug [BK7231T / WB2S / BL0937]

tbgoose, 2023-07-19:
Hey

I have a outdoor Arlec Grid Connect plug (pc44ha). I have successfully flashed the kickstart esp home on it (it was using 1.3.5 firmware as the arlec pc191ha). This is the plug - https://www.bunnings.com.au/arlec-grid-connect-smart-ip44-heavy-duty-timer_p0273373

Using the work you from [#84](https://github.com/kuba2k2/libretiny/issues/84) and some other templates etc I have found around the place I have created the following esphome yaml, but it's not really correct. Can anyone help figure out what I am doing wrong?

There is talk of the device here including tear-down and pin-out info - https://www.elektroda.com/rtvforum/topic3980140.html. Based on the talk here I think the pin selection is correct, but obviously I am missing something RE power monitoring.

EDIT: I should mention that I am mainly using this plug to monitor power use of my pool pump + filter. I very rarely use the switch. It does turn the device on and off from the web-portal and the button on the device, but the power monitoring is completely wrong as logs here show. This was compared to a kilawatt type plug on the same load (~15-18 watts). I am in Aus so voltage is ~230-250V

```
13:46:19 | [D] | [hlw8012:082] | Got power=2.7W, voltage=0.7V
13:46:19 | [D] | [sensor:093] | 'Voltage': Sending state 0.65699 V with 1 decimals of accuracy
13:46:19 | [D] | [sensor:093] | 'Power': Sending state 2.65077 W with 1 decimals of accuracy
13:46:19 | [D] | [sensor:093] | 'Daily Consumption': Sending state 0.00109 kWh with 3 decimals of accuracy
```

esphome yaml
```
substitutions:
  device_name: pool_pump_socket
  device_friendly_name: "Pool Pump Socket"
  device_ip: 192.168.1.183
  outlet_icon: mdi:power-socket-au
  switch_restore: ALWAYS_ON

  # Higher value gives lower watt readout
  current_res: "0.0009 ohm"
  # Lower value gives lower voltage readout
  voltage_div: "769"

esphome:
  name: ${device_name}
  friendly_name: ${device_friendly_name}
  #name_add_mac_suffix: true

libretiny:
  board: wb2s
  framework:
    version: dev

logger:

web_server:
  port: 80

api:
  reboot_timeout: 15min
  encryption:
    key: !secret api_key_pool_pump

ota:
  password: !secret ota_password

captive_portal:

mdns:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: ${device_ip}
    gateway: 192.168.1.1
    subnet: 255.255.255.0
    dns1: 192.168.1.1
  ap:
    password: !secret ap_password

button:
  - platform: restart
    name: Restart

debug:
  update_interval: 30s

sensor:
  - platform: uptime
    name: Uptime
  - platform: hlw8012
    model: BL0937
    cf_pin:
      number: P7
      inverted: true
    cf1_pin:
      number: P8
      inverted: true
    sel_pin:
      number: P24
      inverted: true
    current:
      name: Current
      #unit_of_measurement: A
    voltage:
      name: Voltage
      #unit_of_measurement: V
    power:
      id: ${device_name}_power
      name: Power
      #unit_of_measurement: W
    current_resistor: ${current_res}
    voltage_divider: ${voltage_div}
    change_mode_every: 8
    update_interval: 15s
  - platform: total_daily_energy
    name: "Daily Consumption"
    power_id: ${device_name}_power
    filters:
      - multiply: 0.001
    unit_of_measurement: kWh

  - platform: wifi_signal
    name: RSSI
    update_interval: 5min

  - platform: uptime
    id: uptime_sec
    name: Uptime
    update_interval: 5min

text_sensor:
  - platform: debug
    reset_reason:
      name: "Reset Reason"
  - platform: libretiny
    version:
      name: "Libretiny Version"
  - platform: template
    name: "Up Format"
    lambda: |-
      uint32_t dur = id(uptime_sec).state;
      int dys = 0;
      int hrs = 0;
      int mnts = 0;
      if (dur > 86399) {
        dys = trunc(dur / 86400);
        dur = dur - (dys * 86400);
      }
      if (dur > 3599) {
        hrs = trunc(dur / 3600);
        dur = dur - (hrs * 3600);
      }
      if (dur > 59) {
        mnts = trunc(dur / 60);
        dur = dur - (mnts * 60);
      }
      char buffer[17];
      sprintf(buffer, "%ud %02uh %02um %02us", dys, hrs, mnts, dur);
      return {buffer};
    icon: mdi:clock-start
    update_interval: 5min
  - platform: wifi_info
    ip_address:
      name: "IP Address"
      disabled_by_default: false

output:
  - platform: ledc
    id: output_led_1
    pin: P11

light:
  - platform: monochromatic
    id: light_switch_1
    output: output_led_1

binary_sensor:
  - platform: gpio
    id: binary_switch_1
    pin:
      number: P10
      inverted: true
      mode: INPUT_PULLUP
    on_press:
      then:
        - switch.toggle: outlet

switch:
  - platform: gpio
    id: outlet
    name: None
    pin: P26
    icon: ${outlet_icon}
    on_turn_on:
      - light.turn_on: light_switch_1
    on_turn_off:
      - light.turn_off: light_switch_1
    restore_mode: ${switch_restore}
    device_class: outlet

status_led:
  pin: P6

time:
  - platform: homeassistant
    id: homeassistant_time

Cossid, 2023-07-19:
> I have created the following esphome yaml, but it's not really correct. Can anyone help figure out what I am doing wrong?

You're going to have to be more specific.  What isn't correct?  Is it simply that your voltage and wattage readings are incorrect?  You need to adjust the current resistor and voltage divider seen here in your code:

```yaml
  # Higher value gives lower watt readout
  current_res: "0.0009 ohm"
  # Lower value gives lower voltage readout
  voltage_div: "769"
```

According to `https://upk.libretiny.eu/` for the device you specified, the current resistor should be `0.001 ohm` and usually that doesn't change, in which case you need to change the voltage divider substantially until you get closer to real values.  Again, the device example shows voltage divider should be closer to `1600`.

This isn't really a LibreTiny issue, but general ESPHome component configuration.  See https://esphome.io/components/sensor/hlw8012.html?highlight=voltage_divider for notes about setting the voltage divider.

tbgoose, 2023-07-19:
Thanks for the reply - yes my power metering is incorrect. Even when I change the voltage_div: 1600, I still get wildly inaccurate voltage. To get Voltage anywhere near 

```  
  # Higher value gives lower watt readout
  current_res: "0.001 ohm"
  # Lower value gives lower voltage readout
  voltage_div: "1600"
```

This is with a 9.9-10W constant draw on the circuit from a dc vornado fan

Voltage is so far from where it needs to be, I will work on current/power first

```
13:11:27 | [D] | [hlw8012:076] | Got power=2.9W, current=8.9A
13:11:27 | [D] | [sensor:093] | 'Current': Sending state 8.86578 A with 2 decimals of accuracy
13:11:27 | [D] | [sensor:093] | 'Power': Sending state 2.94147 W with 1 decimals of accuracy
13:11:40 | [D] | [sensor:093] | 'Uptime': Sending state 837.06000 s with 0 decimals of accuracy
13:11:42 | [D] | [sensor:093] | 'Power': Sending state 3.03339 W with 1 decimals of accuracy
13:11:57 | [D] | [hlw8012:082] | Got power=2.9W, voltage=1.5V
13:11:57 | [D] | [sensor:093] | 'Voltage': Sending state 1.51040 V with 1 decimals of accuracy
13:11:57 | [D] | [sensor:093] | 'Power': Sending state 2.94147 W with 1 decimals of accuracy
13:12:12 | [D] | [hlw8012:082] | Got power=2.9W, voltage=1.5V
13:12:12 | [D] | [sensor:093] | 'Voltage': Sending state 1.51884 V with 1 decimals of accuracy
13:12:12 | [D] | [sensor:093] | 'Power': Sending state 2.94147 W with 1 decimals of accuracy
```

So to get 2.94147W to be 10W I would need to multiple by 3.399661. I can't really see in the esp-home docs how to figure out what to do with the current res value to get to the value I want though...

Cossid, 2023-07-19:
Some of these chips are just really inaccurate and you can't correct them with voltage dividers or current resistors.  The generally recommendation is to measure a known load and use a [calibrate_linear filter](https://esphome.io/components/sensor/index.html?highlight=calibrate_linear#calibrate-linear) on the fields that are off.

I have several plugs by the same brand and one of them is massively off from all the others, calibrate_linear is the only way I can get usable results with it.

tbgoose, 2023-07-19:
Ok cool, calibrating these things is a job for another day then! I'll put a pin in this one for now and read up on that process when I have the time.

Thanks!

kuba2k2, 2023-08-17:
I'll close this since it's not an issue. If you need any more help with this, feel free to reopen it whenever you like.

tech2urdoor, 2024-11-02:
I know this is an old issue but I'm having the same problem. I'm pulling my hair out trying to get this right. @tbgoose did you ever  solve this?

tbgoose, 2024-11-03:
Hi, I didn't end up fixing it. I ended up scrapping all my arlec/tuya plugs for zigbee versions.

tech2urdoor, 2024-11-03:
Thanks for replying. I had a feeling you would say that. Did you find a suitable outdoor ZigBee pm plug? I only bought this because it's weather proof. Anything you recommend?