libretiny-eu/libretiny #150: expected an identifier error

herobrine30396, 2023-07-26:
I added the library ESPAsyncWebServer-esphome which is listed as a compatible library in the docs. But I get this error in ESPAsyncWebServer.h

Here is the code:
```
#ifndef WEBSERVER_H
typedef enum {
  HTTP_GET     = 0b00000001,
  HTTP_POST    = 0b00000010,
  HTTP_DELETE  = 0b00000100,
  HTTP_PUT     = 0b00001000,
  HTTP_PATCH   = 0b00010000,
  HTTP_HEAD    = 0b00100000,
  HTTP_OPTIONS = 0b01000000,
  HTTP_ANY     = 0b01111111   <=====This is the line with the error
} WebRequestMethod;
#endif
```

Cossid, 2023-07-26:
It looks like this hasn't been updated to use the new LibreTiny naming yet.  You may need to use the external for for the time being: https://github.com/libretiny-eu/ESPAsyncWebServer

herobrine30396, 2023-07-27:
Well, even after using the fork, I'm still getting the error. While compiling, I also get this:
```
.pio/libdeps/cb3l/ESPAsyncWebServer-esphome/src/ESPAsyncWebServer.h:70:1: error: expected declaration before '}' token
   70 | } WebRequestMethod;
      | ^
.pio/libdeps/cb3l/ESPAsyncWebServer-esphome/src/ESPAsyncWebServer.h:70:3: error: 'WebRequestMethod' does not name a type
   70 | } WebRequestMethod;
      |   ^~~~~~~~~~~~~~~~
*** [.pio\build\cb3l\src\main.cpp.o] Error 1
```

herobrine30396, 2023-07-27:
Never mind. The problem was that the AsyncWebServer library was conflicting with the WebServer library. I'll close the issue now

closed 2023-07-27 by herobrine30396