libretiny-eu/libretiny #151: Failed to start task

herobrine30396, 2023-07-27:
I keep getting this error in the serial monitor, and the AsyncTcpWebserver doesn't work:
```
E [      5.316] failed to start task
E [      5.316] failed to start task
E [      5.316] failed to start task
```

herobrine30396, 2023-07-28:
It's a problem with the memory allocation of AsyncTcpWebserver. I created a fork that decreases the memory allocated to the library, and now it works!

herobrine30396, 2023-07-28:
I have another problem now. There's some issue that stops `LwIPUDP::beginMulticast` so that I get this in the serial monitor:
`E [      5.322] could not join igmp: 125`. Any help?

kuba2k2, 2023-08-02:
Yes, the task failing to start is nearly always because of lack of memory. I don't know which library are you using, but LT's fork adds another #define that decreases the stack size of the task.
About the IGMP issue, check the errno codes for code 125. From a quick look in Google it might be "too many links" or "cancelled".

Cossid, 2023-08-04:
> I have another problem now. There's some issue that stops `LwIPUDP::beginMulticast` so that I get this in the serial monitor: `E [ 5.322] could not join igmp: 125`. Any help?

I recently ran into the same issue, and the problem appears lwip wasn't fully initialized yet.  I added a delay before attempting beginMulticast (in this case, waiting until esphome's setup function had completed), and it worked fine afterwards.

herobrine30396, 2023-08-04:
Oh, okay. Will try that, but right now, I'm not able to upload anything to my board to test it. I've created a separate issue regarding that

closed 2023-08-18 by kuba2k2

Phil1pp, 2025-03-30:
> I have another problem now. There's some issue that stops `LwIPUDP::beginMulticast` so that I get this in the serial monitor: `E [ 5.322] could not join igmp: 125`. Any help?

Did you ever get that resolved?
I am facing the same issue.