libretiny-eu/libretiny #152: CRC error

herobrine30396, 2023-08-03:
I was able to flash my CB3L before. But recently, whenever I try flashing, I get this error:
`ValueError: Chip CRC value 8F407C8C does not match calculated CRC value 98F6727E
`

kuba2k2, 2023-08-04:
What method of flashing are you using? How big is the binary you're trying to flash? Does power cycling help? Did you update/downgrade versions of anything, like LT, ltchiptool? Does the error occur during flashing or at the end?

herobrine30396, 2023-08-04:
I'm using ltchiptool to flash the binary. The size probably isn't the problem as I'm getting this problem even when flashing the PinScan example. Also, the error occurs at the end of the process.

herobrine30396, 2023-08-05:
I've tried flashing both files, but they still fail with the same error. I'll post the screenshots later.

herobrine30396, 2023-08-11:
[image]

Here it is

herobrine30396, 2023-08-11:
This is the log of UART2:

```
@   b x   N   kG  R      L^  s          { 0 
V:BK7231N_1.0.1
REG:cpsr     spsr     r13      r14
SVC:000000D3          00401C1C 000033AC
IRQ:000000d2 00000010 00401e0c 27e65d3d 
FIR:000000d1 00000010 00401ffc fedd3cfa 
SYS:000000df          0040192c 00000158 
ST:00000000
J 0x10000
undefined instruction
Current regs:
r00:0xea000b48 r01:0xe59ff014 r02:0xe59ff014 r03:0xe59ff014
r04:0xe59ff014 r05:0xe59ff014 r06:0xe59ff014 r07:0xe59ff014
r08:0x00003104 r09:0x00003098 r10:0x00003114
fp :0x00003124 ip :0x00003134
sp :0x000030a8 lr :0x000030b8 pc :0xdeadbeef
SPSR:0xdeadbeef
CPSR:0x10aabbee

separate regs:
SYS:cpsr r8-r14
0x000000df
0x00000000
0x00000000
0x00000000
0x00000000
0x00200a00
0x0040aff0
0x00000158
IRQ:cpsr spsr r8-r14
0x000000d2
0xa0000053
0x00000000
0x00000000
0x00000000
0x00000000
0x00200a00
0x0040ac00
0x00005320
FIR:cpsr spsr r8-r14
0x000000d1
0x00000010
0x00000000
0x00000000
0x00000000
0x00000000
0x00000000
0x00409c10
0xfedd3cfa
ABT:cpsr spsr r8-r14
0x000000d7
0x00000010
0x00000000
0x00000000
0x00000000
0x00000000
0x00200a00
0x00409410
0x35e1afe7
UND:cpsr spsr r8-r14
0x000000db
0x00000010
0x00000000
0x00000000
0x00000000
0x00000000
0x00200a00
0x00409410
0xdffef523
SVC:cpsr spsr r8-r14
0x000000d3
0x00000010
0x00000000
0x00000000
0x00000000
0x00000000
0x00200a00
0x0040b348
0x00014deb

shutdown...

```

kuba2k2, 2023-08-11:
If it worked before, you have to work out what changed since then. Any upgrades of any software or something similar. You can also try using a different tool. Otherwise it's a hardware fault of the flash, which could occur by writing too many times.

herobrine30396, 2023-08-12:
The CEN pin stopped working for some reason, that's when I reset the chip manually by unplugging and plugging GND back in. Since then, this error has been happening. It also could be that the flash burnt out, because I did write to it too many times.

kuba2k2, 2023-08-12:
If CEN stopped working then you likely have some hardware damage. You shouldn't ever unplug GND, as current from the VCC and RX pins might flow where it shouldn't. If not using CEN, unplug VCC and not GND.

herobrine30396, 2023-08-12:
Oh! Then is my chip done for?

kuba2k2, 2023-08-12:
I don't know. It's hard to tell if you're doing something wrong (i.e. broken CEN wire caused it to malfunction) or that it's a hardware fault. All I can say is that the flashing tools haven't been modified, so nothing changed there since the time it still worked for you.

closed 2023-08-18 by kuba2k2