libretiny-eu/libretiny #154: Arduino ring buffer is not suitable for serial interface

hn, 2023-08-11:
Libretiny uses the [Arduino Ring Buffer for serial port i/o](https://github.com/kuba2k2/libretiny/blob/master/cores/realtek-ambz/arduino/libraries/Serial/SerialPrivate.h). However, the implementation is [not interrupt-safe and there are no plans to change this](https://github.com/arduino/ArduinoCore-API/issues/195).

This causes [hard to debug problems with serial i/o](https://github.com/hn/ginlong-solis/issues/4). One pain point is that the Arduino Ring buffer is very error prone when used in such a situation (it uses a redundant way to store its state). And if it breaks, it will never be good again (no self-healing). You have to (in most cases) reboot the device or restart the component in question.

As far as I can (quickly) see the ESP8266 and ESP32 implementations do not use the Arduino Ring Buffer, but implement some own 'magic' for the buffer.

I suggest replacing the Arduino Ring Buffer within the serial component with a better option (at least it should not break permanently).

hn, 2023-08-12:
Please consider the work of @KurtE for a possible solution: https://github.com/hn/ginlong-solis/issues/4#issuecomment-1673194618 and https://github.com/hn/ginlong-solis/issues/4#issuecomment-1674801822

I'm using https://github.com/hn/ginlong-solis/commit/76d81ce4d0418884d14362b2f8980c9652d40ebd as a temporary workaround.

kuba2k2, 2023-08-12:
It seems that using the SafeRingBuffer would be a viable option, since it's only a small wrapper around the RingBuffer from Arduino API. On that note, it's pretty disappointing that the API is so much different from most actual official Arduino cores (even things like Serial.printf(), which makes Arduino devs very angry when mentioned).

If the wrapper adds too much overhead by using the synchronized blocks, it could be rewritten to use FreeRTOS' mutexes or semaphores instead.

hn, 2023-08-20:
If I understand correctly, [ESP32 uses this](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-uart.c):

```
#define UART_MUTEX_LOCK()    do {} while (xSemaphoreTake(uart->lock, portMAX_DELAY) != pdPASS)
#define UART_MUTEX_UNLOCK()  xSemaphoreGive(uart->lock)
```

clsergent, 2025-07-14:
Hi,

Any update on this issue? I'm dealing with a ESPHome component relying on UART for RTL8710BN and fails occur very frequently... Increasing buffer size or patching ArduinoCoreAPI using @hn forks didn't solve the issue.

hn, 2025-07-14:
@clsergent Well, at the time I wrote the patches it definitely fixed the problem (for me).

By coincidence, I upgraded my Solis S3 stick last week and I encountered a serial error two days ago (just one time, can't say if it is a real problem or just a glitch in the matrix).

Maybe you try the `RingBufferSelfHeal` branch of the ArduinoCore-API, which uses a slightly different approach.

clsergent, 2025-07-15:
Hello @hn,
Thank you for replying so quickly. I tried a new firmware with your `RingBufferFix` and an increased buffer size (1024, seems huge but it seemed a desperate case at the moment). 
I found in the meanwhile that the RTL8710 may have OTA instability issues, so I'm not so sure that my issues were caused by UART only...
Anyway, 15 hours later and no crash, so I may be good.

My initial question still remain for the maintainers of this repository: a fix is it planned for the ring buffer issue?

mwolter805, 2026-09-09:
I can reproduce a concrete lost-update failure behind this issue on BK7231N with LibreTiny v1.13.0.

The UART receive callback calls `RingBufferN::store_char()`, while foreground code reaches `RingBufferN::read_char()` through `SerialClass::read()`. Both paths modify `_numElems`. A minimal failing schedule is:

1. The foreground reader loads `_numElems == 2` before its decrement.
2. The receive interrupt stores one or more bytes and increments `_numElems`.
3. The foreground reader stores its stale result, `1`, overwriting the interrupt's increments.

I observed the resulting coherent state on a running BK7231N device: 20 occupied ring positions while `_numElems` was zero. The bytes remained in the buffer, but `available()` reported no data. Restarting cleared the state; increasing the buffer size would not prevent this counter race.

I built a deterministic host harness around the exact LibreTiny v1.13.0 `SerialClass::read()` and ArduinoCore-API `RingBufferN` implementation. The unprotected caller reproduces the lost update. A BK7231N patch that saves and masks IRQ and FIQ for one complete read, then restores both saved masks together, passes 128 interrupt schedules at `-O0` and `-O2` with AddressSanitizer and UndefinedBehaviorSanitizer. The controls reject an unprotected read, sequential IRQ/FIQ restoration, unconditional re-enable, and a restore that does not preserve the caller's prior masks. The patched path also preserves byte order, empty and null-buffer `-1`, all 256 buffer slots, drop-new-on-full behavior, wraparound, nesting, and pending interrupts.

The patch applies to current `master` (`6514b26078a6b23c85290d7400194f92d50a5990`) without fuzz. Its tested scope is BK7231N with the current FreeRTOS Beken SDK and one foreground consumer; I have not generalized the result to the other LibreTiny families. I also did not establish a numerical worst-case interrupt-off duration.

Would you prefer a pull request that keeps the critical section in the Beken Serial implementation, or a small common hook implemented by each family so the shared `SerialClass::read()` can protect the whole read? I can prepare the narrow patch and failure-capable tests once the ownership shape is agreed.

AndrewDemsDS, 2026-10-28:
We hit this on RTL8720CF (AmebaZ2) with ESPHome on LibreTiny 1.13.0: the RX side of UART0 stopped for good while TX kept working, and only `end()`/`begin()` brought it back. It's the same lost update as described above. With UART0 in internal loopback, the count fell behind 21 times in 142558 bytes at 115200 baud without masking, and 0 times with it.

@mwolter805, on the question of where the fix should live: https://github.com/libretiny-eu/libretiny/pull/432 puts it in the common `SerialClass::read()` with `taskENTER_CRITICAL()`/`taskEXIT_CRITICAL()`, so every family is covered without a per-family hook. It builds for RTL8720C, RTL8710B, BK7231N and LN882H, and has run 20 h so far on an RTL8720CF without the application-side lock and without a stall. A run of your Beken harness against it would be welcome.