libretiny-eu/libretiny #155: HLW8012... W [      0.102] CHANGE interrupts not supported

alexxx113, 2023-08-11:


kuba2k2, 2023-08-11:
Please uninstall all versions of LibreTiny from .platformio directory. Your dev version is too old, that's why setting 1.1.0 manually works.

alexxx113, 2023-08-14:
Can I help?
logs?
tests?

kuba2k2, 2023-08-15:
Not really. Seems like CHANGE interrupts were "supported" previously, only due to a bug that produced undefined behavior. Realtek doesn't support CHANGE interrupts properly and that's why it's failing now.

When you tried 1.1.0, did it actually report correct energy measurements? Or just produced no warning?

alexxx113, 2023-08-15:
version: 1.1.0
work no problem

sensor:
  - platform: hlw8012
    model: BL0937

log

01:12:01 | [D] | [sensor:093] | 'Voltage': Sending state 225.81564 V with 1 decimals of accuracy
01:12:01 | [D] | [sensor:093] | 'Power': Sending state 25.43914 W with 1 decimals of accuracy
01:12:01 | [D] | [sensor:093] | 'Energy': Sending state 541.63861 Wh with 1 decimals of accuracy
01:12:06 | [D] | [hlw8012:082] | Got power=22.6W, voltage=226.0V
01:12:06 | [D] | [sensor:093] | 'Voltage': Sending state 226.04918 V with 1 decimals of accuracy
01:12:06 | [D] | [sensor:093] | 'Power': Sending state 22.57724 W with 1 decimals of accuracy
01:12:06 | [D] | [sensor:093] | 'Energy': Sending state 541.66998 Wh with 1 decimals of accuracy

mihsu81, 2023-08-21:
My guess is that it is related to the change introduced in 1.2.0 which adds preliminary support for RTL8720C.

https://github.com/kuba2k2/libretiny/blob/f3871388ce54bf277f2067370ee99c57b677e186/cores/realtek-amb/arduino/src/wiring_irq.c#L65

SuperXL2023, 2023-11-26:
I found a temporary solution.
Slightly modified the code in the file: < config dir >/.esphome/platformio/platforms/libretiny/cores/realtek-amb/arduino/src/wiring_irq.c

```
file: wiring_irq.c

...<other code> 
62: #if LT_RTL8720C
63:                        event = IRQ_FALL_RISE; // <- BUG  [ IRQ_FALL_RISE ] not defined, the compiler stops at this point with an error
64: #elif LT_RTL8710B
65:                        event = IRQ_RISE;
66: #else
67:                        LT_W("CHANGE interrupts not supported !!!!!!");
68: #endif
...<other code>
```
First I tried to add to line 62: `#if LT_RTL8720C || LT_RTL8710B`, but the compiler stopped here with the error `IRQ_FALL_RISE not defined`
after that, I added lines 64, 65 and everything compiled perfectly and worked.

I hope this bug will be fixed in the next version of the library, and I hope my temporary solution will help you.

appleimperio, 2024-01-29:
Having the same issue on the WR3E using the RTL8710BN in ESPHome with a MOES iR and RF system. when use the PA12 pin which is the one for the ir receiver can't compile it this si the log.
````
OTAx SELE[fffffffc]
OTA1 USE
IMG2 DATA[0x80899c0:3844:0x10005000]
IMG2 SIGN[RTKWin(10005008)]
IMG2 ENTRY[0x10005000:0x803c7f9]
System_Init1
System_Init2
I [      0.000] LibreTiny v1.4.1 on wr3e, compiled at Jan 28 2024 13:04:19, GCC 10.3.1 (-Os)

W [      0.021] CHANGE interrupts not supported
<RTL8195A>ROM:[V0.1]
FLASHRATE:4
===== Enter Image 1 ====
Img2 Sign: RTKWin, InfaStart @ 0x0803c7f9 
System_Init1
System_Init2
I [      0.000] LibreTiny v1.4.1 on wr3e, compiled at Jan 28 2024 13:04:19, GCC 10.3.1 (-Os)
W [      0.045] CHANGE interrupts not supported
io driver parameters error! file_name: rtl8711b_gpio_rom.c, line: 61
````
Any update on this issue?

Thank You.

flohav, 2024-02-04:
Hi,

I had the exact same issue, and the following solution by SuperXL2023 works perfectly. Thanks !

> I found a temporary solution. Slightly modified the code in the file: < config dir >/.esphome/platformio/platforms/libretiny/cores/realtek-amb/arduino/src/wiring_irq.c
>
> ```
> file: wiring_irq.c [...]

appleimperio, 2024-02-04:
Hi, Im using the latest version of Home assistant add-on of ESPHome and there is no trace of the .esphome folder. Thank you

RF197, 2024-03-03:
Thanks a lot, this worked for me as well (on a BX chip)!

raufaser, 2024-04-05:
@SuperXL2023 Thanks a lot. It worked.

Jaco1990, 2024-04-30:
@SuperXL2023 the solution worked for me aswell

flohav, 2024-06-08:
Do you know how to apply this fix to Home Assistant Esphome? I couldn't fine the correct file to modify...

Jaco1990, 2024-06-08:
> Do you know how to apply this fix to Home Assistant Esphome? I couldn't fine the correct file to modify...

A temporary fix was included in release v1.6.0.
At the moment the only way to run the fix in home assistant and esphome is to add the following line to your Board declaration.

rtl87xx:
  board: xxxx
  framework:
    version: dev

kuba2k2, 2024-06-08:
Or you can probably use version: 1.6.0
Let me know if it works fine.

flohav, 2024-06-09:
It works fine, I used version: dev, updated via Esphome on HA and no issue at all.

foltifekiedu, 2024-07-30:
> I found a temporary solution. Slightly modified the code in the file: < config dir >/.esphome/platformio/platforms/libretiny/cores/realtek-amb/arduino/src/wiring_irq.c
>
> ```
> file: wiring_irq.c [...]

Where can I find this .esphome folder?

I have a simple esphome folder that contains the yaml files

kuba2k2, 2025-07-16:
Yes, it would be better to have an actual solution, instead of the "workaround" which #282 is.

olivluca, 2025-07-21:
Well, no joy, the aliexpress lottery sent me a plug based on the [LN882HKI](https://www.elektroda.com/rtvforum/topic4091325.html) chip.
Is that supported by libretiny or my only option is openbeken?

Cossid, 2025-07-21:
> Well, no joy, the aliexpress lottery sent me a plug based on the [LN882HKI](https://www.elektroda.com/rtvforum/topic4091325.html) chip. Is that supported by libretiny or my only option is openbeken?

LN882HKI support was recently added to [ESPHome via LibreTiny](https://esphome.io/components/libretiny.html) & [PR](https://github.com/esphome/esphome/pull/8954) though there may be some bugs as boards get straightened out.