libretiny-eu/libretiny #160: Missing Components

DJBenson, 2023-08-25:
I've just got a generic motion night light which I've managed to flash with LT, the extracted UPK has a lot of details around the PIR sensor and I presume there is a lux sensor in there somewhere too (there's one on the board) but they are not (yet?) supported by LT.

Is there any work being done on adding such components or is the idea these could/should be supported by ESPHome itself? Would I be right in assuming if I can find the model of sensor(s) and they are supported by ESPHome natively it's just a case of adding in the relevant components? This sounds too easy and I'm guessing these components probably aren't natively supported.

This [CC](https://[raw.githubusercontent.com/tuya-cloudcutter/tuya-cloudcutter.github.io/master/devices/tuya-generic-yd001-motion-night-light.json](https://raw.githubusercontent.com/tuya-cloudcutter/tuya-cloudcutter.github.io/master/devices/tuya-generic-yd001-motion-night-light.json)) device is very similar to the one I have.

DJBenson, 2023-08-26:
Still looking for advice on the lux sensor if anybody has any ideas.

catalin2402, 2023-08-29:
Can you post the UPK extracted from the device ?

DJBenson, 2023-08-30:
I just received another device which resulted in no components being added - this is a generic PIR sensor (1.0.6). UPK for both devices below;

**Generic PIR Sensor (1.0.6):**

> {
    "basic_pin_lv": 1,
    "basic_pin_pin": 8,
    "basic_st": 1,
    "bt_lv": 0,
    "bt_pin": 7,
    "crc": 92,
    "jv": "1.0.1",
    "max_V": 3000,
    "min_V": 2400,
    "module": "CB3S",
    "net_t": 180,
    "rstcnt": 3,
    "samp_pin": 23,
    "samp_sw_lv": 1,
    "samp_sw_pin": 14,
    "samp_type": 1,
    "status_led_lv": 1,
    "status_led_pin": 26,
    "tamper_st": 0
}

**PIR Night Light (1.0.8):**
> {
    "Jsonver": "1.0.8",
    "b_lv": 1,
    "b_pin": 6,
    "brightmax": 100,
    "brightmin": 10,
    "c_lv": 1,
    "c_pin": 8,
    "cagt": 20,
    "category": "0505",
    "cdsval": 1,
    "cmod": "rgbcw",
    "colormax": 100,
    "colormaxp": 100,
    "colormin": 10,
    "colorpfun": 0,
    "crc": 54,
    "cwmaxp": 100,
    "cwtype": 0,
    "day": 200,
    "defbright": 100,
    "defcolor": "c",
    "deftemp": 100,
    "dimmod": 1,
    "dimt": 1,
    "dimval": 0,
    "dmod": 0,
    "dusk": 400,
    "evenfall": 1800,
    "evening": 2100,
    "g_lv": 1,
    "g_pin": 24,
    "gmkb": 60,
    "gmkg": 60,
    "gmkr": 80,
    "gmwb": 75,
    "gmwg": 70,
    "gmwr": 100,
    "key_lv": 0,
    "key_pin": 20,
    "keyfunc": 1,
    "ktime": 3,
    "lfunc": 1,
    "lockt": 4,
    "module": "CBU",
    "night": 2180,
    "notdisturb": 0,
    "onoffmode": 0,
    "pairt": 18,
    "pirfreq": 20000,
    "pirhduty": 2,
    "pirin_lv": 1,
    "pirin_pin": 14,
    "pirlduty": 16,
    "pirmduty": 7,
    "pirmod": 0,
    "pirrange": 0,
    "pirsense_lv": 1,
    "pirsense_pin": 9,
    "pirwarn": 0,
    "pmemory": 1,
    "preheatt": 20,
    "prodagain": 0,
    "pwmhz": 1000,
    "r_lv": 1,
    "r_pin": 26,
    "remdmode": 1,
    "rgbt": 10,
    "rstbr": 50,
    "rstcor": "c",
    "rstnum": 3,
    "rsttemp": 100,
    "sfunc": 1,
    "title20": 0,
    "trigdelay": 30,
    "trigmod": 0,
    "w_lv": 1,
    "w_pin": 7,
    "wfcfg": "spcl",
    "wfct": 3,
    "wt": 20
}

For this device I was able to get the following working;
- LED (brightness/colour/on/off)
- PIR Sensor
- Button

I could not work out the light sensor but then I can't actually see any pins defined for it but there is a sensor built into the device (probably using i2c?)

Here's the main part of my custom ESPHome config for this device;
```
esphome:
    name: $devicename
    friendly_name: $devicename_upper

bk72xx:
    board: generic-bk7231n-qfn32-tuya
    framework:
        version: latest

output:
  - platform: ledc
    id: output_red
    pin: P26
  - platform: ledc
    id: output_green
    pin: P24
  - platform: ledc
    id: output_blue
    pin: P6
  - platform: ledc
    id: output_cold
    pin: P8
  - platform: ledc
    id: output_warm
    pin: P7

light:
  - platform: rgbww
    id: light1
    name: "Light"
    color_interlock: true
    cold_white_color_temperature: 6500 K
    warm_white_color_temperature: 2700 K
    red: output_red
    green: output_green
    blue: output_blue
    cold_white: output_cold
    warm_white: output_warm

binary_sensor:
  - platform: gpio
    pin: 14
    name: "PIR Sensor"
    id: pir_sensor
    device_class: motion
    publish_initial_state: true
    filters:
      - delayed_off: !lambda return (id(ontime_number).state * 1000);      
    on_state:
      - if:
          condition:
            - binary_sensor.is_on: pir_sensor
          then:
            - light.turn_on: light1
          else:
            - light.turn_off: light1

number:
  - platform: template
    name: "Light Off Delay"
    id: ontime_number
    optimistic: true
    initial_value: 240
    min_value: 30
    max_value: 600
    step: 30
    restore_value: true
```

catalin2402, 2023-08-30:
For Generic PIR Sensor (1.0.6), there's a samp_pin: 23. That's an ADC pin, probably that's the level of the lux sensor. you can use the adc component and check if adc gets different values depending on the light conditions. For PIR Night Light (1.0.8), i don't see any adc (bk72xx have only one ADC on gpio23) maybe there's no lux sensor or it have a different way to comunicate

DJBenson, 2023-08-30:
I don't think the PIR sensor has a lux sensor, there was no value in the Tuya app whereas for the night light there was. I'll stick a sensor on that ADC pin and see what's sent. I mapped all the pins of the night light and there was one pin which was "left" but it didn't seem to return any meaningful data.

DJBenson, 2023-08-30:
The ADC on the generic PIR sensor is reading a constant 2.05v - could this be for battery monitoring?

kuba2k2, 2023-08-30:
Around 2V seems to be the value of ADC floating pin (that's what I get on floating). Also, it's too low for battery voltage. 
You're powering it by USB so maybe that's why battery voltage is floating. Can you try putting batteries in?

As for the lux sensor, do you have access to the device inside? Can you trace which pin is it connected to?

DJBenson, 2023-08-30:
Same on batteries assuming my code is correct;

```
sensor:
  - platform: adc
    pin: 23
    name: "ADC"
    update_interval: 60s
```

DJBenson, 2023-08-30:
> As for the lux sensor, do you have access to the device inside? Can you trace which pin is it connected to?

I'll take it apart later and see if I can trace the sensor with my multimeter.

kuba2k2, 2023-08-30:
> Same on batteries assuming my code is correct;
> 
> ```
> sensor:
>   - platform: adc
>     pin: 23
>     name: "ADC"
>     update_interval: 60s
> ```

I just remembered one thing: theres `samp_sw_pin` in the UPK; I think this means that ADC is disabled normally and has to be enabled by writing HIGH to that pin (14).

DJBenson, 2023-08-30:
> 

I'm struggling to work out how to do that.

catalin2402, 2023-08-30:
Using a gpio switch on that pin

DJBenson, 2023-08-30:
I tried this but it doesn't work. If I add a basic GPIO switch and toggle the switch the reported value changes to 1.47 whilst on USB but I can't work out how to permanently pull the pin high.

```
switch:
  - platform: gpio
    pin:
      number: 14
      mode:
        input: true
        pullup: true
    id: gpio_14
```

DJBenson, 2023-08-30:
The pin is reading 1.07 on batteries.

catalin2402, 2023-08-30:
@DJBenson you can keep that switch and make internal. You can also use the on_boot option and turn that switch on when the device boots

DJBenson, 2023-08-31:
Thanks - for now I've flashed the PIR sensor and a couple of contact sensors I received today with OpenBeken - I'm trying to work out how to get meaningful battery data off the devices.

In OBK it looks like there is a calculation based on the voltage of the batteries and the minimum and maximum voltages for that device which are stored in the flash as min_v and max_v. Once I work out 1) how to get the reading in millivolts from the ADC pin and 2) how to convert that into a percentage I'll switch back to LT. Ideally I want all my devices on ESPHome but there are some gaps I need to fill first.

DJBenson, 2023-08-31:
Also I just spotted that the night light device has thresholds for the lux sensor ("dusk": 400, "evenfall": 1800, "evening": 2100) - I've not had chance to take it apart again yet but pretty sure there is one.

closed 2026-02-07 by kuba2k2