libretiny-eu/libretiny #161: Can't compile in HA add-on

gabi1324, 2023-08-25:
Hi! I am using the HA add-on to install a very basic yaml file on a bk7231t. I managed to install the starting yaml after I reinstalled the add-on, but now I can't compile an updated version. 
Below is the log output from the compiling:
[logs_test2_compile.txt](https://github.com/kuba2k2/libretiny/files/12442886/logs_test2_compile.txt)
I am using the dev framework. 
What should I do to make it work?

catalin2402, 2023-08-25:
Hello! Try uninstalling and installing the addon again. it will remove all the caches and it will pull again the platforms. Your configurations will not be lost.

gabi1324, 2023-08-25:
Well, I reinstalled the add-on, but it has a similar error with the api.
[logs_test2_compile (1).txt](https://github.com/kuba2k2/libretiny/files/12443030/logs_test2_compile.1.txt)
Is this a known problem?

catalin2402, 2023-08-25:
Not something I'm aware off. On what machine are you trying to compile it ? Maybe it's a lack of RAM on the machine...

gabi1324, 2023-08-25:
I'm using a pi 4 2gb for HA and add-ons. I don't know what is causing this issue, but it compiled one time without problems.

catalin2402, 2023-08-25:
2GB is a a bit less for esphome. I had that issue one time so i switched to a 4GB one. You can try and disable some addons and try again to see if it compiles sucessfully. Also you can monitor your RAM using http://<ha_address-here>/config/hardware

gabi1324, 2023-08-25:
I never had any issues with the original Esphome add-on. This time it worked and it compiled. I think that someone should look into this issue.

catalin2402, 2023-08-25:
That was my exact problem. Sometimes it compiles, sometimes it didn't (same hw, with the official esphome addon, not the fork). It was running in a VM with 2GB ram. After i increased the ram, everytime it compiled fine. On the 2GB VM the ram usage spiked to 1.9 GB and then it crashed the compilation.

gabi1324, 2023-08-25:
I will stop some add-ons when compiling in the future. I saw a closed issue in which the solution was to uninstall and install the add-on again. I hope that this is a known issue and it will be resolved soon.
@catalin2402 thank you for your help.

daniel-dona, 2023-09-06:
It looks like this is related to an OOM event, you should check the kernel log (with dmesg or something similar) to check if there is an OOM message.

```arm-none-eabi-g++: fatal error: Killed signal terminated program cc1plus```

If that's the case, the OOM can happen sometimes only, depending on running processes and is not a code bug. It can also happen that the OOM occurs, but it kills a different process, and your compilation ends just fine.

https://www.kernel.org/doc/gorman/html/understand/understand016.html

closed 2023-09-13 by kuba2k2