libretiny-eu/libretiny #162: Feature Request : Bluetooth Proxy Support on BK7231x

rishabmehta7, 2023-08-28:
As Beken ships have bluetooth like ESP32, Having a BT Proxy would be amazing https://esphome.io/components/bluetooth_proxy.html

jeison-souza, 2023-11-24:
This will help a lot, since tons of modules are using these chips right now !

erapade, 2023-12-21:
Yes this would be amazing. For me it would mean that every lightbulb could be a potential Bluetooth proxy for less than $/€ 4. 

Anyone that knows of implementation is ongoing and how far away this is?

rishabmehta7, 2024-01-17:
@kuba2k2 Any idea if anyone is working on this?

kuba2k2, 2024-01-17:
Yes, nobody is working on this.

rishabmehta7, 2024-01-19:
> Yes, nobody is working on this.

I have no idea what to do (I am ready to learn) else I can help in anyway I can.

TuanAIoT, 2024-03-12:
Is there a way to get the bluetooth library file and compile it?
[image]

> Vâng, không ai đang làm việc này.

tiimsvk, 2024-07-08:
no news in this thread?
Today I flashed one of these modules and to my surprise I found out that it also contains bluetooth

erapade, 2024-07-08:
Did you managed to install a BT proxy as well. If so, I will probably pee I'm my pants :-)

rishabmehta7, 2024-07-08:
> Did you managed to install a BT proxy as well. If so, I will most probably pee I'm my pants :-)

Nope. LibreTiny may not be a great platform. Bugs don't get addressed, no features get worked on. If OpenBeken finds a way to implement BTProxy. I am moving to that.

kuba2k2, 2024-07-08:
See my reply in https://github.com/libretiny-eu/libretiny/issues/278#issuecomment-2213807829

erapade, 2024-07-08:
> See my reply in https://github.com/libretiny-eu/libretiny/issues/278#issuecomment-2213807829

Does it mean that the code needs to be refactored before it's worth to add enough features to support BT proxy. 

As you probably have understood, what we are aiming for is to turn every light bulb into a BT proxy

agittins, 2024-07-08:
> LibreTiny may not be a great platform. Bugs don't get addressed, no features get worked on. If OpenBeken finds a way to implement BTProxy. I am moving to that.

This is a pretty disrespectful attitude to spout in comments. Libretiny is not some corporate product that took your money and failed to deliver - it's a free and open source project that is run mostly by a single dev, and frankly it is amazing that such a thing exists!

I understand your frustration with the pace of change, but complaining about it like it's a failing is seriously uncool. It's not a failing, it's just the reality of the situation at hand. I was (and remain) super excited to discover an esphome-based implementation for beken chips, and I had spent some time in the openbeken side of things - I am so glad that Libretiny exists and has achieved as much as it has, and especially that it's easily incorporated into the standard esphome setup (for one thing, I don't have to have multiple places to store my iot codebases, awesome!).

Frankly, it's possible that the beken chips will never be able to do bluetooth proxy - the device might not have the grunt and memory to manage it (proxy is much more demanding than just sending broadcasts or supporting a single connection). I'd love to see bt proxy support happen (I have a vested interest, my Bermuda project) but the reality is it might not even be possible, so it's a significant investment for someone to dive into this to spend the time to possibly just confirm it can't be done - it's deeply unreasonable to criticise anyone for not choosing to take that gamble with their time.

People like kuba2k2 who put in serious work and offer it up to all of us for free deserve to be treated with respect, if you don't like their work pass on by, if you think it could be better, consider sponsoring them or someone else to help out, but eroding a dev's passion for something you directly benefit from by complaining about how much of their life they are donating to you is not helpful, and I'd encourage anyone to think compassionately about how they can best engage with the open source community in ways that benefit everyone and acknowledges contributions, rather than putting them down.

rishabmehta7, 2024-07-08:
> This is a pretty disrespectful attitude to spout in comments. Libretiny is not some corporate product that took your money and failed to deliver - it's a free and open source project that is run [...]

I'd agree I was maybe a tad bit over the line, but that's my passion and love for the project.
I do contribute in any way I can.
What I meant by bugs don't get fixed - here's an example https://github.com/libretiny-eu/libretiny/issues/280.

The frustration is not towards Kuba2k2 but more about even tho LT is added to ESPhome it's not getting the attention and love the main line gets. We did discuss a bit about it on the other bug linked above.

agittins, 2024-07-08:
I agree with you that it would be nice to see a way for the esphome team to chip in here - it seems like a lot of the heavy lifting to get bk chips as a platform in esphome is done and benefits esphome already. I noticed your comment on the other bug and that you have reached out to esphome to see how they can contribute, which is awesome.

I know I came on a bit strong, in part I'm reacting to a pattern of OSS devs getting burnt-out and it's clear that you care about the project.

kuba2k2, 2024-07-08:
I'm not sure I'd say I'm burnt-out with this project. I definitely will work on LibreTiny much more in the future, just don't know when. Right now I'm working on another big project of mine, which I've had for nearly 6 years now, and which needs attention after more than a year of no maintenance whatsoever. When I finish that, I see no reason not to get back to LibreTiny (unless a different particular "commercial" project comes up, I can't tell yet).

erapade, 2024-07-08:
> Did you managed to install a BT proxy as well. If so, I will most probably pee I'm my pants :-)

So my pants will stay dry at least for some months/years... 😅

Cossid, 2024-07-08:
Bluetooth at its core is apparently working in LibreTiny on Beken (at a base proof-of-concept level).  The heavy lift here is in ESPHome, as the current implementation is very ESP32 dependent.  ESPHome's ble_client and bluetooth_proxy would need to be rewritten into agnostic standardized contracts/headers, and then it would be easier to implement platform-specific versions of it.  Prior to LibreTiny, ESP32 was the only platform which had bluetooth.  Whoever takes on such a project for any new platform would need to include that refactor, unless it is addressed by ESPHome in advance.  So far no one has volunteered, and there hasn't been enough user input to increase priority.

rishabmehta7, 2024-07-08:
> Bluetooth at its core is apparently working in LibreTiny on Beken (at a base proof-of-concept level).  The heavy lift here is in ESPHome, as the current implementation is very ESP32 dependent. [...]

In that case if you know what has to be said and where, we can start putting pressure so things can move in the right direction...

fightforlife, 2026-09-06:
It seems all preconditions are merged, is there a ticket for the proxy support that we can follow?

Found: https://github.com/orgs/esphome/discussions/3758

rishabmehta7, 2026-09-06:
It was all implemented in the latest release.

closed 2026-09-06 by rishabmehta7

fightforlife, 2026-09-07:
Bad news: I flashed the latest firmware with Bluetooth Proxy to a device (cb2s) that ran fine for years with multiple firmware iterations.
This morning it is broken, no reaction wat so ever, even on hard reboot.
I hope it is just random, but I will leave this note here if someone else has this problem.

Dr-Blank, 2026-09-07:
> This morning it is broken

device bricking on bk7283 is a known issue when enabling ble support. see https://github.com/libretiny-eu/libretiny/issues/407

but yours is cb2s which is BK7231N and ble is working on many of me cb3s devices which have the same chip. yours is not booting or crashing aftr booting?

kuba2k2, 2026-09-07:
@fightforlife The best thing you can do now is to pull it apart, check UART logs and dump the flash memory as it is. Otherwise we won't know what caused the issue. If the device doesn't boot you'll have to use UART anyway.

Bl00d-B0b, 2026-09-07:
@fightforlife The overnight timing makes this look like the [#407](https://github.com/libretiny-eu/libretiny/issues/407) mechanism on BK7231N's schedule. On BK7238 the BDK bond store erases at BLE bring-up, which is why those bricks are immediate. Hardware testing on the fix branch ([#408](https://github.com/libretiny-eu/libretiny/pull/408), report by @Dr-Blank there) showed BK7231N *defers* the bond-store write instead — so a scan-only proxy runs fine until something first writes the store, and with stock v1.13.0 the unconfigured store address resolves to sector 0: bootloader erased hours after a perfectly healthy boot, dead on the next reset.

One-look confirmation from your flash dump: read `0x0-0x1000`. If the #407 mechanism hit, the bootloader sector is mostly `0xFF` (on BK7238 it was 3980 of 4096 bytes erased); an intact dump rules this out and the search moves elsewhere.

If it matches: recovery is a UART reflash of the full image (the rest of the flash is typically untouched), and [#408](https://github.com/libretiny-eu/libretiny/pull/408) is the fix — it gives every layout a dedicated `ble_bonding` partition and fails the build rather than allowing a zero-length store. It is hardware-validated on BK7238 (T1-3S, the immediate-brick case) and BK7231N (CB3S: full-flash diff clean, bond sector provably untouched). Until it merges and ships, BLE-enabled BK builds can pin the fixed platform:

```yaml
platformio_options:
  platform: https://github.com/Bl00d-B0b/libretiny.git#fix/bk72xx-ble-bonding-partition
```

With bk72xx BLE now in an ESPHome release while LibreTiny v1.13.0 still carries the unfixed store, more of these reports are likely — worth prioritizing #408 for the next LibreTiny release.

fightforlife, 2026-09-07:
I did not have the time yet to solder the UART-USB Adapter.
But from the HA Logs I could see that the device went offline 6min 50s after the successful flash. It is a BK7231N.
And by powering it directly with 3.3V, it was able to boot. So it seems the bootloader is working, and it might be a different bug. (could be hardware)

Bl00d-B0b, 2026-09-08:
Production stability datapoint: 21 BK7231x devices (BK7231N-based thermostats, light switches, LED strip controllers and power sockets, incl. T34 modules) run `bluetooth_proxy` here, all through one shared include on stock ESPHome (the BK72xx tracker ships in-tree since 2026.8). Scanning is gated on the Home Assistant API connection - the tracker boots idle and only scans while HA is actually listening, which also keeps the shared WiFi/BLE radio free otherwise:

```yaml
bluetooth_proxy:

bk72xx_ble_tracker:
  scan_parameters:
    continuous: false    # boot idle; scan only while HA is connected

api:
  on_client_connected:
    - bk72xx_ble_tracker.start_scan:
        continuous: true
  on_client_disconnected:
    # fires for EVERY departing client (log viewer, CLI) - only stop
    # scanning when the LAST client is gone
    - if:
        condition:
          not:
            api.connected:
        then:
          - bk72xx_ble_tracker.stop_scan:
```

Current numbers: all 21 show 7.6 days uptime, i.e. untouched since the last planned fleet OTA on 2026-08-31, scanning continuously the whole time (HA stays connected) with zero unplanned reboots or BLE errors. The proxy has been on this fleet since early August. Default scan timings throughout - no per-device tuning needed.

fightforlife, 2026-09-08:
It seems my issue is something else. My board is booting fine after disconnecting it from power for a few seconds.
But I can reproduce the crash after 6-7 minuted reliably. Will try to get verbose logs.