libretiny-eu/libretiny #165: FastLED support

materight, 2023-09-10:
I'm trying to build an ESPHome image for a bk7231n device that controls an addressable LED strip. So I've added this component:
```yaml
light:
  - platform: fastled_clockless
    chipset: WS2811
    pin: P17
    num_leds: 43
    rgb_order: RGB
    name: light
```
when I try to build it, I first get:
```
Internal Error: require_framework_version does not support this platform configuration
    platform: bk72xx
    framework: arduino
```
I assumed it was just a missing check, so I added a condition for `CORE.is_libretiny` [here](https://github.com/esphome/esphome/blob/e66047e07212aa214542949da68049c914901d6a/esphome/config_validation.py#L1869) and recompiled, but I get this:
```
Compiling .pioenvs/ambilight/lib005/FastLED/FastLED.cpp.o
Compiling .pioenvs/ambilight/lib005/FastLED/bitswap.cpp.o
In file included from .piolibdeps/ambilight/FastLED/led_sysdefs.h:37,
                 from .piolibdeps/ambilight/FastLED/FastLED.h:41,
                 from .piolibdeps/ambilight/FastLED/FastLED.cpp:2:
.piolibdeps/ambilight/FastLED/platforms/avr/led_sysdefs_avr.h:12:10: fatal error: avr/io.h: No such file or directory
   12 | #include <avr/io.h>
      |          ^~~~~~~~~~
compilation terminated.
Compiling .pioenvs/ambilight/lib005/FastLED/colorpalettes.cpp.o
In file included from .piolibdeps/ambilight/FastLED/led_sysdefs.h:37,
                 from .piolibdeps/ambilight/FastLED/FastLED.h:41,
                 from .piolibdeps/ambilight/FastLED/bitswap.cpp:2:
.piolibdeps/ambilight/FastLED/platforms/avr/led_sysdefs_avr.h:12:10: fatal error: avr/io.h: No such file or directory
   12 | #include <avr/io.h>
      |          ^~~~~~~~~~
compilation terminated.
Compiling .pioenvs/ambilight/lib005/FastLED/colorutils.cpp.o
In file included from .piolibdeps/ambilight/FastLED/led_sysdefs.h:37,
                 from .piolibdeps/ambilight/FastLED/FastLED.h:41,
                 from .piolibdeps/ambilight/FastLED/colorpalettes.cpp:4:
.piolibdeps/ambilight/FastLED/platforms/avr/led_sysdefs_avr.h:12:10: fatal error: avr/io.h: No such file or directory
   12 | #include <avr/io.h>
      |          ^~~~~~~~~~
compilation terminated.
Compiling .pioenvs/ambilight/lib005/FastLED/hsv2rgb.cpp.o
Compiling .pioenvs/ambilight/lib005/FastLED/lib8tion.cpp.o
In file included from .piolibdeps/ambilight/FastLED/led_sysdefs.h:37,
                 from .piolibdeps/ambilight/FastLED/FastLED.h:41,
                 from .piolibdeps/ambilight/FastLED/hsv2rgb.cpp:4:
.piolibdeps/ambilight/FastLED/platforms/avr/led_sysdefs_avr.h:12:10: fatal error: avr/io.h: No such file or directory
   12 | #include <avr/io.h>
      |          ^~~~~~~~~~
compilation terminated.
Compiling .pioenvs/ambilight/lib005/FastLED/noise.cpp.o
In file included from .piolibdeps/ambilight/FastLED/led_sysdefs.h:37,
                 from .piolibdeps/ambilight/FastLED/FastLED.h:41,
                 from .piolibdeps/ambilight/FastLED/colorutils.cpp:7:
.piolibdeps/ambilight/FastLED/platforms/avr/led_sysdefs_avr.h:12:10: fatal error: avr/io.h: No such file or directory
   12 | #include <avr/io.h>
      |          ^~~~~~~~~~
compilation terminated.
In file included from .piolibdeps/ambilight/FastLED/led_sysdefs.h:37,
                 from .piolibdeps/ambilight/FastLED/FastLED.h:41,
                 from .piolibdeps/ambilight/FastLED/lib8tion.cpp:3:
.piolibdeps/ambilight/FastLED/platforms/avr/led_sysdefs_avr.h:12:10: fatal error: avr/io.h: No such file or directory
   12 | #include <avr/io.h>
      |          ^~~~~~~~~~
compilation terminated.
*** [.pioenvs/ambilight/lib005/FastLED/FastLED.cpp.o] Error 1
*** [.pioenvs/ambilight/lib005/FastLED/bitswap.cpp.o] Error 1
*** [.pioenvs/ambilight/lib005/FastLED/colorpalettes.cpp.o] Error 1
*** [.pioenvs/ambilight/lib005/FastLED/hsv2rgb.cpp.o] Error 1
*** [.pioenvs/ambilight/lib005/FastLED/lib8tion.cpp.o] Error 1
*** [.pioenvs/ambilight/lib005/FastLED/colorutils.cpp.o] Error 1
Compiling .pioenvs/ambilight/lib005/FastLED/platforms.cpp.o
In file included from .piolibdeps/ambilight/FastLED/led_sysdefs.h:37,
                 from .piolibdeps/ambilight/FastLED/FastLED.h:41,
                 from .piolibdeps/ambilight/FastLED/noise.cpp:2:
.piolibdeps/ambilight/FastLED/platforms/avr/led_sysdefs_avr.h:12:10: fatal error: avr/io.h: No such file or directory
   12 | #include <avr/io.h>
      |          ^~~~~~~~~~
compilation terminated.
*** [.pioenvs/ambilight/lib005/FastLED/noise.cpp.o] Error 1
```
Is FastLED not supported yet or am I missing something?

kuba2k2, 2023-09-10:
It is not supported. FastLED needs a platform compatibility layer for each supported microcontroller. Apparently ESPHome uses FastLED version older than 3.3.3, which assumes every other platform than supported is AVR:
```c
#elif defined(ESP32)
#include "platforms/esp/32/led_sysdefs_esp32.h"
#else
// AVR platforms
#include "platforms/avr/led_sysdefs_avr.h"
#endif
```

Normally, you should get an error stating that this platform is not supported.

materight, 2023-09-10:
Ah thanks for the clarification. So I guess adding support for it depends on the FastLED team?

kuba2k2, 2023-09-10:
That, and supporting the appropriate protocols in LibreTiny.