libretiny-eu/libretiny #167: WR3E & HLW8012 Bootloop

wernerhp, 2023-09-10:
- LibreTiny 1.1.0 works

Setting `current_resistor` on 1.1.0 causes a boot loop.

<details>
<summary>cbi-asc.yaml (Click to expand)</summary>

```yaml
substitutions:
  name: cbi-asc-xxxxxx
  friendly_name: "Astute Smart Controller affee9"
  comment: "The Astute Smart Controller is an energy monitoring, scheduling and control switch with load management and automation capability."
  project_name: "wernerhp.esphome/cbi-asi"
  project_version: "1.0.0"

  button_1_pin: PA22
  relay_1_pin: PA12
  led_i_blue_pin: PA05
  led_i_red_pin: PA23
  sel_i_pin: PA19
  cf_pin: PA15
  cf1_pin: PA14
  voltage_divider: "1742"

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  comment: ${comment}
  # name_add_mac_suffix: true
  
  project:
    name: ${project_name}
    version: ${project_version}

#rtl87xx:
libretiny:
  board: wr3e
  framework:
    version: 1.1.0

# Enable logging
logger:
  level: DEBUG

# Enable Home Assistant API
api:

ota:

wifi:
  networks:
    - ssid: !secret wifi_ssid
      password: !secret wifi_password
  ap:
    ssid: ${name}
    password: "12345678"

captive_portal:

web_server:
  port: 80

binary_sensor:
  - platform: gpio
    pin: ${button_1_pin}
    name: "Power Button"
    filters:
      - delayed_on: 10ms
      - delayed_off: 10ms
    on_press:
      - switch.toggle: relay_1
    internal: true

switch:
  - platform: gpio
    pin: ${relay_1_pin}
    name: "Relay"
    id: relay_1 
    on_turn_on:
      - light.turn_on: light_1
    on_turn_off:
      - light.turn_off: light_1

light:
  - platform: binary
    id: light_1
    output: led_1

output:
  - platform: gpio
    pin: ${led_i_red_pin}
    inverted: true
    id: led_1

status_led:
    pin:
      number: ${led_i_blue_pin}
      inverted: true

button:   
  - platform: restart
    name: "Restart"

sensor:
  - platform: hlw8012
    model: BL0937
    sel_pin:
      number: ${sel_i_pin}
      inverted: true
    cf_pin: 
      number: ${cf_pin}
      inverted: true
    cf1_pin: 
      number: ${cf1_pin}
      inverted: true
    voltage_divider: ${voltage_divider}
    # current_resistor: 0.001
    change_mode_every: 3
    update_interval: 3s
    # Current sensor
    current:
      name: "Current"
      unit_of_measurement: A
      accuracy_decimals: 3      
    # Voltage sensor
    voltage:
      name: "Voltage"
      unit_of_measurement: V
      accuracy_decimals: 1
    # Power sensor
    power:
      id: power
      name: "Power"
      unit_of_measurement: W
      accuracy_decimals: 0
    energy:
      name: "Energy"

  - platform: uptime
    name: "Uptime"

  - platform: wifi_signal
    name: "WiFi Signal"

text_sensor:
  - platform: version
    name: "ESPHome Version"

  - platform: wifi_info
    ip_address:
      name: 'IP Address'
      icon: mdi:wifi
    ssid:
      name: 'SSID'
      icon: mdi:wifi-strength-2
    bssid:
      name: 'BSSID'
      icon: mdi:wifi-strength-2
```

</details>

- LibreTiny 1.2.0 bootloops
<details>
<summary>Log (Click to expand)</summary>

```
<RTL8195A>ROM:[V0.1]
FLASHRATE:4
===== Enter Image 1 ====
Img2 Sign: RTKWin, InfaStart @ 0x08041571 
System_Init1
System_Init2
I [      0.000] LibreTiny v1.2.0 on wr3e, compiled at Sep 10 2023 22:38:44, GCC 10.3.1 (-Os)
[I][logger:302]: Log initialized
[C][status_led:014]: Setting up Status LED...
[C][ota:477]: There have been 3 suspected unsuccessful boot attempts.
[D][lt.preferences:104]: Saving 1 preferences to flash...
[D][lt.preferences:132]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[I][app:029]: Running through setup()...
<RTL8195A>ROM:[V0.1]
FLASHRATE:4
===== Enter Image 1 ====
Img2 Sign: RTKWin, InfaStart @ 0x08041571 
System_Init1
System_Init2
I [      0.000] LibreTiny v1.2.0 on wr3e, compiled at Sep 10 2023 22:38:44, GCC 10.3.1 (-Os)
[I][logger:302]: Log initialized
[C][status_led:014]: Setting up Status LED...
<RTL8195A>ROM:[V0.1]
FLASHRATE:4
===== Enter Image 1 ====
Img2 Sign: RTKWin, InfaStart @ 0x08041571 
System_Init1
System_Init2
I [      0.000] LibreTiny v1.2.0 on wr3e, compiled at Sep 10 2023 22:38:44, GCC 10.3.1 (-Os)
[I][logger:302]: Log initialized
[C][status_led:014]: Setting up Status LED...
[C][ota:477]: There have been 4 suspected unsuccessful boot attempts.
[D][lt.preferences:104]: Saving 1 preferences to flash...
[D][lt.preferences:132]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[I][app:029]: Running through setup()...
```
</details>

- LibreTiny 1.3.0 boot loops.

kuba2k2, 2023-09-11:
How are you powering the device? Please check if you're giving it enough power. Usually a weak power supply is the cause of bootloops, especially on Realtek chips.

wernerhp, 2023-09-11:
I got a boot loop after flashing 1.1.0 with 2023.9.0-dev OTA.

[image]
[image]

Pulled it apart and through trial-and error found that LT 1.1.0 with 2023.7.0-dev seemed to work.
Powered using ESP Flasher Rev 6.1 when flashing.

[image]
[image]

wernerhp, 2023-10-01:
[image]

Power and Energy are not reflecting even though both Current and Voltage are working

wernerhp, 2024-02-09:
Still getting this bootloop with **ESPHome 2023.12.9** and **libretiney 1.4.1**

```
rtl87xx:
  board: wr3e
  framework:
    version: recommended
```

```
binary_sensor:
  - platform: gpio
    pin: ${button_1_pin}
    name: "Power Button"
    filters:
      - delayed_on: 10ms
      - delayed_off: 10ms
    on_press:
      - switch.toggle: relay_1
    internal: true

switch:
  - platform: gpio
    pin: ${relay_1_pin}
    name: "Relay"
    id: relay_1 
    on_turn_on:
      - light.turn_on: light_1
    on_turn_off:
      - light.turn_off: light_1

light:
  - platform: binary
    id: light_1
    output: led_1

output:
  - platform: gpio
    pin: ${led_i_red_pin}
    inverted: true
    id: led_1

status_led:
    pin:
      number: ${led_i_blue_pin}
      inverted: true

button:   
  - platform: restart
    name: "Restart"

sensor:
  - platform: hlw8012
    model: BL0937
    sel_pin:
      number: ${sel_i_pin}
      inverted: true
    cf_pin: 
      number: ${cf_pin}
      inverted: true
    cf1_pin: 
      number: ${cf1_pin}
      inverted: true
    change_mode_every: 3
    update_interval: 1s
    voltage_divider: ${voltage_divider}
    voltage:
      name: "Voltage"
    # current_resistor: 0.001 ohm
    current:
      name: "Current"
    power:
      name: "Power"
      id: "power"
    energy:
      name: "Energy"

  - platform: uptime
    name: "Uptime"

  - platform: wifi_signal
    name: "WiFi Signal"

text_sensor:
  - platform: template
    name: "Firmware version"
    icon: "mdi:label-outline"
    entity_category: diagnostic
    lambda: |-
      return {"${project_version}"};
    update_interval: 6h

  - platform: version
    name: "ESPHome Version"

  - platform: wifi_info
    mac_address:
      name: "Mac Address"
      icon: mdi:wifi
    ip_address:
      name: "IP Address"
      icon: mdi:wifi
    ssid:
      name: "SSID"
      icon: mdi:wifi-strength-2
    bssid:
      name: "BSSID"
      icon: mdi:wifi-strength-2

# Enable time component to reset energy at midnight
time:
  - platform: homeassistant
    id: homeassistant_time
```

RF197, 2024-02-11:
With v1.4.1 and a WR2 board config I get the error message : CHANGE interrupt not supported, and then a bootloop.

Which is due to the pulse counter being used, and pin change interrupts not being supported on the RTL chips. Could this maybe be bypassed by attaching the interrupt function to rising AND falling edge interrupts? 

I don't see the same warning in your logfiles but maybe it's related?