libretiny-eu/libretiny #168: Add new device YAML uses deprecated LibreTiny component

gb0101010101, 2023-09-13:
The LibreTiny component was split between supported chip families but the "New Device" button still generates the old format YAML.

EspHome Libretiny v2023.9.0-dev installed within Home Assistant as Addon.

Reproduce: Click on New Device. Select any LibreTiny board. View YAML config. Shows:

```
libretiny:
  board: cb3s
  framework:
    version: dev
```
When it should be:
```
bk72xx:
  board: cb3s
  framework:
    version: dev
```

kuba2k2, 2023-09-13:
How did you install the add-on? (and where from exactly)
LibreTiny has been merged into ESPHome already, and is available in the official dev channel of ESPHome add-on. That version doesn't have LibreTiny in the wizard at all (decision of the team to postpone modifying the wizard).

catalin2402, 2023-09-13:
Also available in the beta channel of ESPHome :)

gb0101010101, 2023-10-19:
> How did you install the add-on? (and where from exactly)
LibreTiny ESPHome was installed as Addon within HAOS virtual machine. 

> LibreTiny has been merged into ESPHome already...
OK, great to hear that. Will migrate configs.

gb0101010101, 2023-10-22:
Migrated configs from LibreTiny ESPhome and everything works. Uninstalled LibreTiny ESPhome.

Thanks for the great work on this. Its a real game changer that the Beken chips are supported.

closed 2023-10-22 by gb0101010101