libretiny-eu/libretiny #170: Compiling failed on PlatformIO

VinceW31, 2023-09-14:
WB3S 	BK7231T
Using a very simple test script I tried to compile it in PlatformIO and got the following error at the end of compilation:
Error:  BOARD  D:/PlatformIO/Projects/Tuya does not exist.
My project and folder name was "Tuya Testing" in D:/PlatformIO/Projects
I changed my project folder name to just plain "Tuya" and started again and this time it all compiled fine with no errors.
I suspect a glitch somewhere looking for the folder name.

kuba2k2, 2023-09-15:
Please post a full log with the error.

VinceW31, 2023-09-15:
Here is the full Build LOG with the ERROR at the end:

 *  Executing task: C:\Users\vince\.platformio\penv\Scripts\platformio.exe run 

Processing wb3s (platform: libretiny; board: wb3s; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
Warning! Non-Git installations are NOT SUPPORTED.
CONFIGURATION: https://docs.platformio.org/page/boards/libretiny/wb3s.html
PLATFORM: LibreTiny (1.4.0) > WB3S Wi-Fi Module
HARDWARE: BK7231T 120MHz, 256KB RAM, 1.03MB Flash
DEBUG: Current (openocd) External (custom, openocd)
PACKAGES:
 - framework-arduino-api @ 2022.8.24+sha.237b10a 
 - framework-beken-bdk @ 0.0.0+v2021.06.07.sha.6491b8c
 - library-flashdb @ 1.2.0+sha.d5c892f
 - library-freertos @ 9.0.0+sha.95cc959
 - library-freertos-port @ 2023.5.23+sha.a917d93
 - library-lwip @ 2.1.3-bdk+sha.4ee4d34
 - library-printf @ 6.1.0+sha.28a79bd 
 - library-uf2ota @ 5.0.0+sha.f955412
 - toolchain-gccarmnoneeabi @ 1.100301.220327 (10.3.1)
PLATFORM VERSIONS:
 - libretiny @ 1.4.0
 - ltchiptool @ 4.5.1
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 2 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio\build\wb3s\src\main.cpp.o
Compiling .pio\build\wb3s\board_wb3s\wb3s.c.o
Archiving .pio\build\wb3s\libboard_wb3s.a
Compiling .pio\build\wb3s\arduino_api\api\Common.cpp.o
Compiling .pio\build\wb3s\arduino_api\api\IPAddress.cpp.o
Compiling .pio\build\wb3s\arduino_api\api\PluggableUSB.cpp.o
Compiling .pio\build\wb3s\arduino_api\api\Print.cpp.o
Compiling .pio\build\wb3s\arduino_api\api\Stream.cpp.o
Compiling .pio\build\wb3s\arduino_api\api\String.cpp.o
Compiling .pio\build\wb3s\core_bk72xx_arduino_libraries\Serial\Serial.cpp.o
Archiving .pio\build\wb3s\libarduino_api.a
Compiling .pio\build\wb3s\core_bk72xx_arduino_libraries\WiFi\WiFi.cpp.o
Compiling .pio\build\wb3s\core_bk72xx_arduino_libraries\WiFi\WiFiAP.cpp.o
Compiling .pio\build\wb3s\core_bk72xx_arduino_libraries\WiFi\WiFiEvents.cpp.o
Compiling .pio\build\wb3s\core_bk72xx_arduino_libraries\WiFi\WiFiGeneric.cpp.o
Compiling .pio\build\wb3s\core_bk72xx_arduino_libraries\WiFi\WiFiSTA.cpp.o
Compiling .pio\build\wb3s\core_bk72xx_arduino_libraries\WiFi\WiFiScan.cpp.o
Archiving .pio\build\wb3s\libcore_bk72xx_arduino_fixups.a
Archiving .pio\build\wb3s\libcore_bk72xx_arduino_libraries.a
Compiling .pio\build\wb3s\core_bk72xx_arduino\main.cpp.o
Compiling .pio\build\wb3s\core_bk72xx_arduino\wiring.c.o
Compiling .pio\build\wb3s\core_bk72xx_arduino\wiring_analog.c.o
Compiling .pio\build\wb3s\core_bk72xx_arduino\wiring_digital.c.o
Compiling .pio\build\wb3s\core_bk72xx_arduino\wiring_irq.c.o
Compiling .pio\build\wb3s\core_common_arduino_libraries\api\Serial\Serial.cpp.o
Archiving .pio\build\wb3s\libcore_bk72xx_arduino.a
Compiling .pio\build\wb3s\core_common_arduino_libraries\api\SoftwareSerial\SoftwareSerial.cpp.o
Compiling .pio\build\wb3s\core_common_arduino_libraries\api\WiFi\WiFi.cpp.o
Compiling .pio\build\wb3s\core_common_arduino_libraries\api\WiFi\WiFiAP.cpp.o
Compiling .pio\build\wb3s\core_common_arduino_libraries\api\WiFi\WiFiEvents.cpp.o
Compiling .pio\build\wb3s\core_common_arduino_libraries\api\WiFi\WiFiGeneric.cpp.o
Compiling .pio\build\wb3s\core_common_arduino_libraries\api\WiFi\WiFiSTA.cpp.o
Compiling .pio\build\wb3s\core_common_arduino_libraries\api\WiFi\WiFiScan.cpp.o
Compiling .pio\build\wb3s\core_common_arduino_libraries\common\FS\FS.cpp.o
Compiling .pio\build\wb3s\core_common_arduino_libraries\common\IPv6Address\IPv6Address.cpp.o
Compiling .pio\build\wb3s\core_common_arduino_libraries\common\MD5\MD5MbedTLSImpl.cpp.o
Compiling .pio\build\wb3s\core_common_arduino_libraries\common\Update\Update.cpp.o
Compiling .pio\build\wb3s\core_common_arduino_libraries\common\Update\UpdateUtil.cpp.o
Compiling .pio\build\wb3s\core_common_arduino_libraries\common\WiFiClient\LwIPClient.cpp.o
Compiling .pio\build\wb3s\core_common_arduino_libraries\common\WiFiClient\LwIPRxBuffer.cpp.o
[... 323 lines cut ...]
Compiling .pio\build\wb3s\bdk_app\http\utils_httpc.c.o
Compiling .pio\build\wb3s\bdk_app\http\utils_net.c.o
Compiling .pio\build\wb3s\bdk_app\http\utils_timer.c.o
Compiling .pio\build\wb3s\bdk_app\net_work\video_demo_main.c.o
Compiling .pio\build\wb3s\bdk_app\net_work\video_demo_softap.c.o
Compiling .pio\build\wb3s\bdk_app\net_work\video_demo_station.c.o
Compiling .pio\build\wb3s\bdk_app\standalone-ap\sa_ap.c.o
Compiling .pio\build\wb3s\bdk_app\standalone-station\sa_station.c.o
Compiling .pio\build\wb3s\bdk_app\video_work\video_buffer.c.o
Compiling .pio\build\wb3s\bdk_app\video_work\video_transfer_tcp.c.o
Compiling .pio\build\wb3s\bdk_app\video_work\video_transfer_udp.c.o
Compiling .pio\build\wb3s\bdk_core\app\app.c.o
Archiving .pio\build\wb3s\libbdk_app.a
Compiling .pio\build\wb3s\bdk_core\app\config\param_config.c.o
Compiling .pio\build\wb3s\bdk_core\driver\entry\boot_vectors.S.o
Compiling .pio\build\wb3s\bdk_core\func\wlan_ui\wlan_ui.c.o
Compiling .pio\build\wb3s\bdk_boot\boot_handlers_bk7231u.S.o
Archiving .pio\build\wb3s\libbdk_boot.a
Archiving .pio\build\wb3s\libbdk_core.a
Linking .pio\build\wb3s\raw_firmware.elf
Usage: python -m ltchiptool link2bin [OPTIONS] BOARD OTA1 OTA2 [ARGS]...
Try 'python -m ltchiptool link2bin --help' for help.

Error: Invalid value for 'BOARD': Board D:\PlatformIO\Projects\Tuya does not exist
*** [.pio\build\wb3s\raw_firmware.elf] Error 2

VinceW31, 2023-09-16:
I put an underscore in the project folder name to replace the space and it all compiled ok, so it doesn't like the space in a fil/project folder name. Changed "Tuya Testing" to "Tuya_Testing"

closed 2026-02-07 by kuba2k2