libretiny-eu/libretiny #171: Tenda Beli SP3 smart plug write error.

gaz99gh, 2023-09-15:
I have an RTL8710BX based Tenda Beli SP3 smart plug that is not Tuya. When trying to upload the Esphome uf2 file generated from the 2023.9.0b1 beta, I get this error:

`E: ValueError: Failed to write to 0x800B000`
 
I have successfully read the original firmware and the file is 2M and a look at the file shows this:

`This flash type is not supported!
 CHIPID[%08x]`

The flash chip is a H25S80BG.

Any ideas on how to proceed ?

[tenda-stock2.zip](https://github.com/libretiny-eu/libretiny/files/12618419/tenda-stock2.zip)
[image]

kuba2k2, 2023-09-15:
The message you found in the file probably doesn't mean anything.

The writing error, however, might be caused by several problems. How are you powering the device for flashing?

I just tried flashing your firmware dump on a RTL8710BX, and it doesn't start up:
```
<RTL8195A>ROM:[V0.1]
FLASHRATE:4
BOOT TYPE:0 XTAL:40000000
IMG1 DATA[1128:10002000]
IMG1 ENTRY[800053f:100021ef]
IMG1 ENTER
CHIPID[000000f6]
read_mode idx:3, flash_speed idx:4
calibration_result:[1:29:15][1:1d] 
calibration_result:[2:19:11][1:13] 
calibration_result:[3:0:0][ff:ff] 
calibration_ok:[1:29:15] 
FLASH CALIB[NEW OK]
OTA2 ADDR[8080000]
OTAx SELE[fffffffc]
OTA1 USE
IMG2 DATA[0x8069530:-1:0xffffffff]
IMG2 ADDR Invalid
```

gaz99gh, 2023-09-15:
I have tried both the modified FTDI TTL-232R cable I normally use and my bench PSU. There is no obvious CEN pin so I'm using the power reset while grounding Tx.

The programming test points on the module don't say TX2/RX2 but they are all there is.

The back of the module looks like this:
[image]

kuba2k2, 2023-09-15:
Try using the rtltool2 flasher from here:
https://github.com/hn/ginlong-solis/issues/2#issuecomment-1625731577
Make sure to use the correct OTA2 address, which is 0x80000 for your board (I think).

Cossid, 2023-09-15:
Yeah, this is becoming a theme with realtek, where the last letter of the chip, which is supposed to indicate flash size, doesn't actually match the size of the external flash.

gaz99gh, 2023-09-16:
OK, I have reached the limit of my linux command line knowledge !!

I have compiled a new esphome device with the `generic-rtl8710bn-2mb-468k` board type and saved the file as tenda-test22468.u2f

I have managed to read from the device with rtltool but I get an error when trying to write. 
`rtltool: error: unrecognized arguments: /home/gazza/rtltool/tenda-test22468.u2f`

The command I am using is:
`sudo python /home/gazza/rtltool/rtltool.py -p /dev/ttyUSB0 wf 0x8000000 /home/gazza/rtltool/tenda-test22468.u2f
`
Also what do the arguments mean, I used 0x8000000 as that was what was in the the other issue that was linked but 0x800000 was mentioned above.

kuba2k2, 2023-09-16:
The comment I linked uses a different command - please see this comment specifically, not the whole issue. You should use rtltool2, not rtltool.

Also, the file is uf2 not u2f.
Also, you shouldn't be using uf2 for flashing with rtltool2 - use the binary files with offsets, as said in the issue comment above.

0x8000000, 0x800000 and 0x80000 are all different, you have to be careful with the number of zeroes.

gaz99gh, 2023-09-16:
Where do I get rtltool2.py I could only find rtltool.py ?

kuba2k2, 2023-09-16:
Here:
https://github.com/pvvx/RTL0B_SDK/blob/master/mkb/rtltool.py
This requires Python 2.
As in the linked comment, run:
```
python2 ./rtltool2.py -p /dev/ttyUSB0 wf 0xb000 image_0x00B000.ota1.bin
python2 ./rtltool2.py -p /dev/ttyUSB0 wf 0x80000 image_0x080000.ota2.bin
```
Get the image_* files from the build directory of your firmware.

gaz99gh, 2023-09-16:
OK, that is the file I was using and I have confirmed that I am using python2 and I even managed to get the missing python2-serial stuff installed.

Also, when I do a manual download from Esphome 23.9.0b1 I now get options to download the uf2 and the separate binaries.

So I have loaded files and it is up and running, the minimal install is using 81% of the flash, the device is a pretty basic smart plug with no power monitoring so I might work. Will I break anything if it try the generic-rtl8710bn-2mb-788k version ??

Thanks for the help.

`INFO ESPHome 2023.9.0b1
INFO Reading configuration /config/esphome/tenda-test22468.yaml...
INFO Starting log output from tenda-test22468.local using esphome API
INFO Successfully connected to tenda-test22468.local
[21:10:00][I][app:102]: ESPHome version 2023.9.0b1 compiled on Sep 15 2023, 13:16:12
[21:10:00][C][wifi:546]: WiFi:
[21:10:00][C][wifi:382]:   Local MAC: 50:2B:73:F6:09:2C
[21:10:00][C][wifi:383]:   SSID: 'Coffee1'[redacted]
[21:10:00][C][wifi:384]:   IP Address: 10.0.0.146
[21:10:00][C][wifi:385]:   BSSID: 80:2A:A8:D1:CF:D5[redacted]
[21:10:00][C][wifi:387]:   Hostname: 'tenda-test22468'
[21:10:00][C][wifi:389]:   Signal strength: -31 dB ▂▄▆█
[21:10:00][C][wifi:393]:   Channel: 1
[21:10:00][C][wifi:394]:   Subnet: 255.255.0.0
[21:10:00][C][wifi:395]:   Gateway: 10.0.0.1
[21:10:00][C][wifi:396]:   DNS1: 10.0.0.1
[21:10:00][C][wifi:397]:   DNS2: 10.0.0.1
[21:10:00][C][logger:357]: Logger:
[21:10:00][C][logger:358]:   Level: DEBUG
[21:10:00][C][logger:359]:   Log Baud Rate: 115200
[21:10:00][C][logger:361]:   Hardware UART: UART2
[21:10:00][C][mdns:115]: mDNS:
[21:10:00][C][mdns:116]:   Hostname: tenda-test22468
[21:10:00][C][ota:097]: Over-The-Air Updates:
[21:10:00][C][ota:098]:   Address: tenda-test22468.local:8892
[21:10:00][C][ota:101]:   Using Password.
[21:10:00][C][api:138]: API Server:
[21:10:00][C][api:139]:   Address: tenda-test22468.local:6053
[21:10:00][C][api:143]:   Using noise encryption: NO
[21:10:00][C][lt.component:013]: LibreTiny:
[21:10:00][C][lt.component:014]:   Version: v1.4.0 on generic-rtl8710bn-2mb-468k, compiled at Sep 15 2023 13:15:57, GCC 10.3.1 (-Os)
[21:10:00][C][lt.component:015]:   Loglevel: 3`

`INFO ESPHome 2023.9.0b1
INFO Reading configuration /config/esphome/tenda-test22468.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing tenda-test22468 (board: generic-rtl8710bn-2mb-468k; framework: arduino; platform: libretiny)
--------------------------------------------------------------------------------
Warning! Non-Git installations are NOT SUPPORTED.
HARDWARE: RTL8710BN 125MHz, 256KB RAM, 468KB Flash
 - framework-arduino-api @ 2022.8.24+sha.237b10a 
 - framework-realtek-amb1 @ 0.0.0+v2022.06.21.sha.c4e44ef 
 - library-flashdb @ 1.2.0+sha.d5c892f 
 - library-freertos @ 8.1.2+sha.776ae6c 
 - library-freertos-port @ 2023.5.23+sha.a917d93 
 - library-lwip @ 2.1.3-amb1+sha.6297b80 
 - library-printf @ 6.1.0+sha.28a79bd 
 - library-uf2ota @ 5.0.0+sha.f955412 
PLATFORM VERSIONS:
 - libretiny @ 1.4.0
 - ltchiptool @ 4.5.1
CUSTOM OPTIONS:
 - fw_name = esphome
 - fw_version = 2023.9.0b1
No dependencies

Archiving .pioenvs/tenda-test22468/libboot_all.a
Archiving .pioenvs/tenda-test22468/libambz_sdk.a
Linking .pioenvs/tenda-test22468/raw_firmware.elf
|-- Image 1: raw_firmware.ota1.elf
|-- Image 2: raw_firmware.ota2.elf
|   |-- image_ota1.0x00B000.bin
|   |   |-- image_ota1.ram_2.r.bin
|   |   |-- image_ota1.xip_image2.bin
|   |   |-- image_ota1.rdp.bin
|   |-- image_ota2.0x080000.bin
|   |   |-- image_ota2.ram_2.r.bin
|   |   |-- image_ota2.xip_image2.bin
|   |   |-- image_ota2.rdp.bin
RAM:   [==        ]  20.1% (used 52592 bytes from 262144 bytes)
Flash: [========  ]  82.1% (used 393356 bytes from 479232 bytes)
Building UF2 OTA image
|-- esphome_2023.9.0b1_generic-rtl8710bn-2mb-468k_rtl8710bn_lt1.4.0.uf2
|-- firmware.uf2
|-- firmware.bin
========================= [SUCCESS] Took 58.93 seconds =========================`

kuba2k2, 2023-09-16:
It's a small flash chip, so 81% usage is normal. Don't try the 788k version, it's too large for this device. Also, you'll break OTA that way.

Do I correctly assume that you have flashed ESPHome successfully and it is running on the device?

gaz99gh, 2023-09-16:
Yes I Esphome running on it with just the default settings and I have been able to do an OTA update.

gaz99gh, 2023-09-17:
Is there somewhere to record working hardware and the ESPHome code for them ??

kuba2k2, 2023-09-17:
We do have a special channel on our Discord server, but I don't really know of any other place (apart from sites dedicated to ESP devices).

gaz99gh, 2023-09-19:
I think this is complete.

kuba2k2, 2023-09-19:
I understand that using ltchiptool didn't work, but rtltool2 did? If so, I'll keep this issue open to fix this error.

gaz99gh, 2023-09-19:
OK, in ltchiptool I was using the defaults but I also had the firmware extension wrong, u2f instead of uf2.

dennis-kachila, 2025-03-08:
> I have an RTL8710BX based Tenda Beli SP3 smart plug that is not Tuya. When trying to upload the Esphome uf2 file generated from the 2023.9.0b1 beta, I get this error:
>
> `E: ValueError: Failed to [...]