libretiny-eu/libretiny #174: Stuck at 'Checking OTA index' for RTL8710BX

ferbulous, 2023-09-22:
Hi, I have this rl8710bx chip that's stuck with this message when flashing with ltchiptool.

[image]

```Firmware: rtl8710bn-2mb-468k```

I have several of this device, and just this one having this issue, the rest flashed just fine (except for ```E: ValueError: Failed to write to 0x800B000``` error that keeps popping up halfway so I just had to keep reflash them until that error's gone)

Any reason for this?

closed 2023-09-23 by ferbulous

kuba2k2, 2023-09-23:
So, what was the cause?

ferbulous, 2023-09-23:
> So, what was the cause?

I’m not quite sure, switched to 3.3v pin of my banana pi from asm1117 and i can start flashing it. Still had to repeat like 8 times because of the ``` ValueError: Failed to write to 0x800B000 ``` error