libretiny-eu/libretiny #176: Error when compiling climate platform: heatpumpir

zain1144, 2023-09-28:
I get this error every time 

https://media.discordapp.net/attachments/563439987262095361/1156671046645186630/image.png?ex=6515d15b&is=65147fdb&hm=c671b907a0fa3f88dad804f9ae167d230b3fdbeeeaad5f77521a38efe581e68b&=&width=2334&height=1352

`esphome:
  name: rf

bk72xx:
  board: generic-bk7231n-qfn32-tuya

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 192.168.0.124
    gateway: 192.168.0.1
    subnet: 255.255.255.0

logger:

api:

ota:

web_server:
  port: 80

        

climate:
  - platform: heatpumpir
    name: "AC"
    protocol: gree
    horizontal_default: auto
    vertical_default: auto
    min_temperature: 10
    max_temperature: 40

remote_transmitter:
  pin: P7
  carrier_duty_percent: 50%
`

catalin2402, 2023-09-28:
The heatpumpir platform uses external libraries

X-Ryl669, 2023-10-21:
The real question is why `CORE.is_esp8266` or `CORE.is_esp32` is true.

I've patched the libraries in #163 and it's working fine without this IRremoteESP8266 library.

kuba2k2, 2023-10-21:
> The real question is why CORE.is_esp8266 or CORE.is_esp32 is true.

It's not. Only `CORE.is_libretiny` and `CORE.is_bk72xx`/`CORE.is_rtl87xx` is true.

X-Ryl669, 2023-10-21:
I agree, the code in ESPHome/core/__init.py is like this:
```py
    @property
    def is_esp32(self):
        return self.target_platform == PLATFORM_ESP32

    @property
    def is_rp2040(self):
        return self.target_platform == PLATFORM_RP2040

    @property
    def is_bk72xx(self):
        return self.target_platform == PLATFORM_BK72XX

    @property
    def is_rtl87xx(self):
        return self.target_platform == PLATFORM_RTL87XX

    @property
    def is_libretiny(self):
        return self.is_bk72xx or self.is_rtl87xx
```
So it's theorically impossible it would pass this test, yet the library is added and being compiled (and it's failing).

X-Ryl669, 2023-10-21:
The only other place it's referenced in ESPHome is platformio.ini and I also don't see why it would be including in Libretiny.

kuba2k2, 2023-10-21:
Well, that's odd since it doesn't include that library on my machine. Just tested.

It errors out in `HeatpumpIR/IRSenderPWM.cpp`, because it's not adapted for LibreTiny, and tries to use AVR registers.

X-Ryl669, 2023-10-21:
Here's my yml that's having the same error as the OP:
```yaml
substitutions: 
  # Beware, it's a CBU for the 2 old devices (West and South) and not CB3S. 
  # The pinout is different, so adjust accordingly
  # ## usage of secrets-file ## -> comment in ###### Change ME ######
  device_name: "whatever" # Wird im Blueprint benötigt!
  wifi_ssid: !secret wifi_ssid # add in your esphome secrets file.
  wifi_password: !secret wifi_password # add in your esphome secrets file. -> per default this is also used for ota_password and web_password
  ota_password: !secret ota_password # add in your esphome secrets file. - manual change in code required to activate
  web_password: !secret wifi_password # add in your esphome secrets file. - manual change in code required to activate
  api_password: !secret api_key # add in your esphome secrets file. - manual change in code required to activate  
 
  # ## static ip config ##
  ip: "192.168.0.169"
  gw: "192.168.0.254"
  subnet: "255.255.255.0"
  dns: "192.168.0.254"

esphome:
  name: irremotesouth
  friendly_name: ${device_name}

bk72xx:
  board: cbu
  framework:
    version: dev

# Enable logging
logger:

# Enable Home Assistant API
web_server:
  port: 80
  auth:
    username: admin
    password: ${web_password}

# Enable Home Assistant API
api:
  encryption:
    key: ${api_password}

ota:
  password: ${ota_password}

wifi:
  ssid: ${wifi_ssid}
  password: ${wifi_password}
  manual_ip:
   static_ip: ${ip}
   gateway: ${gw}
   subnet: ${subnet}
   dns1: ${dns}
  # domain: ${domain}
  ##### Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "${device_name}"
    password: ${wifi_password}

captive_portal:

remote_transmitter:
  pin: P7
  carrier_duty_percent: 50%

climate:
  - platform: heatpumpir       # adjust to match your AC unit!
    protocol: panasonic_nke
    horizontal_default: mright
    vertical_default: middle
    max_temperature: 30
    min_temperature: 16
    name: "AC Unit"

switch:
  - platform: gpio
    name: "Blue Led"
    pin:
      number: P24
      inverted: false
```

Cossid, 2023-10-21:
```yaml
  framework:
    version: dev
```

If you're using official esphome, you should not be using the dev framework, just remove those lines.

X-Ryl669, 2023-10-21:
You are probably right, but that doesn't change anything to the issue that IRremoteESP8266 is built (while it shouldn't) and it failed.

kuba2k2, 2023-10-21:
Your config still doesn't pull IRremoteESP8266 library. Log:
```
Found 5 compatible libraries
Scanning dependencies...
Dependency Graph
|-- AsyncTCP-esphome @ 2.0.1
|-- ESPAsyncWebServer-esphome @ 3.1.0
|-- DNSServer @ 1.1.0
|-- ArduinoJson @ 6.18.5
|-- HeatpumpIR @ 1.0.23
Building in release mode
```

X-Ryl669, 2023-10-21:
Did you get it to compile without the same errors as the OP ?

kuba2k2, 2023-10-21:
Yes; however, there are other errors in `HeatpumpIR/IRSenderPWM.cpp`, because it's not adapted for LibreTiny, and tries to use AVR registers.

X-Ryl669, 2023-10-21:
That's strange it behaves differently on your version and mine. In the dependency graph, I also have `|-- noise-c @ 0.1.4` which you don't. I've posted a patch for the IRSenderPWM.cpp in #163. But that's not part of Libretiny to solve I guess.

Are you using last release of ESPHome (mine is 2023.10.1) ?

kuba2k2, 2023-10-21:
Yes, I'm using the latest release (dev branch). `noise-c` is used if you have API encryption; I've disabled it since I don't have the required secrets.

X-Ryl669, 2023-10-21:
Here the complete build log:
```txt
INFO ESPHome 2023.10.1
INFO Reading configuration /config/esphome/irremotesouth.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing irremotesouth (board: cbu; framework: arduino; platform: https://github.com/libretiny-eu/libretiny.git)
--------------------------------------------------------------------------------
HARDWARE: BK7231N 120MHz, 256KB RAM, 1.03MB Flash
 - framework-arduino-api @ 2022.8.24+sha.237b10a 
 - framework-beken-bdk @ 0.0.0+v2021.06.07.sha.6491b8c 
 - library-flashdb @ 1.2.0+sha.d5c892f 
 - library-freertos @ 9.0.0+sha.95cc959 
 - library-freertos-port @ 2023.5.23+sha.a917d93 
 - library-lwip @ 2.1.3-bdk+sha.4ee4d34 
 - library-printf @ 6.1.0+sha.28a79bd 
 - library-uf2ota @ 5.0.0+sha.f955412 
PLATFORM VERSIONS:
 - libretiny @ 1.4.1+sha.1ed0000
 - ltchiptool @ 4.6.0
CUSTOM OPTIONS:
 - fw_name = esphome
 - fw_version = 2023.10.1
Dependency Graph
|-- AsyncTCP-esphome @ 2.0.1
|-- ESPAsyncWebServer-esphome @ 3.1.0
|-- DNSServer @ 1.1.0
|-- noise-c @ 0.1.4
|-- ArduinoJson @ 6.18.5
|-- HeatpumpIR @ 1.0.23
Compiling .pioenvs/irremotesouth/lib84c/IRremoteESP8266/IRrecv.cpp.o
Compiling .pioenvs/irremotesouth/lib84c/IRremoteESP8266/ir_Aiwa.cpp.o
Compiling .pioenvs/irremotesouth/lib84c/IRremoteESP8266/ir_Amcor.cpp.o
Compiling .pioenvs/irremotesouth/lib84c/IRremoteESP8266/ir_Argo.cpp.o
Compiling .pioenvs/irremotesouth/lib84c/IRremoteESP8266/ir_Arris.cpp.o
Compiling .pioenvs/irremotesouth/lib84c/IRremoteESP8266/ir_Bosch.cpp.o
Compiling .pioenvs/irremotesouth/lib84c/IRremoteESP8266/ir_Bose.cpp.o
Compiling .pioenvs/irremotesouth/lib84c/IRremoteESP8266/ir_Carrier.cpp.o
/piolibs/IRremoteESP8266/src/IRrecv.cpp:140:7: error: '_IRrecv' has not been declared
  140 | using _IRrecv::timer;
      |       ^~~~~~~
/piolibs/IRremoteESP8266/src/IRrecv.cpp:174:3: error: expected unqualified-id before 'if'
  174 |   if (params.rawlen) params.rcvstate = kStopState;
      |   ^~
/piolibs/IRremoteESP8266/src/IRrecv.cpp:181:1: error: expected declaration before '}' token
  181 | }
      | ^
/piolibs/IRremoteESP8266/src/IRrecv.cpp:184:27: error: expected initializer before 'gpio_intr'
  184 | static void USE_IRAM_ATTR gpio_intr() {
      |                           ^~~~~~~~~
/piolibs/IRremoteESP8266/src/IRrecv.cpp: In member function 'void IRrecv::enableIRIn(bool)':
/piolibs/IRremoteESP8266/src/IRrecv.cpp:389:35: error: 'gpio_intr' was not declared in this scope
  389 |   attachInterrupt(params.recvpin, gpio_intr, CHANGE);
      |                                   ^~~~~~~~~
*** [.pioenvs/irremotesouth/lib84c/IRremoteESP8266/IRrecv.cpp.o] Error 1
========================== [FAILED] Took 6.93 seconds ==========================
```
Can you post yours so I can spot what's different (except for noise-c) ?

kuba2k2, 2023-10-21:
Nothing is different. After applying your patch to `IRSenderPWM.cpp`:
```
INFO ESPHome 2023.10.1
INFO Reading configuration ir.yml...
INFO Generating C++ source...
INFO Compiling app...
Processing irremotesouth (board: cbu; framework: arduino; platform: https://github.com/libretiny-eu/libretiny.git)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/libretiny/cbu.html
PLATFORM: LibreTiny (1.4.0) > CBU Wi-Fi Module
HARDWARE: BK7231N 120MHz, 256KB RAM, 1.03MB Flash
DEBUG: Current (openocd) External (custom, openocd)
PACKAGES:
 - framework-arduino-api @ 2022.8.24+sha.237b10a
 - framework-beken-bdk @ 0.0.0+v2021.06.07.sha.6491b8c
 - library-flashdb @ 1.2.0+sha.d5c892f
 - library-freertos @ 9.0.0+sha.95cc959
 - library-freertos-port @ 2023.5.23+sha.a917d93
 - library-lwip @ 2.1.3-bdk+sha.4ee4d34
 - library-printf @ 6.1.0+sha.28a79bd
 - library-uf2ota @ 5.0.0+sha.f955412
 - toolchain-gccarmnoneeabi @ 1.100301.220327 (10.3.1)
PLATFORM VERSIONS:
 - libretiny @ 1.4.1+sha.1ed0000
 - ltchiptool @ 4.5.1
CUSTOM OPTIONS:
 - fw_name = esphome
 - fw_version = 2023.10.1
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ soft
Found 5 compatible libraries
Scanning dependencies...
Dependency Graph
|-- AsyncTCP-esphome @ 2.0.1
|-- ESPAsyncWebServer-esphome @ 3.1.0
|-- DNSServer @ 1.1.0
|-- ArduinoJson @ 6.18.5
|-- HeatpumpIR @ 1.0.23
Building in release mode
Checking size .pioenvs\irremotesouth\raw_firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [====      ]  36.3% (used 95240 bytes from 262144 bytes)
Flash: [========  ]  81.8% (used 886452 bytes from 1083136 bytes)
======================================================================= [SUCCESS] Took 25.45 seconds =======================================================================
INFO Successfully compiled program.
```

Don't mind the ltchiptool version, it's irrelevant.

X-Ryl669, 2023-10-21:
Puzzling...

It looks like the output is similar, yet it behaves differently. I don't have the LDF stuff and the "Found 5 compatible libraries". I was able to build it with my patches, but it was when I used the docker image (libretiny, before it was merged in official ESPHome), not HA's ESPHome tab. 

How can I track what's going wrong with my setup then? I find the HA integration of ESPHome hard to debug because of the numerous docker images. I can patch the files, but next update will nuke them anyway.

X-Ryl669, 2023-10-21:
I've cleaned everything and entered the ESPHome container to run `esphome -v run myfile.yaml`, the complete output is [here for stderr](https://pastebin.com/raw/XrjgMSqW)
and [here for stdout](http://pastebin.fr/131716)

Maybe you can spot what's wrong.

kuba2k2, 2023-10-21:
Well:
```
Compiling .pioenvs\irremotesouth\lib401\IRremoteESP8266\ir_Daikin.cpp.o
.piolibdeps/irremotesouth/IRremoteESP8266/src/IRrecv.cpp:140:7: error: '_IRrecv' has not been declared
  140 | using _IRrecv::timer;
      |       ^~~~~~~
.piolibdeps/irremotesouth/IRremoteESP8266/src/IRrecv.cpp:174:3: error: expected unqualified-id before 'if'
  174 |   if (params.rawlen) params.rcvstate = kStopState;
      |   ^~
.piolibdeps/irremotesouth/IRremoteESP8266/src/IRrecv.cpp:181:1: error: expected declaration before '}' token
  181 | }
      | ^
.piolibdeps/irremotesouth/IRremoteESP8266/src/IRrecv.cpp:184:27: error: expected initializer before 'gpio_intr'
  184 | static void USE_IRAM_ATTR gpio_intr() {
      |                           ^~~~~~~~~
.piolibdeps/irremotesouth/IRremoteESP8266/src/IRrecv.cpp: In member function 'void IRrecv::enableIRIn(bool)':
.piolibdeps/irremotesouth/IRremoteESP8266/src/IRrecv.cpp:389:35: error: 'gpio_intr' was not declared in this scope
  389 |   attachInterrupt(params.recvpin, gpio_intr, CHANGE);
      |                                   ^~~~~~~~~
*** [.pioenvs\irremotesouth\lib401\IRremoteESP8266\IRrecv.cpp.o] Error 1
======================================================================= [FAILED] Took 17.25 seconds =======================================================================
```

What I did was to change the platform to ESP8266, which made PIO download the library. Then, I changed it back to BK72xx, but the library stayed there.

As soon as I remove the `IRremoteESP8266` directory from `.piolibdeps`, the error goes away. Renaming the directory doesn't help - it seems like PIO is just including every library it finds there. This is a PlatformIO bug.

MayorBug, 2023-11-06:
I also bumped also into this issue, is there be a way to work around this issue?

X-Ryl669, 2023-11-06:
Ok, gave another try today and I can't reproduce your conclusions. In my esphome docker that was updated today, I don't have `IRremoteESP8266` in any `.piolibdeps` folder. The only places I'm seeing them is in `/data/build/irremotesouth/.pioenvs/irremotesouth/lib84c/IRremoteESP8266`. I made sure I did not change the platform to anything other than `bk72xx/cbu`

I've deleted the folder and it instantly reappeared when I retried building and obviously, it fails.

```bash
root@5c53de3b-esphome:/# find -name "IRremoteESP8266"
./data/build/irremotesouth/.pioenvs/irremotesouth/lib84c/IRremoteESP8266
./data/build/irremotewest/.pioenvs/irremotewest/lib84c/IRremoteESP8266
./piolibs/IRremoteESP8266
root@5c53de3b-esphome:/# find -name "IRremoteESP8266" -print0 | xargs -0 rm -r
root@5c53de3b-esphome:/# find -name "IRremoteESP8266"
root@5c53de3b-esphome:/#
```

Building still fails, something is still pulling that library in the current building progress.

MayorBug, 2023-11-06:
@X-Ryl669  Thanks man, that worked, Now I can use this S09 IR controller with my panasonic heatpump

```
INFO ESPHome 2023.9.0-dev
INFO Reading configuration /config/libretuya-esphome/s09-ir.yaml...
INFO Detected timezone 'Europe/Tallinn'
INFO Generating C++ source...
INFO Compiling app...
Processing s09-ir (board: wb3s; framework: arduino; platform: libretiny @ 1.2.0)
--------------------------------------------------------------------------------
Library Manager: Installing esphome/noise-c @ 0.1.4
INFO Installing esphome/noise-c @ 0.1.4
Unpacking  [####################################]  100%
Library Manager: noise-c@0.1.4 has been installed!
INFO noise-c@0.1.4 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/libsodium @ 1.10018.1
INFO Installing esphome/libsodium @ 1.10018.1
Unpacking  [####################################]  100%
Library Manager: libsodium@1.10018.1 has been installed!
INFO libsodium@1.10018.1 has been installed!
Library Manager: Installing git+https://github.com/X-Ryl669/arduino-heatpumpir#Libretiny @ 1.0.24
INFO Installing git+https://github.com/X-Ryl669/arduino-heatpumpir#Libretiny @ 1.0.24
git version 2.30.2
Cloning into '/data/cache/platformio/cache/tmp/pkg-installing-_7i0blu_'...
Library Manager: LTHeatpumpIR@1.0.24+sha.095cfff has been installed!
INFO LTHeatpumpIR@1.0.24+sha.095cfff has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Warning! Non-Git installations are NOT SUPPORTED.
HARDWARE: BK7231T 120MHz, 256KB RAM, 1.03MB Flash
 - framework-arduino-api @ 2022.8.24+sha.237b10a 
 - framework-beken-bdk @ 0.0.0+v2021.06.07.sha.6491b8c 
 - library-flashdb @ 1.2.0+sha.d5c892f 
 - library-freertos @ 9.0.0+sha.95cc959 
 - library-freertos-port @ 2023.5.23+sha.a917d93 
 - library-lwip @ 2.1.3-bdk+sha.4ee4d34 
 - library-printf @ 6.1.0+sha.28a79bd 
 - library-uf2ota @ 5.0.0+sha.f955412 
PLATFORM VERSIONS:
 - libretiny @ 1.2.0
 - ltchiptool @ 4.7.0
CUSTOM OPTIONS:
 - fw_name = esphome
 - fw_version = 2023.9.0-dev
Dependency Graph
|-- noise-c @ 0.1.4
|-- HeatpumpIR @ 1.0.24+sha.095cfff
Compiling /data/s09-ir/.pioenvs/s09-ir/src/esphome/components/api/api_connection.cpp.o
Compiling /data/s09-ir/.pioenvs/s09-ir/src/esphome/components/api/api_frame_helper.cpp.o
Compiling /data/s09-ir/.pioenvs/s09-ir/src/esphome/components/api/api_pb2.cpp.o
Compiling /data/s09-ir/.pioenvs/s09-ir/src/esphome/components/api/api_pb2_service.cpp.o
Compiling /data/s09-ir/.pioenvs/s09-ir/src/esphome/components/api/api_server.cpp.o
Compiling /data/s09-ir/.pioenvs/s09-ir/src/esphome/components/api/list_entities.cpp.o
Compiling /data/s09-ir/.pioenvs/s09-ir/src/esphome/components/api/proto.cpp.o
Compiling /data/s09-ir/.pioenvs/s09-ir/src/esphome/components/api/subscribe_state.cpp.o
Compiling /data/s09-ir/.pioenvs/s09-ir/src/esphome/components/api/user_services.cpp.o
Compiling /data/s09-ir/.pioenvs/s09-ir/src/esphome/components/binary_sensor/automation.cpp.o
Compiling /data/s09-ir/.pioenvs/s09-ir/src/esphome/components/binary_sensor/binary_sensor.cpp.o
Compiling /data/s09-ir/.pioenvs/s09-ir/src/esphome/components/binary_sensor/filter.cpp.o
Compiling /data/s09-ir/.pioenvs/s09-ir/src/esphome/components/climate/climate.cpp.o
Compiling /data/s09-ir/.pioenvs/s09-ir/src/esphome/components/climate/climate_mode.cpp.o
Compiling /data/s09-ir/.pioenvs/s09-ir/src/esphome/components/climate/climate_traits.cpp.o
Compiling /data/s09-ir/.pioenvs/s09-ir/src/esphome/components/climate_ir/climate_ir.cpp.o
Compiling /data/s09-ir/.pioenvs/s09-ir/src/esphome/components/heatpumpir/heatpumpir.cpp.o
Compiling /data/s09-ir/.pioenvs/s09-ir/src/esphome/components/heatpumpir/ir_sender_esphome.cpp.o
Compiling /data/s09-ir/.pioenvs/s09-ir/src/esphome/components/libretiny/core.cpp.o
Compiling /data/s09-ir/.pioenvs/s09-ir/src/esphome/components/libretiny/gpio_arduino.cpp.o
Compiling /data/s09-ir/.pioenvs/s09-ir/src/esphome/components/libretiny/lt_component.cpp.o
[... 568 lines cut ...]
|   |-- image_bk7231t_app.0x011000.rbl
|   |   |-- raw_firmware.bin
|   |-- image_bk7231t_app.0x011000.crc
|   |-- image_bk7231t_app.0x129F0A.rblh
|   |-- image_bk7231t_app.ota.rbl
|   |-- image_bk7231t_app.ota.ug.bin
RAM:   [====      ]  44.2% (used 115896 bytes from 262144 bytes)
Flash: [=======   ]  66.9% (used 724428 bytes from 1083136 bytes)
Building UF2 OTA image
|-- esphome_2023.9.0-dev_wb3s_bk7231t_lt1.2.0.uf2
|-- firmware.uf2
|-- firmware.bin
========================= [SUCCESS] Took 59.89 seconds =========================
Warning! Non-Git installations are NOT SUPPORTED.
INFO Successfully compiled program.
INFO Resolving IP address of s09-ir.local
INFO  -> 192.168.194.4
INFO Uploading /data/s09-ir/.pioenvs/s09-ir/firmware.uf2 (2483200 bytes)
Uploading: [============================================================] 100% Done...

INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from s09-ir.local using esphome API
```

zain1144, 2023-11-07:
@X-Ryl669 Thanks!
I'm using HA OS, how can I remove IRremoteESP8266 files

MayorBug, 2023-11-07:
@zain1144 Follow the guide. use the advanced SSH addon and turn off its protection mode.

huyhpa, 2023-11-14:
Thank you @X-Ryl669 , it worked.

X-Ryl669, 2023-12-05:
Just a note here:
With new ESPHome version the darn library came back. This time, I ran esphome compile in verbose mode and got this:
```
Dependency Graph
|-- AsyncTCP-esphome @ 2.0.1 (License: LGPL-3.0, Path: /config/esphome/.esphome/build/irremotewest/.piolibdeps/irremotewest/AsyncTCP-esphome)
|-- ESPAsyncWebServer-esphome @ 3.1.0 (License: LGPL-3.0, Path: /config/esphome/.esphome/build/irremotewest/.piolibdeps/irremotewest/ESPAsyncWebServer-esphome)
|   |-- AsyncTCP-esphome @ 2.0.1 (License: LGPL-3.0, Path: /config/esphome/.esphome/build/irremotewest/.piolibdeps/irremotewest/AsyncTCP-esphome)
|-- DNSServer @ 1.1.0 (License: Unknown, Path: /config/esphome/.esphome/build/irremotewest/.piolibdeps/irremotewest/DNSServer)
|-- noise-c @ 0.1.4 (License: MIT, Path: /config/esphome/.esphome/build/irremotewest/.piolibdeps/irremotewest/noise-c)
|   |-- libsodium @ 1.10018.1 (License: MIT, Path: /config/esphome/.esphome/build/irremotewest/.piolibdeps/irremotewest/libsodium)
|-- ArduinoJson @ 6.18.5 (License: Unknown, Path: /config/esphome/.esphome/build/irremotewest/.piolibdeps/irremotewest/ArduinoJson)
|-- HeatpumpIR @ 1.0.25+sha.cdfbf72 (License: Unknown, URI: git+https://github.com/X-Ryl669/arduino-heatpumpir#Libretiny, Path: /config/esphome/.esphome/build/irremotewest/.piolibdeps/irremotewest/LTHeatpumpIR)
|   |-- IRremoteESP8266 @ 2.8.6 (License: Unknown, Path: /piolibs/IRremoteESP8266)
```
So it's clear that HeatpumpIR is responsible for bringing the IRremoteESP8266 library in (although its `library.json` specifies that this library is just for ESP8266 platform, it's still being pull in).  

In the end, I've forked this library too and fixed the building issue so it'd build too when updating ESPHome. You'll still need to delete the IRremoteESP8266 library folder if you have downloaded the initial version first, since platformIO is too dumb to check that the source URL has changed and will not download my fork if it finds a folder with the right name.

Let's see if it does on next update!

preussal, 2024-12-30:
Hi X-Ryl669,

i used your code the last year, but now is working without your code.

```
substitutions:
  name: "gem-aircon-remote"
  IRReceiver: "8"
  IRTransmitter: "26"
  StatusLED: "9"
  RXpin: "RX1"
  TXpin: "TX1"

esphome:
  name: ${name}

#external_components:
#  # https://github.com/libretiny-eu/libretiny/issues/176
#  - source: github://X-Ryl669/esphome-hpir
#    components: [heatpumpir]
#    refresh: 0s

bk72xx:
  board: wb3s
  #board: generic-bk7231t-qfn32-tuya
  framework:
    version: recommended    # dev, latest, recommended

logger:
  level: DEBUG
  baud_rate: 0
  logs:
    component: ERROR
    sensor: WARN
    text_sensor: WARN
    
# Enable Home Assistant API
api:
  encryption:
    key: ""

ota:
  - platform: esphome
    id: my_ota
    password: ""
    on_begin:
      then:
        - logger.log: "BLE connection suspended for OTA update"

wifi:
  domain: !secret wifi_domain
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  use_address: 192.168.250.5

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "${name} fallback"
    password: ""

web_server:
  port: 80
  # No internet/intranet required on the clients (all assets are inlined, compressed and served from flash):
  local: true
captive_portal:

time:
  - platform: sntp
    timezone: Asia/Bangkok
    id: sntp_time
[... 137 lines cut ...]
|-- HeatpumpIR @ 1.0.27
Compiling .pioenvs/gem-aircon-remote/src/main.cpp.o
Linking .pioenvs/gem-aircon-remote/raw_firmware.elf
|-- Image 1: raw_firmware.elf
|   |-- image_bk7231t_app.0x011000.rbl
|   |   |-- raw_firmware.bin
|   |-- image_bk7231t_app.0x011000.crc
|   |-- image_bk7231t_app.0x129F0A.rblh
|   |-- image_bk7231t_app.ota.rbl
|   |-- image_bk7231t_app.ota.ug.bin
RAM:   [====      ]  44.9% (used 117808 bytes from 262144 bytes)
Flash: [========  ]  79.5% (used 861524 bytes from 1083136 bytes)
Building UF2 OTA image
|-- esphome_2024.12.2_wb3s_bk7231t_lt1.7.0.uf2
|-- firmware.uf2
|-- firmware.bin
========================= [SUCCESS] Took 10.04 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 192.168.250.5 port 8892...
INFO Connected to 192.168.250.5
INFO Uploading /data/build/gem-aircon-remote/.pioenvs/gem-aircon-remote/firmware.uf2 (2947072 bytes)
Uploading: [============================================================] 100% Done...
```

Thank you