libretiny-eu/libretiny #177: ESPHome 2023.9.1 - error "Platform missing. You must include one of the available platform key esp32, esp8266, rp2040, host"

MnM001, 2023-09-29:
I updated to ESPHome 2023.9.1.

It does mention that libretiny is natively supported now.
However get the error below:

[image]

Cossid, 2023-09-29:
Try removing
```
framework:
  version: dev
  ```

MnM001, 2023-09-29:
Same error.
The error is one the bk72xx line.... 
Cleaning build files did not solve the issue:

[image]

Cossid, 2023-09-29:
That's a different error.  You need to replace `ledc` with `libretiny_pwm` moving forward.

MnM001, 2023-09-29:
And that worked :) - thanks.

Are any more changes like this? I don't think I saw anything documented for this.

Cossid, 2023-09-29:
Nope, that is pretty much it.

closed 2023-09-29 by Cossid