libretiny-eu/libretiny #180: High power consumption on BK7231- power_save_mode in wifi component not working

rbonat, 2023-10-03:
First of all, I'm really impressed of this unexpected and awesome new feature of esphome thanks to libretiny!
I've uploaded (via UART) my first yaml configuration into single channel relay on CB2S, and it is working from the first try, reliable and stable.
But I've noticed the relay is getting warm idle, so I've measured power consumption. In my case it goes to 1.2 - 1.3 W idle, while same relay witch original tuya-soft consumes 0.5 - 0,6 W.
Apart of unnecessary power consumption (which can be an issue when you count your devices in dozens), the real problem can be the heat. Imagine device closed in wall-socket, without any air flow, dispersing 1.3W of energy in heat 24h/7d - it might lead to serious overheating issues.
I've tried different power_save_mode values of wifi component, usually working good witch esp chips, but witch no effect.
Looks like power_save_mode is not implemented for BK7231 - is it planned to be implemented in future releases?

rbonat, 2024-07-05:
Nop, I've just switched back to esp modules. Maybe new releases fixed this, but I've not checked it.

OoSTARTERoO, 2024-09-15:
I've tested it now with

LibreTiny v1.7.0
LSPA9 / BK7231N (CB2S)
hlw8012 (BL0937) enabled
Connected with HA

With power_save_mode HIGH / LOW it consumes 77mA
with it disabled, it draws around 81mA