libretiny-eu/libretiny #181: Flashing issue in ESPHome/ltchiptool while bk7231tools works fine.

evadim, 2023-10-05:
I successfully converted noname RGBCW (3+3+3+10+10 w) led downlight  which have bare BK7231T chip on led driver pcb with test points.

The issue is, what only one flasher reliably work with it is **bk7231tools**, but there are no clear manual how to use with ESPHome. Finally I found what I should flash image_bk7231t_app.0x011000.rbl via _write app firmware (from app-only binary file)_ from examples in github page.

I used 3 different usb-ttl adapters which works for ESP* modules without any issues with same test-clip wires.
There are no difference if I use onboard power or separate HiLink 3.3V power supply (ground connected to adapter).
At the end I use 6-in-1 USB/TTL/232/485 (yellow 2 row pin header) one as it have RTS pin which I connected to CEN pin.

All things tested in Gentoo Linux.

ltchiptool - show message about resetting board and no reaction:

```
ltchiptool flash write  kickstart-bk7231t-2023-08-29.uf2
I: Available COM ports:
I: |-- ttyUSB0 - CP2102 USB to UART Bridge Controller - CP2102 USB to UART Bridge Controller - Silicon Labs (10C4/EA60)
I: |   |-- Selecting this port. To override, use -d/--device
I: Detected file type: UF2 - esphome 2023.9.0-dev
I: Connecting to 'Beken 7231T' on /dev/ttyUSB0 @ 115200
I: Connect UART1 of the BK7231 to the USB-TTL adapter:
I:
I:     --------+        +------------------------
I:          PC |        | BK7231
I:     --------+        +------------------------
I:          RX | ------ | TX1 (GPIO11 / P11)
I:          TX | ------ | RX1 (GPIO10 / P10)
I:         RTS | ------ | CEN (or RST, optional)
I:             |        |
I:         GND | ------ | GND
I:     --------+        +------------------------
I:
I: Make sure to use a good 3.3V power supply, otherwise the adapter might
I: lose power during chip reset. Usually, the adapter's power regulator
I: is not enough and an external power supply is needed (like AMS1117).
I:
I: If you didn't connect RTS to CEN, after running the command you have
I: around 20 seconds to reset the chip manually. In order to do that,
I: you need to bridge CEN to GND with a wire.
E: TimeoutError: Timed out attempting to link with chip
E: |-- File "/home/evadim/ltchiptool/lib/python3.11/site-packages/bk7231tools/serial/__init__.py", line 42, in connect
```
In GUI mode I got random segfaults after flash timeout or similar app crashed errors. (And 100% if I open plugins tab) Apps istalled via pip in wheel for user, as well as bk7231tools.
Same behavior I see while flashing via ESPHome v2023.9.2 release: timeout.

I also try BK7231Flasher with Mono. It show CRC errors, but kind of fixed with increasing UART timeouts in settings. reads fine but write failed after erasing (and nothing erased I suspect).

bk7231tools in same wheel-pip env works without any issues:
full log [bk7231tools_flash.log](https://github.com/libretiny-eu/libretiny/files/12816536/bk7231tools_flash.log)
```
bk7231tools write_flash -d /dev/ttyUSB1 -s 0x11000 -S 0 image_bk7231t_app.0x011000.rbl
Connected! Chip info: BK7231S_1.0.5 / Flash ID: 00 00 00 / Protocol: BASIC_TUYA
Writing 1150832 bytes to 0x11000
Erasing and writing at 0x211000 (0.00%)
Erasing and writing at 0x212000 (0.36%)
Erasing and writing at 0x213000 (0.71%)
Erasing and writing at 0x214000 (1.07%)
Erasing and writing at 0x215000 (1.42%)
Erasing and writing at 0x216000 (1.78%)
Erasing and writing at 0x217000 (2.14%)
Erasing and writing at 0x218000 (2.49%)
.....
Erasing and writing at 0x2D2000 (68.69%)
Erasing and writing at 0x2D3000 (69.05%)
Erasing and writing at 0x2D4000 (69.40%)
Erasing at 0x2D5000 (69.76%)
Erasing at 0x2D6000 (70.12%)
Erasing at 0x2D7000 (70.47%)
Erasing at 0x2D8000 (70.83%)
Erasing at 0x2D9000 (71.18%)
Erasing at 0x2DA000 (71.54%)
Erasing at 0x2DB000 (71.90%)
Erasing at 0x2DC000 (72.25%)
.....
Erasing at 0x326000 (98.59%)
Erasing at 0x327000 (98.94%)
Erasing at 0x328000 (99.30%)
Erasing and writing at 0x329000 (99.66%)
Verifying CRC
The current command timeout of 1.0 second(s) is too low for reading 1150976 bytes CRC. Increasing to 3 second(s).
OK!

```

kuba2k2, 2023-10-05:
`bk7231tools write_flash -d /dev/ttyUSB1`
`|-- ttyUSB0
|   |-- Selecting this port. `

Do you have multiple UART adapters plugged in? It seems that bk7231tools uses `ttyUSB1` as you specified on command line, but ltchiptool tries to use ttyUSB0, because that's what it finds suitable.

evadim, 2023-10-05:
> Do you have multiple UART adapters plugged in?

Only one at time. I just copy logs from long console history. At some point they start to use ttyUSB1 name not sure why. 
I still have in dmesg that moment, where we can see this change in names:

```
[Oct 5 05:32] traps: ltchiptool[29437] general protection fault ip:7fdc3cfb1180 sp:7ffdf85c1cd0 error:0 in libpango-1.0.so.0.5000.14[7fdc3cf95000+37000]
[Oct 5 05:36] ltchiptool[31613]: segfault at fffffffffffffffc ip 00007f544d0f427e sp 00007f5440c0d678 error 5 in libgtk-3.so.0.2406.32[7f544ce78000+37d000] likely on CPU 1 (core 1, socket 0)
[  +0.000034] Code: 83 f9 01 7e 19 44 8d 49 ff 31 c0 90 45 8b 14 80 44 39 14 87 75 c6 48 ff c0 49 39 c1 75 ee 48 c1 e1 02 4d 8d 44 08 fc 41 8b 00 <39> 44 0f fc 7f ac ff c0 41 89 00 48 8b 52 08 48 85 d2 75 a7 c3 66
[Oct 5 06:23] traps: xxdiff[24197] general protection fault ip:7f867b7041d6 sp:7fff13cbd868 error:0 in libQt5Gui.so.5.15.10[7f867b522000+3df000]
[Oct 5 06:55] traps: ltchiptool[4712] general protection fault ip:7f3a39a9b180 sp:7f3a17ffe2d0 error:0 in libpango-1.0.so.0.5000.14[7f3a39a7f000+37000]
[Oct 5 08:14] usb 1-1: USB disconnect, device number 34
[  +0.000420] cp210x ttyUSB0: cp210x converter now disconnected from ttyUSB0
[  +0.000096] cp210x 1-1:1.0: device disconnected
[  +2.868990] usb 1-1: new full-speed USB device number 35 using xhci_hcd
[  +0.158941] usb 1-1: New USB device found, idVendor=10c4, idProduct=ea60, bcdDevice= 1.00
[  +0.000020] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  +0.000007] usb 1-1: Product: CP2102 USB to UART Bridge Controller
[  +0.000005] usb 1-1: Manufacturer: Silicon Labs
[  +0.000029] usb 1-1: SerialNumber: 0001
[  +0.009953] cp210x 1-1:1.0: cp210x converter detected
[  +0.001248] usb 1-1: cp210x converter now attached to ttyUSB0
[Oct 5 08:16] usb 1-1: USB disconnect, device number 35
[  +0.000213] cp210x ttyUSB0: failed set request 0x7 status: -19
[  +0.000015] cp210x ttyUSB0: failed set request 0x12 status: -19
[  +0.000007] cp210x ttyUSB0: failed set request 0x0 status: -19
[  +0.000266] cp210x ttyUSB0: cp210x converter now disconnected from ttyUSB0
[  +0.000059] cp210x 1-1:1.0: device disconnected
[Oct 5 08:18] usb 1-1: new full-speed USB device number 36 using xhci_hcd
[  +0.157955] usb 1-1: New USB device found, idVendor=10c4, idProduct=ea60, bcdDevice= 1.00
[  +0.000021] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  +0.000008] usb 1-1: Product: CP2102 USB to UART Bridge Controller
[  +0.000007] usb 1-1: Manufacturer: Silicon Labs
[  +0.000006] usb 1-1: SerialNumber: 0001
[  +0.009782] cp210x 1-1:1.0: cp210x converter detected
[  +0.001015] usb 1-1: cp210x converter now attached to ttyUSB1
[Oct 5 08:30] ltchiptool[22389]: segfault at fffffffffffffffc ip 00007fddfdbaa27e sp 00007fddea101678 error 5 in libgtk-3.so.0.2406.32[7fddfd92e000+37d000] likely on CPU 5 (core 1, socket 0)
[  +0.000020] Code: 83 f9 01 7e 19 44 8d 49 ff 31 c0 90 45 8b 14 80 44 39 14 87 75 c6 48 ff c0 49 39 c1 75 ee 48 c1 e1 02 4d 8d 44 08 fc 41 8b 00 <39> 44 0f fc 7f ac ff c0 41 89 00 48 8b 52 08 48 85 d2 75 a7 c3 66
[Oct 5 08:31] ltchiptool[22416]: segfault at 10008 ip 00007fcf8fc07e2f sp 00007ffc28052a30 error 4 in libpango-1.0.so.0.5000.14[7fcf8fbf6000+37000] likely on CPU 2 (core 2, socket 0)
[  +0.000065] Code: 1f 84 00 00 00 00 00 0f 1f 40 00 55 53 48 83 ec 08 48 85 ff 74 45 48 89 fb e8 bd ff ff ff 48 8b 7b 08 48 89 c5 48 85 ff 74 07 <8b> 47 08 85 c0 75 0a 48 83 c4 08 5b 48 89 e8 5d c3 48 8d 35 19 ef
[Oct 5 10:07] usb 1-1: USB disconnect, device number 36
[  +0.000464] cp210x ttyUSB1: cp210x converter now disconnected from ttyUSB1
[  +0.000136] cp210x 1-1:1.0: device disconnected
[Oct 5 11:05] elogind-daemon[1643]: Suspending system...

```
IIRC after experiments with BK7231Flasher. Today after resume from hibernation they again use ttyUSB0.
Below logs from ltchiptool gui run, it just have a lot more noise in console:

```
(ltchiptool) evadim@t480 ~/ltchiptool $ ./bin/ltchiptool gui
21:29:35: Debug: Adding duplicate image handler for 'Windows bitmap file'
21:29:35: Debug: Adding duplicate animation handler for '1' type
21:29:35: Debug: Adding duplicate animation handler for '2' type

(ltchiptool:23023): Gtk-WARNING **: 21:29:35.366: Negative content height -3 (allocation 7, extents 5x5) while allocating gadget (node scrollbar, owner GtkScrollbar)

(ltchiptool:23023): Gtk-CRITICAL **: 21:29:35.366: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkScrollbar
E: wxAssertionError: C++ assertion ""start >= 0 && end <= l"" failed at /tmp/pip-install-s8d5w2tf/wxpython_dac74386d5414e4b8cbdb95bc2a81cf9/ext/wxWidgets/src/gtk/textctrl.cpp(1980) in SetStyle(): invalid range in wxTextCtrl::SetStyle
E: |-- File "/home/evadim/ltchiptool/lib/python3.11/site-packages/ltchiptool/gui/colors.py", line 128, in apply
I: Found new device: ttyUSB1 - CP2102 USB to UART Bridge Controller - CP2102 USB to UART Bridge Controller - Silicon Labs (10C4/EA60)
I: Connect UART1 of the BK7231 to the USB-TTL adapter:
I:
I:     --------+        +------------------------
I:          PC |        | BK7231
I:     --------+        +------------------------
I:          RX | ------ | TX1 (GPIO11 / P11)
I:          TX | ------ | RX1 (GPIO10 / P10)
I:         RTS | ------ | CEN (or RST, optional)
I:             |        |
[... 51 lines cut ...]
Reading 4k page at 0x3E8000 (95.31%)
Reading 4k page at 0x3E9000 (95.51%)
Reading 4k page at 0x3EA000 (95.70%)
[... cut ...]
Reading 4k page at 0x3F3000 (97.46%)
Reading 4k page at 0x3F4000 (97.66%)
Reading 4k page at 0x3F5000 (97.85%)
Reading 4k page at 0x3F6000 (98.05%)
Reading 4k page at 0x3F7000 (98.24%)
Reading 4k page at 0x3F8000 (98.44%)
Reading 4k page at 0x3F9000 (98.63%)
Reading 4k page at 0x3FA000 (98.83%)
Reading 4k page at 0x3FB000 (99.02%)
Reading 4k page at 0x3FC000 (99.22%)
Reading 4k page at 0x3FD000 (99.41%)
Reading 4k page at 0x3FE000 (99.61%)
Reading 4k page at 0x3FF000 (99.80%)
(ltchiptool) evadim@t480 ~/ltchiptool $

kuba2k2, 2023-10-06:
1. It is trying to reset the chip by using RTS or DTR lines. This, however, doesn't really work, so it's best to not connect these lines. On the other hand, when you flash ESPHome to the chip, ltchiptool will be able to communicate with it and reset it automatically.
2. You can enable verbose mode with -v or -vv on the ltchiptool command line. You can also enable -s (or -S, not sure) to make it dump all bytes transferred over UART.

evadim, 2023-10-06:
With bk7231tools auto reset works 100% time for me.
With ltchiptool it not work and disconnecting +3.3 also not help.

I will try to get time, solder wires back and try to debug.