libretiny-eu/libretiny #182: WiFi connection error after running function WiFi.begin(ssid,password)

TuanAIoT, 2023-10-07:
WiFi connection error after running function WiFi.begin(ssid,password), 
I use CBU module. Does anyone have a sample example of connecting to wifi, then connecting to mqtt?WiFi connection error after running function WiFi.begin(ssid,password), 
i, then connecting to mqtt?

[image]

kuba2k2, 2023-10-07:
The reset reason says "power on", which means that the chip just received power to boot up. This is likely caused by loose (or too thin) wires, or a weak power supply. Connecting to wifi draws too much current, and the chip reboots.

TuanAIoT, 2023-10-09:
Thank you, initially I used VCC = 3.3V but when I re-measured the voltage it was only 3.2V, that's why the chip rebooted.
I repaired it using a 3.6V power source and the machine works normally, thank you very much

closed 2023-10-10 by Cossid